ardera/flutter-pi #108: How to use own firebase_core_linux plugin

matgrz1993, 2020-10-17:
Hello again,

I've create an initial [firebase_core_linux](https://github.com/matgrz1993/firebase_linux_app/tree/main/firebase_core/firebase_core_linux) plugin 
based on [url_launcher_linux](https://github.com/flutter/plugins/tree/master/packages/url_launcher/url_launcher_linux) plugin.
But I don't know how to call MethodChannel `initializeCore`.
When running `flutter-pi` it crashed with:
```
flutter: Observatory listening on http://127.0.0.1:33735/fASSxqf3JC4=/
[ERROR:engine/src/flutter/lib/ui/ui_dart_state.cc(177)] Unhandled Exception: MissingPluginException(No implementation found for method Firebase#initializeCore on channel plugins.flutter.io/firebase_core)
#0      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:157:7)
<asynchronous suspension>
#1      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:332:12)
#2      MethodChannel.invokeListMethod (package:flutter/src/services/platform_channel.dart:345:41)
#3      MethodChannelFirebase._initializeCore (package:firebase_core_platform_interface/src/method_channel/method_channel_firebase.dart:30:36)
#4      MethodChannelFirebase.initializeApp (package:firebase_core_platform_interface/src/method_channel/method_channel_firebase.dart:75:13)
#5      Firebase.initializeApp (package:firebase_core/src/firebase.dart:43:25)
#6      main (package:firebase_linux_app/main.dart:6:18)
#7      _runMainZoned.<anonymous closure>.<anonymous closure> (dart:ui/hooks.dart:231:25)
#8      _rootRun (dart:async/zone.dart:1190:13)
#9      _CustomZone.run (dart:async/zone.dart:1093:19)
#10     _runZoned (dart:async/zone.dart:1630:10)
#11     runZonedGuarded (dart:async/zone.dart:1618:12)
#12     _runMainZoned.<anonymous closure> (dart:ui/hooks.dart:223:5)
#13     _startIsolate.<anonymous closure> (dart:isolate-patch/isolate_patch.dart:301:19)
#14     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:168:12)
```

ardera, 2020-10-19:
Do you want to use flutter-pi or the official linux desktop embedding? Since you're using GTK and flutter_linux headers in your plugin, neither of which are present in flutter-pi. The whole flutter plugin API is not present in flutter-pi.

If you want to add native code to flutter-pi, you need to write it inside flutter-pi. See the [`src/plugins` directory](https://github.com/ardera/flutter-pi/tree/master/src/plugins). So just create a fork and write your plugin in there. I know, that's suboptimal and isn't really "compliant" with the flutter way of doing things, but there's no better solution.

matgrz1993, 2020-10-22:
Ok, thank you!

closed 2020-10-22 by matgrz1993