ardera/flutter-pi #109: Observatory 'print' command support

GoRadioGo, 2020-10-18:
flutter-pi is very interesting.
Set up a RPI4, running `flutter_gallery` application, and that work well.

Exploring the Observatory operations via Chrome browser.
Breakpoints work well.
I noticed that 'print' operations do not work. Example ..
$ print _textScaleFactor
{details: _compileExpression: No compilation service available; cannot evaluate from source.}

This would help make debugging that much more powerful. Is there something that can enable 'print' commands ?

ardera, 2021-04-07:
Sorry, completely forgot about this. It's probably caused by `flutter build bundle` stripping away the part of the `kernel_blob.bin` that can compile dart code into code that can be executed by the dart VM. I don't have a solution right now, but we'll probably have to use `flutter assemble` somehow to get it working.

GoRadioGo, 2021-04-07:
Hey no worries.  
We have enjoyed using flutter-pi, and are close to finishing off an application using it.
Kudos for the good work.

I'm still running an older version of Flutter engine binaries (1.22.6), but will migrate to the 2.x engine binaries in a couple of weeks.

On Observatory related question, would you expect Flutter-pi to support the newer DevTools ?

ardera, 2021-04-07:
> Hey no worries.
> We have enjoyed using flutter-pi, and are close to finishing off an application using it.
> Kudos for the good work.

thx!

> On Observatory related question, would you expect Flutter-pi to support the newer DevTools ?

So generally, everything observatory related is happening inside the flutter engine, so as long as the flutter engine works with your devtools everything else should work fine too. I'm not sure they make any guarantees about inter-version compatibility though. I think the best way to find out would be just asking on the [flutter discord](https://discord.gg/BS8KZyg)