ardera/flutter-pi #112: Running flutter-pi on PiTFT 2.8" doesn't work

MrCustomizer, 2020-10-24:
I'm trying to show Flutter apps on my PiTFT 2.8" capacitive display. I've connected the PiTFT to a Raspberry Pi 3 Model B and have followed [this guide](https://learn.adafruit.com/adafruit-pitft-3-dot-5-touch-screen-for-raspberry-pi/easy-install-2) to set it up in HDMI mirror mode. I chose this mode as it is mentioned in the guide that only this mode supports OpenGL and flutter-pi seems to be dependent on OpenGL.
Trying to start the flutter gallery app with flutter-pi results in the following error:
```
[flutter-pi] Could not query DRM device list: No such file or directory
```
I'm guessing it tries to list /dev/dri, but this directory does not exist in my current setup.

Is it theoretically possible to run Flutter apps on this PiTFT and if so, what can I do to help find the cause of the problem? I've attached the output of raspinfo.
[raspinfo.txt](https://github.com/ardera/flutter-pi/files/5433932/raspinfo.txt)

ardera, 2020-10-25:
`/dev/dri` should not be empty, thats the directory of connected GPU devices. Did you make sure you have the fake-KMS gl driver enabled in `raspi-config`? Maybe something in the TFT setup turned it back to `legacy`

Also, can you give me the complete output of `raspinfo`?

MrCustomizer, 2020-10-25:
> /dev/dri should not be empty, thats the directory of connected GPU devices. Did you make sure you have the fake-KMS gl driver enabled in raspi-config? Maybe something in the TFT setup turned it back to legacy

Good guess! I ran the setup script for the TFT multiple times with different settings. One (or all) of these runs must have turned it back.
After switching to fake-KMS again, it started working. Thank you very much.
There seems to be only one problem left: The screen resolution isn't set correctly. Is there a way to force a resolution? I haven't found one yet.
```
[flutter-pi] WARNING: display didn't provide valid physical dimensions.
```

ardera, 2020-10-25:
Can you try adding the following lines to your `/boot/config.txt`?

```
hdmi_cvt=320 240 60 1
hdmi_group=2
mide_mode=87
hdmi_drive=2
hdmi_force_mode
```

See also [this reference on the `config.txt` video options](https://www.raspberrypi.org/documentation/configuration/config-txt/video.md).

MrCustomizer, 2020-10-25:
Most of the options were already set. I only had to add `hdmi_drive=2` and `hdmi_force_mode=1` (btw. you probably meant hdmi_mode instead of mide_mode, right?). Sadly, these settings didn't change anything.

I've started experimenting with the line `hdmi_cvt=320 240 60 1` (which has some more parameters on my machine: `hdmi_cvt=320 240 60 1 0 0 0`). Reducing the width from 320 to 80 has the effect, that the display of the console on the TFT is pretty much distorted, but when starting `flutter-pi`, the horizontal resolution is changed:

```
[flutter-pi] WARNING: display didn't provide valid physical dimensions.
             The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
===================================
display mode:
  resolution: 320 x 240
  refresh rate: 51Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
===================================
```

As the displayed vertical resolution has been correct from the start, I'm guessing there is something wrong with the interpretation of the horizontal resolution.

> Btw, do you have a HDMI monitor connected to the Pi (to use the mirroring)?

Currently not. `hdmi_force_hotplug` is already set on my machine.

> I also recommend invoking flutter-pi with the --dimensions option 

Thanks. I will do that.

ardera, 2020-10-25:
> which has some more parameters on my machine: `hdmi_cvt=320 240 60 1 0 0 0`

The additional parameters are not necessary. If they're not specified, they are initialized to their default values (which is `0` for these specific params)

`hdmi_cvt` seems to not like setting resolutions that low. I got it working using `hdmi_timings`:
```
hdmi_force_hotplug=1
hdmi_timings=320 0 8 32 40 240 1 3 4 6 0 0 0 40 0 4048000 1
hdmi_group=2
hdmi_mode=87
hdmi_force_mode=1
```

Though, if you actually connect a HDMI monitor to the HDMI port, it won't show anything since the Pi doesn't work with pixel clocks as low as 4MHz. (The 2nd-last parameter of hdmi_timings).
This btw also enforces a screen refresh-rate of 40Hz, since `fbcp` copies the buffers to the TFT at 40Hz too, so it doesn't make sense to use anything more than that. (You could probably modify `fbcp` to copy it faster than that, though, and modify the hdmi timings accordingly)

MrCustomizer, 2020-10-25:
I have to admit, I don't know what you did here, but it works! :rofl: I guess I will dig into the specifics of these settings, later. As I don't want to connect a monitor to the HDMI port, that's a perfectly suitable solution for me. Thanks a lot.
I'm closing this issue now as everything seems to be working to my satisfaction.

closed 2020-10-25 by MrCustomizer