ardera/flutter-pi #119: Dart Error: Can't load Kernel binary: Invalid kernel binary format version.

nanfang2000, 2020-11-08:
I've done "flutter channel stable && flutter upgrade"
but still get this error when run the simple app:
[ERROR:engine/src/flutter/shell/common/shell.cc(210)] Dart Error: Can't load Kernel binary: Invalid kernel binary format version.
[ERROR:engine/src/flutter/shell/common/engine.cc(220)] Could not prepare to run the isolate.
[ERROR:engine/src/flutter/shell/common/engine.cc(159)] Engine not prepare and launch isolate.
[ERROR:engine/src/flutter/shell/common/shell.cc(491)] Could not launch engine with configuration.
flutter: Observatory listening on http://127.0.0.1:32771/uTjhwBkUzr4=/

flutter doctor shows:
Flutter (Channel stable, 1.22.3, on Mac OS X 10.15.3 19D76

Could you help on that?
do I need to config flutter or something?

I just tried to run 'flutter build bundle' to get the asset and not the release mode.

Thank you very much!

ardera, 2020-11-08:
Have you made sure the asset bundle on your Raspberry Pi is up to date?
To be really certain, you can check the modification timestamp of the `flutter_assets/kernel_blob.bin` file on your Raspberry Pi matches the modification timestamp of the same file on your host machine. (Using `stat`) If they don't match, something went wrong when deploying the assets.

nanfang2000, 2020-11-08:
Thank you for your advise. The app bundle must be not updated with the latest flutter because it is an old app. after I create a new app it works well.
Closing the issue.

closed 2020-11-08 by nanfang2000