ardera/flutter-pi #121: Can not find sdk-root option when building kernel snapshot

nanfang2000, 2020-11-08:
I'm building the kernel snapshot. The dart command shows:
`Could not find an option named "sdk-root".`
Here is my commands to build the kernel snapshot:
```
../flutter/bin/cache/dart-sdk/bin/dart \
  ../flutter/bin/cache/dartsdk/bin/snapshots/frontend_server.dart.snapshot \
  --sdk-root ../flutter/bin/cache/artifacts/engine/common/flutter_patched_sdk_product \
  --target=flutter \
  --aot \
  --tfa \
  -Ddart.vm.product=true \
  --packages .packages \
  --output-dill build\kernel_snapshot.dill \
  --verbose \
  --depfile build\kernel_snapshot.d \
  package:gallery/main.dart
```

ardera, 2020-11-08:
You have a typo in line 2. You wrote `dartsdk` instead of `dart-sdk`.

nanfang2000, 2020-11-09:
Sorry, my bad. Now it works.
One more question, what does 'gallery/main.dart' option mean? I see the main.dart is locate in lib folder and there is no gallery folder.

ardera, 2020-11-09:
The `package:gallery/main.dart` is not real path that exists on your computer. It's more like a pseudo path. The real path of it can be found out in combination with the `.packages` file.

If you look into the .packages file, somewhere there'll be the line:
```
gallery:file://<some path here>/lib
```

When the `frontend_server.dart.snapshot` then tries to find out the actual path on disk to `package:gallery/main.dart` it substitutes the `package:gallery` with whatever filepath it finds for the `gallery` package in the `.packages` file. So
the actual path on disk is then `file://<some path here>/lib/main.dart`.

nanfang2000, 2020-11-09:
@ardera Thank you very much, I'm a newbie to flutter. 
I finally failed at building the app.so:
./buildapp.sh: line 1: 11578 Killed                  ./gen_snapshot_linux_x64 --causal_async_stacks --deterministic --snapshot_kind=app-aot-elf --elf=build/app.so --strip --sim_use_hardfp --no-use-integer-division build/kernel_snapshot.dill

Not sure why it got killed before building finish

ardera, 2020-11-09:
The 'Killed' message means linux killed either your script or the `gen_snapshot_linux_x64` executable for some reason. Maybe you don't have enough memory.

See https://stackoverflow.com/questions/726690/what-killed-my-process-and-why for mote information.

nanfang2000, 2020-11-09:
thank you, @ardera ! yes, after I increase the memory of my Linux vm to 4GB, the building pass. 
now I successfully ran the app in release mode. BTW, my hardware is a custom a33 sbc. your flutter pi is awsome and works without even a line changed!

closed 2020-11-11 by nanfang2000

ardera, 2020-11-11:
> thank you, @ardera ! yes, after I increase the memory of my Linux vm to 4GB, the building pass.
> now I successfully ran the app in release mode. BTW, my hardware is a custom a33 sbc. your flutter pi is awsome and works without even a line changed!

Nice! Yeah I always try to make sure I'm not relying on raspberry pi specific features too much.