ardera/flutter-pi #123: Touch Issue

vigno88, 2020-11-19:
Hi!

First thanks a lot for your port, this is amazing! I'm new to flutter and I would like to use it as a ui framework for raspberry pi so I buil the 'flutter gallery' example and everything worked fine, I used a touchscreen and it worked like a charm. So I did one of
the example that is on the google docs website build it on my host. Then I repeated the exact same procedure that I did for
the flutter gallery to run it on the pi, but with my new super basic app (  [the shoplist example](https://flutter.dev/docs/development/ui/widgets-intro) ) won't respond to any touch input. I tried rebuilding it many time, adding some small changes but nothing works. Is there something that I am missing ? Do I need to add a specific package or configure something for it to works?

Thanks a lot for your help,

Nathan

ardera, 2020-11-19:
Hey,
if it works with flutter gallery but doesn't with your test app, it's likely caused by your test app. Flutter-pi's touch support should be the same on all apps.

There's a subreddit and a discord where you can get help debugging your app, see the [flutter community page](https://flutter.dev/community).

When you change the state of any stateful widgets, make sure you're calling the `setState` function, since otherwise the `build` function won't be called. Maybe you forgot to call `setState` in some touch/ button press callback. When you don't call `setState`, flutter assumes the state of your stateful widget didn't change, or that it did change, but not in a way affecting the output of the `build` function.

vigno88, 2020-11-19:
My test app works when I use the android emulator on my desktop. I have tested it before building it for the pi. But thanks for your quick answer! Now I know that the problem is somewhere on my side and that I haven't missed anything important.

vigno88, 2020-11-20:
Hey again,

My app runs fine on an android emulator on my pc. I tried doing the exact same thing that you said in the readme and I still don't have any touch support even though it works well for the flutter gallery. I tried to replace the main of the flutter_gallery project with the main of my own project and i still don't have touch support. Do you have a sample project near that you know will work so I can try to debug with it?

I use a raspberry pi model 3 b+

ardera, 2020-11-20:
Actually, I mostly use flutter gallery for testing. All the other apps I have depend on specific peripherals (for example an ELM327 connected via Bluetooth, a SI468X connected via SPI)

If you want, you can send me your app and I'll try to get it working (or publish it as a github repo so I can fork it)

vigno88, 2020-11-20:
Oh nice! You are building application for cars? 

I think I was not patient enough.. In --release i have still no touch, but without the --release the app wouldn't open so I always shut it after a few seconds but this morning a waited a little bit more, the app opened and now everything works like a charm!

Thanks again for you port!

ardera, 2020-11-20:
> Oh nice! You are building application for cars?

Exactly, actually I only developed flutter-pi so I can have a nice & modern UI in my car

> I think I was not patient enough.. In --release i have still no touch, but without the --release the app wouldn't open so I always shut it after a few seconds but this morning a waited a little bit more, the app opened and now everything works like a charm!
> 
> Thanks again for you port!

Nice! Yeah debug mode takes a bit longer to start, I think it's mostly spend on disk IO. So a faster SD card may help

It's still weird it doesn't work in release mode, it should work fine there too

vigno88, 2020-11-20:
>  Exactly, actually I only developed flutter-pi so I can have a nice & modern UI in my car

That must be so perfect, you can have any data you want from car at any time. Good luck with this project!

> Nice! Yeah debug mode takes a bit longer to start, I think it's mostly spend on disk IO. So a faster SD card may help

I note that down.

ardera, 2021-01-31:
> That must be so perfect, you can have any data you want from car at any time. Good luck with this project!

(a bit late but) thanks! Closing this now, since it seems to be resolved

closed 2021-01-31 by ardera