ardera/flutter-pi #124: Problem running plotter-pi in virtual environment

irisrucia97, 2020-11-20:
[flutter-pi] WARNING: display didn't provide valid physical dimensions.
The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display. 
WARNING: Error getting last vblank timestamp. drmCrtcGetSequence: Invalid argument
VSync will be disabled.
See https://github.com/ardera/flutter-pi/issues/38 for more info. 
=================================== 
display mode:
resolution: 800 X 600 
refresh rate: 60Hz
 physical Size: 0mm X 0mm
flutter device pixel ratio: 1.000000
 =================================== 
VMware: No 3D enabled (0, Success). 
[flutter-pi] Could not create GBM Surface. gbm_surface_create_with_modifiers: Function not implemented

The execution environment ran on ubuntu 18.04 and the virtual machine used is oracle VM virtualBox. 
Doesn't work with virtual machines? Or is the other part wrong?

ardera, 2020-11-20:
The cause is right there in the log:
```
VMware: No 3D enabled (0, Success).
```

flutter-pi needs 3D acceleration / OpenGL. Otherwise it won't work. I don't know how to enable 3D acceleration for VMware.

irisrucia97, 2020-11-23:
Thank you for answer!

closed 2020-11-23 by irisrucia97