ardera/flutter-pi #128: Document how to build for debug mode

Hixie, 2020-11-27:
The README.md file describes how to build for release but it's not clear exactly how to build a debug build.

ardera, 2020-11-27:
Yeah I agree.

It's pretty simple actually, the `flutter build bundle` command always builds the debug artifacts (except if given the `--precompiled` argument). So the debug artifacts (namely `kernel_blob.bin`) are contained in the asset bundle. If you don't want release mode, you can just skip the `Building the app.so` section and go straight to `Running your app with flutter-pi`

compenguy, 2021-01-29:
Relatedly, and I can move this into a separate issue if you'd like, is how to build/start in profiling mode.

I have a bundle that I built with --profile, but I get no observatory announcement when I start it with --release, and it won't start without it because of a missing kernel.blob.

When I use a bundle built for debug and start flutter-pi in debug I get the observatory announcement, but all the documentation for flutter says not to profile in debug.  So I'm just unclear on how to accomplish this.

ardera, 2021-01-30:
> I have a bundle that I built with --profile, but I get no observatory announcement when I start it with --release, and it won't start without it because of a missing kernel.blob.

Actually, the asset bundle (so the stuff inside `build/flutter_assets`) is the same whether you build in release, profile or debug mode. The default build mode is `--release`, so invoking `flutter build bundle` without args as is done in the README to build the debug flutter assets will actually build them in release mode. But it's not important since they're all the same anyway.

If you want profile mode you need a custom `gen_snapshot` & flutter engine, configured with `--runtime-mode profile` as an argument to `gn`. I'll consider including it engine-binaries branch. Main point against including it is that I'm already building 4 engines (debug & release for ARM & ARM64) and also I'm building those on my private machine, so I while it's building I can't do anything else.

compenguy, 2021-01-30:
I've made a go at cross compiling those myself, but failed miserably at every turn (llvm was a pain, but I got it built, but I never succeeded at getting libcxx to build, and I can't even remember any more what problems I had with libflutter_engine.so).  

When I get some free time, maybe I'll have another go at it.  If I run into problems, is there a good IRC or Discord channel for people who are using flutter in embedded?

ardera, 2021-01-30:
> When I get some free time, maybe I'll have another go at it. If I run into problems, is there a good IRC or Discord channel for people who are using flutter in embedded?

Sadly, there isn't. There's a flutter-official discord you can give a try, but it wasn't that helpful for me in the past

That's a good idea though. Maybe the flutter discord server mods could add a new channel `flutter-embedders`? Alternatively, I could create a flutter-embedded telegram group or a discord, but if it was on the official discord, we'd have everything in one place. Downside may be that that's confusing since everything else on the server is google-official. People may assume they will get google official support while actually, the people there are just hacking together some embedders. What do you think @Hixie?

> I've made a go at cross compiling those myself, but failed miserably at every turn (llvm was a pain, but I got it built, but I never succeeded at getting libcxx to build, and I can't even remember any more what problems I had with libflutter_engine.so).

Yeah I agree it's pretty hard. I can provide you with my build scripts tough: [flutter_engine.zip](https://github.com/ardera/flutter-pi/files/5898288/flutter_engine.zip)

Steps to get it running:
- In every script, there's an absolute path to the build system location. By default it's `/home/hannes/devel/flutter_engine`. Change that to wherever you extracted the zip but including the `flutter_engine` folder contained in the zipfile in the path.
- cd into the extracted directory
- `rm binutils_sources && git clone git://sourceware.org/git/binutils-gdb.git binutils_sources`
- `rm llvm_sources && git clone -b release/11.x https://github.com/llvm/llvm-project.git llvm_sources`
- `rm engine-binaries && git clone -b engine-binaries https://github.com/ardera/flutter-pi.git engine-binaries`
- edit the rsync invocations in `arm32/fetch_sysroot` to fit your pi's hostname
- install googles depot tools if you haven't already
- `sudo apt install python2`
- `sudo apt install python-is-python2` (if it exists for your distro)
- right now it's impossible to build flutter stable without modifications if you aren't a google employee (since it erronously depends on a google internal repo) I have a [fork of the engine](https://github.com/ardera/engine-1.git) that has these modifications. The latest stable branch is `flutter-1.22-candidate.12`
  - in the future, when it works again, you need to edit the `engine/.gclient` file so it points to the official repo again
  - for now, `pushd engine && gclient sync --ref=src/flutter@2328d8b91ee74c284fbffd126da3598313bdb05f && popd` (the latest commit of the `flutter-1.22-candidate.12` branch)
- `./arm32/fetch_sysroot` (will make modifications on your pi, more precisely change all absolute symlinks to relative ones)
- `./arm32/build_binutils`
- `./arm32/build_llvm`
- edit `arm32/build_engine_release`
  - replace `--runtime-mode release` with `--runtime-mode profile`
  - change any path with `linux_release_arm` to `linux_profile_arm`
- `./arm32/build_engine_release`
- engine binaries are now in `engine_out/out/linux_profile_arm/`

Once you have that setup, all you need to do on a new flutter version is the `gclient sync` and `arm32/build_engine_release`.

Everything I do on a new version is run `cbcd_engine` which checks out the latest stable engine, builds, collects the binaries into the engine-binaries dir and deploys them to my pi(s), but that doesn't work right now bc flutter builds are broken outside of google as mentioned.

When you have the binaries, copy the `libflutter_engine.so` to your pi to `/usr/lib/libflutter_engine.so.release`, build your `app.so` using the `gen_snapshot` you just build instead of the one I provide and it should just work.

Hixie, 2021-01-30:
We're happy to add new channels, see https://github.com/flutter/flutter/wiki/Chat for details. Basically, ask in #server-support.
That server isn't "Google"-official, by the way. It's "Flutter"-official. Google is a big contributor to Flutter, but not the only one. And nothing there is Google giving support to anyone.

ardera, 2021-01-30:
> Basically, ask in #server-support.

Okay, I'll do that.

> That server isn't "Google"-official, by the way. It's "Flutter"-official. Google is a big contributor to Flutter, but not the only one. And nothing there is Google giving support to anyone.

My mistake. I just assumed that bc most of the people there are working for google.

eximius313, 2023-09-23:
@ardera, isn't it already documented?

ardera, 2023-09-24:
> @ardera, isn't it already documented?

Yes it is, thanks for letting me know.

closed 2023-09-24 by ardera