ardera/flutter-pi #137: Native build

hhk7734, 2020-12-15:
I tryed to build on arm64. https://wiki.loliot.net/docs/lang/flutter/engine/flutter-app-for-linux-arm64

After `flutter build bundle`, I got
`./build/linux/debug/bundle/data/flutter_assets/kernel_blob.bin`
`./.dart_tool/flutter_build/a2391edd92a619ff72fea4c3443e4345/app.so`

on My testbed

```shell
$ flutter-pi sample/
[flutter-pi] Could not query DRM device list: No such file or directory
```

So, I failed to test the release mode. Are the above files related to the file in your release description?

hhk7734, 2020-12-15:
I just tryed `xinit /path/to/sample/build/linux/release/bundle/sample`. it works well. but not full screen.

I'll soon test it with flutter-pi and see if there is a speed difference. :)

ardera, 2020-12-15:
> After `flutter build bundle`, I got
> `./build/linux/debug/bundle/data/flutter_assets/kernel_blob.bin`
> `./.dart_tool/flutter_build/a2391edd92a619ff72fea4c3443e4345/app.so`

I'm a bit confused. The normal path for the flutter assets should be just `build/flutter_assets`. If you follow my instructions in the readme, the `app.so` should be located inside `build/flutter_assets` too. I also don't understand what you mean by

> Are the above files related to the file in your release description?

What release description? What file in my release description? I'm sorry, I'm probably missing something obvious here 😅 

> ```shell
> $ flutter-pi sample/
> [flutter-pi] Could not query DRM device list: No such file or directory
> ```
> 
> So, I failed to test the release mode.

That is unrelated to any of the flutter artifacts though, `flutter-pi` hasn't even reached the point in it's execution where the artifacts are loaded / the paths to them checked.

That error means the call to [`drmGetDevices2`](https://github.com/grate-driver/libdrm/blob/3e3c53ed85c30331a6c0ba2af349223654caa5dd/xf86drm.c#L4007) is failing. Don't know why that should happen though. Maybe the x server / wayland server was still running on your testbed? Though I'm not sure this could cause the error.

Besides, you need to invoke flutter-pi with the `--release` option to launch it in release mode.

hhk7734, 2020-12-15:
Oh, Sorry.

I build app on linux-arm64 board(ODROID-C4).

```shell
flutter create sample && cd sample
```

```shell
flutter build bundle
```

After this, I got
`./build/linux/debug/bundle/data/flutter_assets/kernel_blob.bin`
`./.dart_tool/flutter_build/a2391edd92a619ff72fea4c3443e4345/app.so`

In README.md, Running your App on the Raspberry Pi -> Building the app.so (for running your app in Release/Profile mode).
If I follow your description, maybe, `app.so and kernel_blob.bin` is in `/home/pi/my_app`. right?

So, I was wondering if the file I created could replace the file in the description.

> That error means the call to drmGetDevices2 is failing. Don't know why that should happen though. Maybe the x server / wayland server was still running on your testbed? Though I'm not sure this could cause the error.

On my testbed, perhaps, the kernel doesn't support DRM. So I wanted to test if the above files were the same, but I couldn't do it right away, so I asked if you knew.

---

Anyway, my ultimate goal is to build right in native and run it with flutter-pi.

ardera, 2020-12-15:
Oh okay, I understand.

> In README.md, Running your App on the Raspberry Pi -> Building the app.so (for running your app in Release/Profile mode).
> If I follow your description, maybe, `app.so and kernel_blob.bin` is in [...]

Yeah, I think you can safely replace them. `kernel_blob.bin` should be the same anyway, it doesn't contain any native code and since you also replaced `gen_snapshot`, the `app.so` should work too.

Your way of just replacing the x64 artifacts in the SDK with arm64 ones is a glorious hack. I never thought you could get the SDK to run on arm that easily.

mikekosulin, 2021-03-18:
so strange.
Got the same error before manual setup Fake KMS via raspi-config.

caught this because looks like my Ansible task doesn't work:
```yaml
vars:
  GLDRIVER: "G2"
tasks:
  - name: Set OpenGL desktop driver
    become: true
    shell: "raspi-config nonint do_gdriver {{ GLDRIVER}} "
```

Raspberry pi 3 model B+

```shell
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

hhk7734, 2021-04-07:
flutter master branch already supports it. :)

closed 2021-04-07 by hhk7734