ardera/flutter-pi #138: flutter_map issue

matzesoft, 2020-12-16:
I want to use [flutter_map](https://github.com/fleaflet/flutter_map) in my project, however I run into some problems. I already created a [issue](https://github.com/fleaflet/flutter_map/issues/791) at the flutter_map repo so _read this first._
Maybe the problem of the zoom resetting is related to the touch input of flutter-pi. Maybe someone can try the flutter_map example and share his experience?

ardera, 2020-12-16:
I just tested with the example, I'm somehow getting a different issue. Sometimes after zooming in/out, the map just disappears for me, leaving just a grey rectangle. I didn't run into the zoom reset issue, since the map disappears so early. Maybe what manifests for you as the zoom resetting manifests itself for me as the map greying out? Either way, weird issue

I can zoom in using double taps / double clicks just fine though.

Additionally, `flutter_gallery` has a widget for testing zoom. I tested there and it works fine

The Pi 7inch screens touch detection isn't that high resolution, so two very close touchpoints could get merged into one. Don't know how that could cause the issue though.

matzesoft, 2020-12-17:
Ok, so this is issue is probably created by `flutter_map` if `flutter_gallery` is working fine or could you imagine any other problems apart from the touch input which could be created by `flutter-pi`?

About my UI: Thanks! If I can help you with anything just [contact](mailto:matzesoft@gmail.com) me.

ardera, 2020-12-19:
> Ok, so this issue is probably created by `flutter_map` if `flutter_gallery` is working fine or could you imagine any other problems apart from the touch input which could be created by `flutter-pi`?

I wouldn't exclude that flutter-pi is creating the problem. Though I think it's likely the problem is on the flutter side. Somehow I now get errors with flutter_gallery too. (I have absolutely no idea why I didn't get them before)

The error & stacktrace is the following:
```
flutter: ══╡ EXCEPTION CAUGHT BY GESTURE ╞═══════════════════════════════════════════════════════════════════
flutter: The following assertion was thrown while handling a gesture:
flutter: 'package:flutter/src/widgets/interactive_viewer.dart': Failed assertion: line 644 pos 12: 'scale !=
flutter: 0.0': is not true.
flutter:
flutter: Either the assertion indicates an error in the framework itself, or we should provide substantially
flutter: more information in this error message to help you determine and fix the underlying cause.
flutter: In either case, please report this assertion by filing a bug on GitHub:
flutter:   https://github.com/flutter/flutter/issues/new?template=BUG.md
flutter:
flutter: When the exception was thrown, this was the stack:
flutter: #2      _InteractiveViewerState._matrixScale (package:flutter/src/widgets/interactive_viewer.dart:644:12)
flutter: #3      _InteractiveViewerState._onScaleUpdate (package:flutter/src/widgets/interactive_viewer.dart:779:43)
flutter: #4      ScaleGestureRecognizer._advanceStateMachine.<anonymous closure> (package:flutter/src/gestures/scale.dart:437:18)
flutter: #5      GestureRecognizer.invokeCallback (package:flutter/src/gestures/recognizer.dart:183:24)
flutter: #6      ScaleGestureRecognizer._advanceStateMachine (package:flutter/src/gestures/scale.dart:436:7)
flutter: #7      ScaleGestureRecognizer.handleEvent (package:flutter/src/gestures/scale.dart:332:7)
flutter: #8      PointerRouter._dispatch (package:flutter/src/gestures/pointer_router.dart:77:12)
flutter: #9      PointerRouter._dispatchEventToRoutes.<anonymous closure> (package:flutter/src/gestures/pointer_router.dart:122:9)
flutter: #10     _LinkedHashMapMixin.forEach (dart:collection-patch/compact_hash.dart:377:8)
flutter: #11     PointerRouter._dispatchEventToRoutes (package:flutter/src/gestures/pointer_router.dart:120:18)
flutter: #12     PointerRouter.route (package:flutter/src/gestures/pointer_router.dart:106:7)
flutter: #13     GestureBinding.handleEvent (package:flutter/src/gestures/binding.dart:358:19)
flutter: #14     GestureBinding.dispatchEvent (package:flutter/src/gestures/binding.dart:338:22)
flutter: #15     RendererBinding.dispatchEvent (package:flutter/src/rendering/binding.dart:267:11)
flutter: #16     GestureBinding._handlePointerEvent (package:flutter/src/gestures/binding.dart:295:7)
flutter: #17     GestureBinding._flushPointerEventQueue (package:flutter/src/gestures/binding.dart:240:7)
flutter: #18     GestureBinding._handlePointerDataPacket (package:flutter/src/gestures/binding.dart:213:7)
flutter: #22     _invoke1 (dart:ui/hooks.dart:265:10)
flutter: #23     _dispatchPointerDataPacket (dart:ui/hooks.dart:174:5)
flutter: (elided 5 frames from class _AssertionError and dart:async)
flutter:
flutter: Handler: "onUpdate"
flutter: Recognizer:
flutter:   ScaleGestureRecognizer#315d1
flutter: ════════════════════════════════════════════════════════════════════════════════════════════════════
```

So the `ScaleGestureRecognizer` invokes its listener with a scale of `0.0`. Maybe that's caused by the problem pi touch detection problem, where 2 close points are being coalesced into one?

I think it'd be cool to build an overlay for the zooming widget, the see the touch positions

ardera, 2020-12-19:
Here's a touch position overlay I built: https://gist.github.com/ardera/6b9787b7ae26bfb3e42459728dea4de3

I used it with flutter_gallery, and as predicted, the error message is logged at the exact time the two touch positions get merged into one when they are too close.

I also tested with flutter_map but I didn't notice any pattern there. However I noticed the weird jumping / zooming doesn't ocurr when your first lift one finger, and then the other. Though even then, it sometimes ocurrs too if you lift the first finger too soon after the actual zooming gesture.

matzesoft, 2020-12-29:
I tested now with your position overlay. In `flutter_map` with my own map data the map always zoomed completely out when the two touch positions merged.

> However I noticed the weird jumping / zooming doesn't ocurr when your first lift one finger, and then the other.

I also tested this but for me it didn't change anything. Sometimes the map still zoomed completely out.
Maybe this also depends on map size. I could imagine that testing with the small `flutter_map` example is hard because there isn't a lot of range to test with. I created a [small guide](https://github.com/matzesoft/kart-project/tree/master/assets/map) if you want test with my bigger map data.

I also wanna mention that I am not using the official RaspberryPi touch display ([my display](https://www.amazon.de/gp/product/B076J8ZWFF/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1)). But as I am getting quite similar results I don't think this makes a lot of difference here.

matzesoft, 2021-01-15:
So I finally switched off from the `flutter_map` package and I am using the [`map`](https://pub.dev/packages/map) now. This package doesn't have the capabilities `flutter_map` has but for my purposes thats enough.
Aswell, I didn't run into any zooming issues with the `map` package. But still thanks for the research.
Closing for now.

closed 2021-01-15 by matzesoft