ardera/flutter-pi #139: Can't run flutter-pi app after shutting down flutter-pi app

Hixie, 2020-12-18:
Killing `flutter-pi` prevents the app from working again, due to "permission denied" errors.

Steps to reproduce:

1. I reboot my Pi (a 3B+).
2. I ssh into it and run `flutter-pi --release .` in the relevant directory.
3. App runs fine.
4. I run `killall flutter-pi`.
5. I run the app again.
6. The app fails with 
```
[modesetting] Could not commit atomic request. drmModeAtomicCommit: Permission denied
[compositor] Could not present frame. drmModeAtomicCommit: Permission denied
```

Admittedly this doesn't seem like a great way to stop the app, but I'm not sure what the right way to stop it so I can restart it is. :-)

Hixie, 2020-12-18:
(This sequence used to work a few weeks ago.)
(It seems if I kill the app that flutter-pi starts, nothing happens, but if I then kill flutter-pi, the problem described above doesn't.)

ardera, 2020-12-20:
There can only be one process (the DRM master) at a time on the whole system that is allowed to control the video output. So either the first process somehow still exists and is still DRM master on the 2nd invocation of flutter-pi, or the kernel somehow thinks there's still a DRM master even after the first process has finished.

> * I reboot my Pi (a 3B+).
> * I ssh into it and run `flutter-pi --release .` in the relevant directory.
> * App runs fine.
> * I run `killall flutter-pi`.
> * I run the app again.
> * The app fails with

Do you call these commands in a script? POSIX signals have no concept of "completion", so `killall`, by default, doesn't wait for the process to finish, it just sends a `SIGTERM`. There's a `--wait` flag for `killall`, can you try if that works? Maybe that helps

Hixie, 2020-12-20:
Waiting doesn't seem to help. It seems that killing the process started by `flutter-pi` is the key; killing `flutter-pi` itself no longer (?) seems to kill that.

ardera, 2020-12-20:
> It seems that killing the process started by `flutter-pi` is the key; killing `flutter-pi` itself no longer (?) seems to kill that.

Isn't that the same thing? If you kill `flutter-pi` itself using `killall`, what way do you use to kill the process?

Hixie, 2020-12-21:
My process structure looks like this:
```
  │   ├─bash
  │   │   └─bash
  │   │       └─flutter-pi --release .
  │   │           ├─laundry-room-mo
  │   │           ├─{flutter-pi}
  │   │           ├─{flutter-pi}
  │   │           ├─{flutter-pi}
  │   │           ├─{flutter-pi}
  │   │           ├─{flutter-pi}
  │   │           ├─{flutter-pi}
  │   │           ├─{io.flutter.io}
  │   │           ├─{io.flutter.ui}
  │   │           └─{io.flutter.ui}
```
If I kill `laundry-room-mo` first, then kill `flutter-pi`, then rerun `flutter-pi`, I don't get the error described above.

ardera, 2020-12-21:
For me, the process tree looks like this when launching `flutter_gallery` in release mode:

```
  ├─sshd -D
  │   ├─sshd
  │   │   └─sshd
  │   │       └─bash
  │   │           └─pstree -aclt
  │   └─sshd
  │       └─sshd
  │           └─bash
  │               └─flutter-pi-debu --release .
  │                   ├─{flutter-pi-debu}
  │                   ├─{flutter-pi-debu}
  │                   ├─{flutter-pi-debu}
  │                   ├─{flutter-pi-debu}
  │                   ├─{flutter-pi-debu}
  │                   ├─{flutter-pi-debu}
  │                   ├─{io.flutter.io}
  │                   └─{io.flutter.ui}
```

So for me, no child process is created, there's only one process and that is `flutter-pi`. Do you have any idea what that child process `laundry-room-mo` is in your case? If you don't know, you can maybe set a breakpoint on `fork` & `execve` (and all the different versions of them) in `gdb` and see what creates it.

Hixie, 2021-02-09:
I'm not able to reproduce this. I have no idea what was going on.

ardera, 2021-02-23:
Weird, okay, closing this now

closed 2021-02-23 by ardera

krishna123sah, 2021-08-03:
I'm getting the above errors
`[modesetting] Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
[flutter-pi] Could not create drmdev from device at "/dev/dri/card1". Continuing.
[flutter-pi] WARNING: display didn't provide valid physical dimensions.
             The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
===================================
display mode:
  resolution: 1920 x 1080
  refresh rate: 50Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 3.1 Mesa 19.3.2"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_primitive_bounding_box GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_EXT_texture_norm16 GL_KHR_context_flush_control GL_NV_image_formats GL_OES_draw_elements_base_vertex GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_EXT_buffer_storage GL_EXT_float_blend GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_shader_image_atomic GL_EXT_texture_compression_s3tc_srgb GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_MESA_framebuffer_flip_y GL_EXT_texture_query_lod "
===================================
flutter: Observatory listening on http://127.0.0.1:35911/0f253WQN62E=/
flutter: Directory: '/home/pi'
[modesetting] Could not commit atomic request. drmModeAtomicCommit: Permission denied
[compositor] Could not present frame. drmModeAtomicCommit: Permission denied
[modesetting] Could not commit atomic request. drmModeAtomicCommit: Permission denied
[compositor] Could not present frame. drmModeAtomicCommit: Permission denied
[modesetting] Could not commit atomic request. drmModeAtomicCommit: Permission denied
[compositor] Could not present frame. drmModeAtomicCommit: Permission denied
`

johannphilippe, 2021-08-06:
> I'm getting the above errors
> 
> ```
> [flutter-pi] Could not create drmdev from device at "/dev/dri/card1". Continuing.
> [flutter-pi] WARNING: display didn't provide valid physical dimensions.
>              The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
> ===================================
> display mode:
>   resolution: 1920 x 1080
>   refresh rate: 50Hz
>   physical size: 0mm x 0mm
>   flutter device pixel ratio: 1.000000
> ===================================
> EGL information:
>   version: 1.4
>   vendor: "Mesa Project"
>   client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
>   display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
> ===================================
> OpenGL ES information:
>   version: "OpenGL ES 3.1 Mesa 19.3.2"
>   shading language version: "OpenGL ES GLSL ES 3.10"
>   vendor: "Broadcom"
>   renderer: "V3D 4.2"
>   extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_primitive_bounding_box GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_EXT_texture_norm16 GL_KHR_context_flush_control GL_NV_image_formats GL_OES_draw_elements_base_vertex GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_EXT_buffer_storage GL_EXT_float_blend GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_shader_image_atomic GL_EXT_texture_compression_s3tc_srgb GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_MESA_framebuffer_flip_y GL_EXT_texture_query_lod "
> ===================================
> flutter: Observatory listening on http://127.0.0.1:35911/0f253WQN62E=/
> flutter: Directory: '/home/pi'
> [modesetting] Could not commit atomic request. drmModeAtomicCommit: Permission denied
> [compositor] Could not present frame. drmModeAtomicCommit: Permission denied
> [modesetting] Could not commit atomic request. drmModeAtomicCommit: Permission denied
> [compositor] Could not present frame. drmModeAtomicCommit: Permission denied
> [modesetting] Could not commit atomic request. drmModeAtomicCommit: Permission denied
> [compositor] Could not present frame. drmModeAtomicCommit: Permission denied
> ```

Same issue here, on Raspberry pi 3, Raspberry pi OS 32 bits.

ysyfff, 2022-04-08:
> There can only be one process (the DRM master) at a time on the whole system that is allowed to control the video output. So either the first process somehow still exists and is still DRM master on the 2nd invocation of flutter-pi, or the kernel somehow thinks there's still a DRM master even after the first process has finished.
> 
> > * I reboot my Pi (a 3B+).
> > * I ssh into it and run `flutter-pi --release .` in the relevant directory.
> > * App runs fine.
> > * I run `killall flutter-pi`.
> > * I run the app again.
> > * The app fails with
> 
> Do you call these commands in a script? POSIX signals have no concept of "completion", so `killall`, by default, doesn't wait for the process to terminate, it just sends a `SIGTERM` and immediately returns. There's a `--wait` flag for `killall`, can you try if that works? Maybe that helps

**There can only be one process** reminds me, so I reboot my pi and rerun `flutter-pi <flutter_assets_b>` and surprisedly it works!!! 

My case is:

- run `flutter-pi <flutter_assets_a>`
- <flutter_assets_a> failed somehow, which <flutter_assets_a> is built on windows. I update flutter from 2.5.3 to 2.10.4 on my Mac, and rebuild a bundle and then scp <flutter_assets_b> to my pi
- run `flutter-pi <flutter_assets_b>`
- then the problem show up

officialluq, 2024-08-29:
> I'm getting the above errors
>
> ```
> [flutter-pi] Could not create drmdev from device at "/dev/dri/card1". Continuing.
> [flutter-pi] WARNING: display didn't provide valid physical [...]

I was able to reproduce this issue and fix is quite simple. Do you use some kind of compositor for wayland or X11? Try to disable it first :) DRI cannot be used by two compositors