ardera/flutter-pi #144: undefined reference to 'libinput_device_touch_get_touch_count'

biagiopietro, 2021-01-19:
Hello, I followed [GETTING_STARTED.md](https://github.com/ardera/flutter-pi/blob/master/GETTING_STARTED.md) but I got stuck when I carried out **make -j\`nproc\`**.

# Command history
Execution of `cmake ..`, output:

```bash
puma@puma:~/dev/flutter-pi/build(master)$ sudo cmake ..
-- CMAKE_BUILD_TYPE not set, defaulting to Release.
-- The C compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Generator .............. Unix Makefiles
-- Build Type ............. Release
-- Flutter Channel not set, defaulting to stable
-- Flutter Channel ........ stable
-- Engine SHA1 ............ ae90085a8437c0ae94d6b5ad2741739ebc742cb4
-- PKG_CONFIG_PATH ........ 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libdrm'
--   Found libdrm, version 2.4.101
-- Checking for module 'gbm'
--   Found gbm, version 20.0.8
-- Checking for module 'egl'
--   Found egl, version 20.0.8
-- Checking for module 'glesv2'
--   Found glesv2, version 20.0.8
-- Checking for module 'libsystemd'
--   Found libsystemd, version 237
-- Checking for module 'libinput'
--   Found libinput, version 1.10.4
-- Checking for module 'xkbcommon'
--   Found xkbcommon, version 0.8.2
-- Checking for module 'libudev'
--   Found libudev, version 237
-- Configuring done
-- Generating done
-- Build files have been written to: /home/puma/dev/flutter-pi/build
```
Execution of   **sudo make -j\`nproc\`**, partial output:
```bash
[100%] Linking C executable flutter-pi
CMakeFiles/flutter-pi.dir/src/flutter-pi.c.o: In function `on_libinput_ready':
flutter-pi.c:(.text+0x1b2f): undefined reference to `libinput_device_touch_get_touch_count'
collect2: error: ld returned 1 exit status
CMakeFiles/flutter-pi.dir/build.make:282: recipe for target 'flutter-pi' failed
make[2]: *** [flutter-pi] Error 1
CMakeFiles/Makefile2:94: recipe for target 'CMakeFiles/flutter-pi.dir/all' failed
make[1]: *** [CMakeFiles/flutter-pi.dir/all] Error 2
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2
```
I got a lot of warning related to `%lld` as well, for instance:
```bash
/home/puma/dev/flutter-pi/src/plugins/omxplayer_video_player.c: In function ‘on_create’:
/home/puma/dev/flutter-pi/src/plugins/omxplayer_video_player.c:1111:56: warning: format ‘%lld’ expects argument of type ‘long long int’, but argument 4 has type ‘int64_t {aka long int}’ [-Wformat=]
         "flutter.io/omxplayerVideoPlayer/videoEvents%lld",
                                                     ~~~^
                                                     %ld
         player->player_id
         ~~~~~~~~~~~~~~~~~    
```
Inside the warnings list, I found also this:

```bash
[ 35%] Building C object CMakeFiles/flutter-pi.dir/src/compositor.c.o
/home/puma/dev/flutter-pi/src/flutter-pi.c: In function ‘on_libinput_ready’:
/home/puma/dev/flutter-pi/src/flutter-pi.c:1942:23: warning: implicit declaration of function ‘libinput_device_touch_get_touch_count’; did you mean ‘libinput_event_get_touch_event’? [-Wimplicit-function-declaration]
     int touch_count = libinput_device_touch_get_touch_count(device);
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                       libinput_event_get_touch_event

```

# System setup

## OS
```bash
puma@puma:~/dev/flutter-pi/build(master)$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.5 LTS
Release:	18.04
Codename:	bionic
```
## cmake
```bash
puma@puma:~/dev/flutter-pi/build(master)$ cmake --version
cmake version 3.19.3

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```
## make
```bash
puma@puma:~/dev/flutter-pi/build(master)$ make --version
GNU Make 4.1
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law
```
I hope I have provided you with all the information you need.

Thanks in advance for your support.

ardera, 2021-01-20:
The `%lld` errors are probably due to x86, I never tested that before. Should be an easy fix though (using `inttypes.h`)

The `libinput_device_touch_get_touch_count` is a function introduced in newer versions of libinput. Raspbian for example uses libinput 1.12.6, your system has 1.10.4. For now, it's best to work around that by [compiling the latest version of libinput yourself](https://wayland.freedesktop.org/libinput/doc/latest/building.html). In the future, I can work around that and avoid the call to `libinput_device_touch_get_touch_count` since I don't strictly need it, but it does make things easier a bit which is why I used it initially.

ardera, 2021-01-21:
First part of this issue should be resolved by #145