ardera/flutter-pi #149: Apps crashing due to a cursor

jianxizg, 2021-01-29:
I have just installed flutter-pi and tried to run an app in debug mode. The app started fine, with a cursor appearing on the screen. A few seconds later, it has crashed with:      
```
[compositor] Preferred framebuffer depth for hardware cursor is not supported by flutter-pi.
Segmentation fault 
```
P.S. a cursor is not needed    

Thanks!

ardera, 2021-01-29:
That hardware cursor message is not the reason for the crash.

Can you provide a coredump so I can look into it? (Just execute `ulimit -c unlimited` before, then run flutter-pi and wait till it crashes, then send me the file named `core`)

jianxizg, 2021-01-29:
Here is the core file    

https://wetransfer.com/downloads/c26b08880c69b07fa267d000fb8bd1c120210129160852/ea4ad564c6f249265b1f35ad9c6be17920210129160913/879da7

It's a pi4 4GB, official 7' touch screen, with 32 bit raspbian lite, new installation, not modified in any way.   

Here is the full output:    
```
[flutter-pi] WARNING: display has non-square pixels. Non-square-pixels are not supported by flutter.
===================================
display mode:
  resolution: 800 x 480
  refresh rate: 60Hz
  physical size: 155mm x 86mm
  flutter device pixel ratio: 1.358234
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 3.1 Mesa 19.3.2"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_primitive_bounding_box GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_EXT_texture_norm16 GL_KHR_context_flush_control GL_NV_image_formats GL_OES_draw_elements_base_vertex GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_EXT_buffer_storage GL_EXT_float_blend GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_shader_image_atomic GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_MESA_framebuffer_flip_y GL_EXT_texture_query_lod "
===================================
flutter: Observatory listening on http://127.0.0.1:35145/a-lpegpqDNM=/
[compositor] Preferred framebuffer depth for hardware cursor is not supported by flutter-pi.
Segmentation fault (core dumped)
```

ardera, 2021-01-29:
Somehow gdb can't make sense of the core file.

Can you try building flutter-pi in debug mode?
- Go into your build dir
- run `cmake .. -DCMAKE_BUILD_TYPE=Debug`
- `ninja`
- `ulimit -c unlimited`
- run your app using flutter-pi again
- upload both the core file and the flutter-pi executable

Btw what app are you using?

jianxizg, 2021-01-29:
Building in debug mode - success.    
Somehow ninja runs in the terminal and I cannot type in more commands unless I stop ninja.    
Please advise    

Funny enough, I am running the official Hello World example to test it out

ardera, 2021-01-29:
> Somehow ninja runs in the terminal and I cannot type in more commands unless I stop ninja.

That's definitely not normal. Try deleting your `flutter-pi` folder and then following the instructions again to clone, configure and build it. Also try reinstalling the engine binaries.

It's weird bc I have pretty much the same setup (except Pi 4 2GB instead of 4GB) and I justed tested and everythings fine for me.

jianxizg, 2021-01-31:
Pulled out a 2gb pi4, it doesn't crash as often now, but if I move the cursor around, it will crash soon, is there a way to completely disable the cursor?. Very strange LOL

ardera, 2021-01-31:
Yeah that's strange

> is there a way to completely disable the cursor?

I mean, you _could_ edit the sources and disable the cursor, but that doesn't fix the problem, just works around it. I want to fix the bug, but since I can't reproduce it on my Pi's I have to rely on a core dump to look into it

Btw what input devices do you use? Can you attach the output of `cat /proc/bus/input/devices`?

DariusBotha95, 2021-11-10:
Hi there Hannes,

I am running into the same problem, whenever I add the dimensions of the screen size to the command `flutter-pi -d "306,230" my_apps_flutter_assets`, my display looks much better and the scaling is perfect, however, the moment I move the mouse connected to the Pi, it closes the screen and gives me a `Segmentation fault` error

ardera, 2021-11-13:
> Hi there Hannes,
> 
> I am running into the same problem, whenever I add the dimensions of the screen size to the command `flutter-pi -d "306,230" my_apps_flutter_assets`, my display looks much better and the scaling is perfect, however, the moment I move the mouse connected to the Pi, it closes the screen and gives me a `Segmentation fault` error
> 
> In line with your previous comment, my output of `cat /proc/bus/input/devices` is attached. [cat.txt](https://github.com/ardera/flutter-pi/files/7510527/cat.txt)

Hey,
thanks, but what would help me much more is actually a coredump of flutter-pi:

> * Go into your build dir
> * run `cmake .. -DCMAKE_BUILD_TYPE=Debug`
> * `ninja`
> * `ulimit -c unlimited`
> * run your app using flutter-pi again
> * upload both the core file and the flutter-pi executable