ardera/flutter-pi #152: Flutter-pi command not found!

Sarthak-22, 2021-02-09:
I am new to both flutter and Raspberry Pi 
Whenever I try to run "flutter-pi  /home/pi/flutter_gallery" , it gives an error saying that "flutter-pi command not found".
Can anyone tell me how and where (which directory) should the flutter-pi command be used so that it starts to run my app.

DisDis, 2021-02-09:
Hi, please use './flutter-pi'.

Sarthak-22, 2021-02-09:
Is there any specific directory in which we should open the command line and run this command?

Sarthak-22, 2021-02-09:
Also, when I use" ./flutter-pi /home/pi/flutter_gallery", it's now throwing an error "No such file or directory", though I have correctly installed (cloned) flutter-pi repo in engine-binaries directory in Rpi

ardera, 2021-02-10:
`./flutter-pi ...` needs to be executed in the directory where the `flutter-pi` executable is located.

For example, if you've followed the instructions, you've cloned `flutter-pi` somewhere and created a build folder inside it. The `flutter-pi` executable is located inside that `build` folder if you've built it successfully.

If you've cloned flutter-pi to `/home/pi/flutter-pi`, you can launch flutter-pi by executing
```
/home/pi/flutter-pi/build/flutter-pi ...
```
or
```
cd /home/pi/flutter-pi/build
./flutter-pi ...
```

Sarthak-22, 2021-02-10:
@ardera Thank you for your help. I re-executed all the commands and finally **flutter-pi** executable was installed successfully. However, on running the command, `flutter-pi  /home/pi/myapp` , the console showed the following output along with 4 errors at the end
```
[flutter-pi] WARNING: display didn't provide valid physical dimensions.
             The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
===================================
display mode:
  resolution: 1024 x 600
  refresh rate: 60Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 2.0 Mesa 19.3.2"
  shading language version: "OpenGL ES GLSL ES 1.0.16"
  vendor: "Broadcom"
  renderer: "VC4 V3D 2.1"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_EGL_image GL_OES_depth_texture GL_AMD_performance_monitor GL_OES_packed_depth_stencil GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_separate_shader_objects GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_KHR_parallel_shader_compile GL_MESA_tile_raster_order "
===================================
detected VideoCore IV as underlying graphics chip, and VC4 as the driver.
Reporting modified GL_EXTENSIONS string that doesn't contain non-working extensions.
[ERROR:engine/src/flutter/shell/common/shell.cc(210)] Dart Error: Can't load Kernel binary: Invalid SDK hash.
[ERROR:engine/src/flutter/shell/common/engine.cc(220)] Could not prepare to run the isolate.
[ERROR:engine/src/flutter/shell/common/engine.cc(159)] Engine not prepare and launch isolate.
[ERROR:engine/src/flutter/shell/common/shell.cc(491)] Could not launch engine with configuration.
flutter: Observatory listening on http://127.0.0.1:39309/UsVbxXFWFkw=/
```
If possible, can u indicate the source of these errors and how they can be rectified?
I used `flutter build appbundle` to build my app on host machine (Windows cmd) and used the `scp` command to transfer them in a directory in Rpi

GeneticSYS, 2021-02-10:
Hey @Sarthak-22

The error `Dart Error: Can't load Kernel binary: Invalid SDK hash.` means, that your installed engine-binaries don't match the Flutter version of your built app. In your case, the current engine-binaries of flutter-pi are using Flutter 1.22.6 and you built your app with Flutter 1.22.0.

In order to fix that, you have to build your Flutter app with Flutter 1.22.6.

Sarthak-22, 2021-02-10:
Thank you so much, my touchscreen is working fine now..all the issues resolved
Closing the issue

closed 2021-02-10 by Sarthak-22