ardera/flutter-pi #153: Virtual-Keyboard support in flutter-pi

Sarthak-22, 2021-02-16:
Is there any way in which the virtual keyboard automatically comes up in Flutter without explicitly coding for virtual keyboard package using Dart.

I installed virtual keyboard in Raspbian OS with the command `sudo apt install matchbox-keyboard`. However, after running `flutter-pi` command and launching the flutter app, the matchbox-keyboard does not detect the input field/box in flutter app to automatically come up. Is there any solution for this (without coding for virtual-keyboard flutter package) ?

ardera, 2021-02-17:
Are you sure matchbox-keyboard would work with flutter-pi if the automatic input detection worked? matchbox-keyboard actually shouldn't work with flutter-pi at all since it depends on X11, while flutter-pi is completely incompatible with X11.

I've thought about touchscreen keyboard input for some time, and while it's definitely something that needs to be implemented, currently there's no "nice" way to do it. I can maybe add a kind of loopback interface that could be used by the virtual keyboard package so you can use its text input just like you'd use real text input on any other platform. However it seems like the owner doesn't maintain it anymore, so I'm not sure that could work.

mortenboye, 2021-04-14:
I had the same issue and ended up forking https://github.com/Im-Kevin/cool_ui which intercepts the platform messages that would trigger a native keyboard. 
Its not a perfect solution though, as you can run into some conflicts between the virtual keyboard and various overlays, alerts, etc.

mortenboye, 2021-08-30:
@Taha-Firoz I mostly removed stuff I didn't need, as I was only interested in the keyboard.
To get it to work I basically just followed the example setup, register the keyboard, insert a KeyboardRootWidget high in the widget hierarchy and specify the wanted keyboard types on your textfields.
If you need new keyboard types you have too add them and their identifiers to the package.

What issues are you seeing?

mortenboye, 2023-03-12:
I've tried a new approach (compared to the one described above), using the new TextInputControl features:
https://github.com/flutter/flutter/pull/76072
https://api.flutter.dev/flutter/services/TextInputControl-mixin.html

These APIs makes it much easier to implement a virtual keyboard.

And while the onscreen keyboard works as intended, plugging in a HW keyboard and typing on that, does not result in any characters when running on the Pi (Yocto). It does work in Mac and Linux standalone Flutter apps.

On the Pi I get this in the terminal:
```
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(illegalargument, Expected `arg[1]['inputType']['name']` to be a string-ification of `TextInputType`., null, null)
#0      JSONMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:171:7)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:315:18)
<asynchronous suspension>
```

@ardera do you have any clue what causes this? My HW keyboard works fine in the shell.