ardera/flutter-pi #154: Run flutter-pi app after boot

matzesoft, 2021-02-17:
I want to run my flutter-pi app right after the RaspberryPi has booted up.
At first, I created a script, which runs my flutter-pi project. When running the script just from the command line it worked out completely fine. The goal of the script is it, to restart the flutter-pi app if anything has crashed. After 10 fails the Raspberry Pi will be shutdown.
```bash
#!/bin/bash

# Counter how often flutter-pi failed to run.
counter=0
while [ $counter -lt 10 ]
do
    sudo ~/sdks/flutter-pi/out/flutter-pi ~/projects/my_project | tee ~/logs/log.txt-$(date +"%Y-%m-%d-%T")
    echo "MyProject will be restarted in 3 seconds..."
    sleep 3s
    counter=`expr $counter + 1`
done

echo "Unable to start MyProject correctly. The Raspberry will be shutdown in 5 seconds..."
sleep 5s
sudo poweroff
```

#### Run the script on boot
I tried to use `crontab` and `systemd` (following [this](https://www.baeldung.com/linux/run-script-on-startup) tutorial). Both of them resulted in the same problem. After rebooting the Raspberry Pi at first nothing happend. After about 35 seconds the Raspberry Pi just shutdown.
Following of the script from above, flutter-pi failed to run and finally just shutdown.

So how can I run flutter-pi after boot? Thanks for the help!

DisDis, 2021-02-17:
Hi, it is easy:)
Autorun script './run.sh' in home dir:
`nano ~/run.sh
`
```
#!/bin/sh
cd ~/workspace/MyProject
./flutter-pi ./
```

Init autorun in system
`sudo nano /etc/profile`
added first line:
`/home/pi/run.sh`

DisDis, 2021-02-17:
And yes, you have to set 'Auto login in console' in Raspi-config
See https://www.raspberrypi-spy.co.uk/wp-content/uploads/2015/09/raspi_config_boot_02.png

matzesoft, 2021-02-17:
Thanks, works good!
But if I am getting it correct this will run the app always when opening a new terminal or SSH session. Is there a way to start the app only one time at boot?
I also wondered if there is an option to quit to the CMD out of the flutter-pi app (for example when pressing a button).

So my goal would be to start the flutter-pi app on boot, but being still able to use the normal shell with SSH or when working locally on the Raspberry Pi.

w3p706, 2021-02-23:
I used systemd on ubuntu for that. SSH is still possible, login on device not. Also tty2… is not accessible.

```
sudo vi /etc/systemd/system/flutter-gui.service
```

```
[Unit]
starts the gui

[Service]
Type=Simple
ExecStart=flutter-pi --release /home/hef/flutter_assets

[Install]
WantedBy=multi-user.target
```

```
sudo systemctl enable flutter-gui
sudo systemctl start flutter-gui
```

ardera, 2021-02-23:
@w3p706 's method should work perfectly I think

> I also wondered if there is an option to quit to the CMD out of the flutter-pi app (for example when pressing a button).

You can't switch to terminal without terminating flutter-pi, but if you meant terminating flutter-pi without pressing `ctrl+c` you can also make it terminate by sending it a SIGTERM like this: `pkill flutter-pi`

matzesoft, 2021-02-24:
Thanks, finally got it working using `systemd`!
My mistake was quite simple. I always used `~` instead of the full path `/home/pi/`. After replacing all `~` to the complete home path everything worked fine...

> you can also make it terminate by sending it a SIGTERM like this: `pkill flutter-pi`

Yeah I meant to exit flutter-pi without having to use a keyboard, just with a software button. Works, thanks for that.

> You can also look at `journalctl -u myservice.service` and `systemctl status myservice.service` to find out how/if your service is failing

Thank you for that aswell. Helped me getting on the right way.

closed 2021-02-24 by matzesoft

kekko7072, 2021-04-14:
> I used systemd on ubuntu for that. SSH is still possible, login on device not. Also tty2… is not accessible.
>
> ```
> sudo vi /etc/systemd/system/flutter-gui.service
> ```
>
> ```
> [Unit]
> starts the gui [...]

Hey how can I add an update function? I mean the user from the app will download the new directory with the update and then how can I delete the old and start the new?

this is the c code I wrote but it doesn't work.
```

`------
SETUP THIS PROGRAMM AT STARTUP
    sudo nano /etc/rc.local

    echo Running at boot
    sudo  /home/pi/start-os
------
*/

#include <unistd.h>
#include <stdio.h>
#include <stdlib.h>

int main(){
    
    /* CHECK IF UPDATED */
    if (access("/home/pi/update", F_OK) == 0) {
       
        /*RENAME OLD OS*/
        rename("/home/pi/os", "/home/pi/old");

        /*RENAME NEW OS*/
        rename("/home/pi/update", "/home/pi/os");

        /*DELETE OLD OS*/
        system("rm -r /home/pi/old");
    }
    

    system("flutter-pi --release ~/home/pi/os");
    return 0;
}` 
```

kekko7072, 2021-04-14:
Hey I am trying to do the same using an executable file from c. In this code I also check if there is an update which means there is a new directory and rename it to run it. But my code is not working I edited the  `sudo nano /etc/rc.local` and added this line at the end 
```
echo Running at boot
sudo  /home/pi/start-os
```

Can someone fix it, or suggest me another way?
```
`
#include <unistd.h>
#include <stdio.h>
#include <stdlib.h>

int main(){
    
    /* CHECK IF UPDATED */
    if (access("/home/pi/update", F_OK) == 0) {
       
        /*RENAME OLD OS*/
        rename("/home/pi/os", "/home/pi/old");

        /*RENAME NEW OS*/
        rename("/home/pi/update", "/home/pi/os");

        /*DELETE OLD OS*/
        system("rm -r /home/pi/old");
    }
    

    system("flutter-pi --release /home/pi/os");
    return 0;
}`
```

kekko7072, 2021-04-14:
> sudo systemctl start flutter-gui

I have used this type and finally enabled adapting  this code to my case. But I don't know how to implement the update check.
```
[Service]
Type=Simple
ExecStart=flutter-pi --release /home/pi/os

[Install]
WantedBy=multi-user.target
```