ardera/flutter-pi #158: Flutter 2.0.0, channel stable, is out.

pezi, 2021-03-04:
Flutter 2.0.0, channel stable, is out and the actual engine runs into Dart Error: Can't load Kernel binary: Invalid kernel binary format version. This is as is - this issie is more an info.

pezi, 2021-03-04:
I see you started the migration job - https://github.com/flutter/flutter/issues/74322
Thanks for yout great work!

closed 2021-03-04 by pezi

ardera, 2021-03-04:
Yeah exactly, I tried to build the engine yesterday. Arm64 engine works fine, arm32 doesnt bc of the issue you mentioned. I'll see if I can fix it myself. If it takes too long, I'll just push the arm64 binaries without the arm32 ones.

GabrielAraujo, 2021-05-29:
Hello Guys, I'm trying to build the gallery app into my RPi 3 which is 32bit and facing this same issue.
The only solution is to wait for the guys to solve the issue [flutter/flutter#74322](https://github.com/flutter/flutter/issues/74322) and then push arm32 binaries too right?

@ardera thank you for this awesome project, I'm planning on building an bike computer with sensors and ANT+ connection. =D