ardera/flutter-pi #161: Could not prepare to run the isolate

manishbuttan, 2021-03-18:
Hello,
I upgraded Flutter to the latest stable release. This was 2.0.2. Then downloaded the flutter_gallery master project and ran flutter build bundle (the specific release you mentioned won't run). After building and rsyning to RPi, I am not able to run it on RPi. (I had successfully built and run a previous version on RPi).

The error I am getting is: Could not prepare to run the isolate. Engine not prepare and run the isolate. could not launch engine with configuration.

Please help. Thanks.

manishbuttan, 2021-03-18:
Also, typically, it would be great to run the latest version of at least the stable channel of flutter on flutter-pi. Will this be possible? What do we need to do to upgrade from a previous version?

manishbuttan, 2021-03-18:
So Adera, I managed to build a new app with stable build flutter 2.0.2 and ran it successfully on the RPi 4. So I am good for now. Thank you.
(Still wonder why the flutter_gallery did not run with the same procedure.)

ardera, 2021-03-18:
`flutter_gallery` master doesn't even build for me, but that specific commit in the readme works fine for me at least. Nice that you got it working with your own app though.

manishbuttan, 2021-03-18:
Is there any relation between flutter version and flutter-pi version? Will the latest flutter version automatically work with flutter pi? or will flutter-pi also need to be upgraded?

ardera, 2021-03-18:
When a new flutter version is released, flutter-pi should be working with it without any changes, I can't remember an update where that was not the case. However, it's not guaranteed. The embedder API provided by flutter is not stable and can theoretically change at any time (at least that was the case the last time I looked)

manishbuttan, 2021-03-18:
That’s good to know... Thanks.

manishbuttan, 2021-03-18:
Btw, your efforts are very valuable and greatly appreciated. I have worked with Qt on RPI for over a year and have been excited about flutter creating a truly free option for devices. Your efforts have made that possible. Keep up the good work. It matters a lot.

ardera, 2021-03-18:
> Btw, your efforts are very valuable and greatly appreciated. I have worked with Qt on RPI for over a year and have been excited about flutter creating a truly free option for devices. Your efforts have made that possible. Keep up the good work. It matters a lot.

Thanks, appreciate it!

manishbuttan, 2021-03-20:
Hi Hannes,
Would you be able to help with the following errors:

[ERROR:engine/src/flutter/runtime/dart_isolate.cc(169)] Could not prepare isolate.
[ERROR:engine/src/flutter/runtime/runtime_controller.cc(401)] Could not create root isolate.
[ERROR:engine/src/flutter/shell/common/shell.cc(571)] Could not launch engine with configuration.

When I switched my flutter channel to Beta and upgraded the project, it again stopped working on flutter-pi.

Thanks.

ardera, 2021-03-20:
For any version of the flutter SDK, the exact matching engine versions need to be installed on the Raspberry Pi. So if you use the stable 2.0.2 SDK, you need to have the binaries for 2.0.2 on the Raspberry Pi.

So you just need the correct engine binaries for the current beta version. I only provide ones for stable though. So you'll probably need to compile them yourself

I don't provide them myself because they're rarely useful and will increase the time I spend compiling engine stuff significantly, can't use it for anything else while I do that

manishbuttan, 2021-03-20:
Where would I get the correct engine source code? Currently the latest flutter stable version is 2.0.3. Where would I find the download for 2.0.3? 
Would apprecaite it if you can help with the steps to find, download and build binaries of any version I want, so I don't have to keep bothering you.
Thanks

manishbuttan, 2021-03-20:
I also have another question for you. What build is working on the RPi? I don't think it is the Android build. Will it be the Linux version? Where would I set the launch screen for my app? Sometimes some features of Flutter only work on certain platforms. Would the RPi version be compared to a linux desktop build?

ardera, 2021-03-20:
> Where would I get the correct engine source code? Currently the latest flutter stable version is 2.0.3. Where would I find the download for 2.0.3?

For any version of the flutter SDK, the exact commit hash of the engine to use is in the `bin/internal/engine.version` file. For example, for latest stable its [this](https://github.com/flutter/flutter/blob/stable/bin/internal/engine.version).

That commit hash is the specific commit in the [engine repo](https://github.com/flutter/engine) you should use to build the engine.

There's a lot of information about engine building in [this issue](https://github.com/ardera/flutter-pi/issues/49). Please make sure to read that first, contains a lot of fixes for already known problems. (You'll definitely encounter those when you build the engine yourself) Generally, there's a [blog post](https://medium.com/flutter/flutter-on-raspberry-pi-mostly-from-scratch-2824c5e7dcb1) by a google engineer describing the rough outline of the engine build and everything in the issue I mentioned is basically fixups on top of that guidance. There's also [some instructions for building for arm64](https://docs.loliot.net/docs/lang/flutter/engine/flutter-engine-for-linux-arm64) on a wiki page by hhk. Though I advice you to first start off with the blog post I mentioned earlier, the instructions on that wiki page use custom engine forks and stuff like that, better to start off with mainline first.

If you encounter any problems, just leave a comment on the flutter engine building issue.

> I also have another question for you. What build is working on the RPi? I don't think it is the Android build. Will it be the Linux version? Where would I set the launch screen for my app? Sometimes [...]

What do you mean what build? For flutter-pi, you're building the asset bundle, which is completely platform-agnostic; you're not building any linux / android bundle. Flutter-pi is sadly not comparable to a linux desktop build, anything you do in the `linux` subdirectory of your flutter project is completely meaningless with flutter-pi. Though, work is being done to bring more support to embedders like flutter-pi into the flutter SDK, so maybe it will work that way at some point.

manishbuttan, 2021-03-20:
I guess I am trying to figure out how to make Firebase work on Flutter pi. It is supported for iOS, android and web. Any ideas what approach to take for Firebase?

ardera, 2021-03-20:
You're not the first one, see #108 and #122. Generally, if you need want to write any native code, you need to do it inside flutter-pi.

To make firebase work for you, you need to implement the platform interfaces for any of the firebase packages you want to use. See [here](https://github.com/FirebaseExtended/flutterfire/tree/master/packages), each of those packages has a `<package>_platform_interface` sub directory with the specific platform interface as a class somewhere inside it.

I think it'd be best to generate some C wrappers for the firebase C++ API (there's surely some tool out there that does this for you) and then use dart ffi to call those wrappers (you can use `ffigen` to generate the dart bindings) in your platform interface. You can't call C++ directly from dart ffi, that's the reason for the wrappers.

And the wrappers you just put somewhere inside flutter-pi. You can also merge it into upstream flutter-pi at some point if you want to and publish the platform interfaces on pub.dev.