ardera/flutter-pi #162: How to use flutter-pi to choose chinese

tsuiwade, 2021-03-24:
First of all thank you for your dedication.

I followed your steps and used the **gallery** on my raspberrypi4.

When I select locale as chinese, the text will display ☐ ☐ ☐.

I am a newbie in this area, so I would like to ask for advice.

1. I installed the Raspberry Pi desktop system, and there is no problem with Chinese fonts displayed on the desktop system.

2. I followed your method and set the Raspberry Pi to start in the console mode. At this time, the Chinese I displayed is also ☐☐. I don't know if this is the problem.
3. I have installed Chinese ttf according to the method on the Internet, and set the locale to zh_CN.utf8.

I don’t know what went wrong. Raspberry Pi system or console mode or flutter-pi or gallery program.
Thank you!

ardera, 2021-04-03:
I have no idea how flutter resolves fonts, but I think this not something related to flutter-pi. You can try opening an issue on the flutter repo or asking in the #help section of the flutter discord

kokocooler2020, 2021-04-09:
@tsuiwade 
I solved this problem by using **Custom-Fonts** in flutter, You can use **Chinese font .ttf file**, 
I passed the test on Raspberry Pi 4B, It works great.

@ardera **Thank you** for your selfless dedication, this is great！

ardera, 2021-04-10:
@kokocooler2020 thx!
Closing this now

closed 2021-04-10 by ardera

tsuiwade, 2021-04-13:
Thank you!
I downloaded the 'FontCreator' software and checked the default ` GalleryIcons.ttf ` font, not including Chinese.
So I changed to a file with Chinese font and it was successful！