ardera/flutter-pi #164: Render to 3.5inch RPi Display

fusion44, 2021-03-28:
Hi,

I'm trying to render to a [3.5inch RPi Display](http://lcdwiki.com/3.5inch_RPi_Display) but I can't get it working. It's not connected through HDMI but the GPIO pins. Not sure if this makes a difference for flutter-pi? The standard command line output to the display is working well. 

As soon as I plug in another touchscreen via HDMI flutter-pi renders to it just fine.

A full log is attached at the end.

I notice some warnings which depend on how I start the app:
Command: `dev/flutter-pi/build/flutter-pi -d "73.44, 48.96" --release ~/dev/hello_pi/`

> [flutter-pi] WARNING: display didn't provide valid physical dimensions.
             The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
WARNING: Kernel didn't return a valid vblank timestamp. (timestamp == 0)
         VSync will be disabled.
         See https://github.com/ardera/flutter-pi/issues/38 for more info.
===================================
display mode:
  resolution: 1024 x 768
  refresh rate: 60Hz
  physical size: 73mm x 0mm
  flutter device pixel ratio: 1.000000
===================================
...

Command: `dev/flutter-pi/build/flutter-pi -d "73, 49" --release ~/dev/hello_pi/`
>[flutter-pi] WARNING: display has non-square pixels. Non-square-pixels are not supported by flutter.
WARNING: Kernel didn't return a valid vblank timestamp. (timestamp == 0)
         VSync will be disabled.
         See https://github.com/ardera/flutter-pi/issues/38 for more info.
===================================
display mode:
  resolution: 1024 x 768
  refresh rate: 60Hz
  physical size: 73mm x 49mm
  flutter device pixel ratio: 3.691420
===================================

Can flutter-pi handle fractional dimensions?

Also I notice that flutter-pi wants to use 1024 x 768 as resolution where the display only can handle 320 x 480 px. Where to change this?

Another warning which I get independent from which display I use is this:
>[modesetting] Could not commit atomic request. drmModeAtomicCommit: Device or resource busy
[compositor] Non-blocking drmModeAtomicCommit failed with EBUSY.
             Future drmModeAtomicCommits will be executed blockingly.
             This may have have an impact on performance.

The HDMI display works even with this warning. 

It's a PI 4 with 4GB of ram running BUSTER.
```sh
admin@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux
```

Any idea what might cause this? Thank you.

**Full log:**
>admin@raspberrypi:~ $ sudo dev/flutter-pi/build/flutter-pi -d "73, 49" --release ~/dev/hello_pi/
[flutter-pi] WARNING: display has non-square pixels. Non-square-pixels are not supported by flutter.
WARNING: Kernel didn't return a valid vblank timestamp. (timestamp == 0)
         VSync will be disabled.
         See https://github.com/ardera/flutter-pi/issues/38 for more info.
===================================
display mode:
  resolution: 1024 x 768
  refresh rate: 60Hz
  physical size: 73mm x 49mm
  flutter device pixel ratio: 3.691420
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 3.1 Mesa 19.3.2"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_primitive_bounding_box GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_EXT_texture_norm16 GL_KHR_context_flush_control GL_NV_image_formats GL_OES_draw_elements_base_vertex GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_EXT_buffer_storage GL_EXT_float_blend GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_shader_image_atomic GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_MESA_framebuffer_flip_y GL_EXT_texture_query_lod "
===================================
[modesetting] Could not commit atomic request. drmModeAtomicCommit: Device or resource busy
[compositor] Non-blocking drmModeAtomicCommit failed with EBUSY.
             Future drmModeAtomicCommits will be executed blockingly.
             This may have have an impact on performance.

ardera, 2021-03-28:
> It's not connected through HDMI but the GPIO pins. Not sure if this makes a difference for flutter-pi? The standard command line output to the display is working well.

The way it's connected doesn't make a difference, but the fact that it has an fbdev driver rather than a KMS driver. (KMS is the newer way to control display / graphics output in linux, fbdev the older one, flutter-pi uses KMS) So your display isn't supported per-se but it's still possible to get it working.

Actually, did you make sure you installed your TFT correctly according to http://www.lcdwiki.com/MHS-3.5inch_RPi_Display?
`rpi-fbcp` and the correct video mode in `/boot/config.txt` should be installed in the process.

> Can flutter-pi handle fractional dimensions?

No, but that also shouldn't be necessary, the physical dimensions are only used to calculate the display density and a fraction of a millimeter shouldn't make a difference in display density.

> Another warning which I get independent from which display I use is this:
>
>
> The HDMI display works even with this warning.

Yeah I don't know either what causes this. The warning said it may affect performance, but I didn't really experience that so it may be fine. It's unrelated to your TFT display though

`rpi-fbcp` may not be particularly fast and may also cost some performance. (It copies over at 40FPS at max) Official `fbdev` support is on my TODO list, so this all should work out of the box at some point.

fusion44, 2021-04-03:
Yes I can render images via FBI to the screen. I take that this is a sign that the driver is installed correctly.

Do you mean this [fbcp](https://github.com/tasanakorn/rpi-fbcp)? 

I've tried running it but I get an error:
>admin@raspberrypi:~ $ sudo fbcp 
assertion failure:/build/raspberrypi-userland-HqXp8U/raspberrypi-userland-2+git20201022~151804+e432bc3/interface/vmcs_host/vc_vchi_dispmanx.c:201:vc_vchi_dispmanx_init():success == 0
Aborted

I've traced it to a [userland](https://github.com/raspberrypi/userland/blob/3fd8527eefd8790b4e8393458efc5f94eb21a615/interface/vmcs_host/vc_vchi_dispmanx.c#L148) driver from Broadcom, so actually the driver might still have some problems.

ardera, 2021-04-03:
> Do you mean this [fbcp](https://github.com/tasanakorn/rpi-fbcp)?

Exactly

> I've tried running it but I get an error:
>
>
> I've traced it to a [...]

Are you using raspbian 64bit? I think that may have some problems. Other than that maybe full-KMS could cause issues if you're using that

fusion44, 2021-04-04:
It is a modified version of the 64 bit Raspbian OS. Possibly some modification causes this. 
I'll take a deeper look into this.

>admin@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux

fusion44, 2021-05-08:
Sorry this took so long. I've had another try on this today. 

Steps I took:
* Flash an SD card with the latest 64 bit raspiOS
* Install display driver from https://github.com/tux1c/wavesharelcd-64bit-rpi.git
* Try LXDE desktop (renders fine to the LCD, touchscreen OK)
* Change settings (Switch to console mode, Enable V3D driver, Set GPU memory)
* Install engine binaries and build flutter-pi
* Build asset bundle and transfer to the PI

Running the app yields this output now:
```
[keyboard] Could not find "XKBMODEL" property inside "/etc/default/keyboard". Default value will be used.
[keyboard] Could not find "XKBLAYOUT" property inside "/etc/default/keyboard". Default value will be used.
[keyboard] Could not find "XKBVARIANT" property inside "/etc/default/keyboard". Default value will be used.
[keyboard] Could not find "XKBOPTIONS" property inside "/etc/default/keyboard". Default value will be used.
[flutter-pi] Could not find a connected connector!
```

Not sure what this connector is that flutter-pi is looking for. 

The readme of the waveshare driver says that `dtoverlay=vc4-fkms-v3d` must not be present in the `/boot/config.txt` file but it is added automatically by raspi-config when enabling the V3D driver. So I suspect that the two systems are incompatible with each other.

Thank you very much.

ardera, 2021-05-08:
I actually doubt they're incompatible. I'm not excluding it, but at least the "Could not find a connected connector!" message is not related to any incompatiblity. It should most likely disappear if you put `hdmi_force_hotplug=1` inside the `config.txt`.

You can also take a look at #112, someone else got it to work with a similiar display. Those small LCDs are all pretty similiar in their software implementation, I really think you can get it to work

fusion44, 2021-05-08:
Yay!

Adding this did the trick!

```
hdmi_force_hotplug=1
hdmi_timings=320 0 8 32 40 240 1 3 4 6 0 0 0 40 0 4048000 1
hdmi_group=2
hdmi_mode=87
hdmi_force_mode=1
``` 
It renders correctly, although slow. Now I just need to get the touchscreen working again.

fusion44, 2021-05-16:
OK, raw touch input from the system is actually working but flutter-pi isn't picking it up. 
I've ran a [test python script](https://stackoverflow.com/a/16682549) to intercept input events from `/dev/input/event0` and I get lots of touch events like these:
```
Event type 3, code 0, value 1403 at 1621157063.231088
Event type 3, code 1, value 3286 at 1621157063.231088
Event type 3, code 24, value 138 at 1621157063.231088
```

running flutter-pi with `-i /dev/input/event{0}` didn't help

I'll dig into this a little more this evening.

fusion44, 2021-05-16:
Ok, input works! At least kind of. Seems like something is off while calculating the touch point x / y coordinates in [user_input.c](https://github.com/ardera/flutter-pi/blob/a9d635cb663f503b6afad8c8ad2d9042bd32ff62/src/user_input.c#L862).

I've added a debug print to user_input.c after the two calls:
```c
printf("x: %f y: %f display_width: %d display_height: %d\n", x, y, input->display_width, input->display_height);
```
Here's the output when I tap near the **top left corner**
`x: 38.825195 y: 301.476807 display_width: 480 display_height: 320`

**bottom left corner**
`x: 446.840576 y: 303.190186 display_width: 480 display_height: 320`

**bottom right corner**
`x: 450.231934 y: 25.389160 display_width: 480 display_height: 320`

**top right corner**
`x: 38.825195 y: 12.538818 display_width: 480 display_height: 320`

Seems like the the input is assigned to the wrong sides? 
On the short side (**x**) I see values from ~40 to ~450 and on the long side (**y**) I see values from 13 to 300.
The thing is that `display_width` and `display_height` are actually correct but libinput seems to calculate the touch coordinated flipped.

ardera, 2021-05-16:
Seems like your coords are rotated 90 degrees clock-wise (or your screen -90 degrees ccw).

flutter-pi doesn't have a way to deal with that itself, but you can fix this a layer lower by setting some udev properties for the device.

this may work:
1. find out the name of your touchscreen device. For example, if your touchscreen device is `/dev/input/event0`, execute
    ```
    udevadm info -a -p /sys/class/input/event0
    ```
    The value of `ATTRS{name}` is your device name.

2. create a new udev config file
    ```bash
    sudo nano /etc/udev/rules.d/98-touchscreen-cal.rules
    ```
3. add a new rule for your touchscreen which has the correct calibration matrix. According to [this](https://wayland.freedesktop.org/libinput/doc/1.11.3/group__config.html#ga09a798f58cc601edd2797780096e9804), the setup for 270 CW rotation (which is same as rotating 90 degrees CCW of course) is this:
    ```
    ATTRS{name}=="device name", ENV{LIBINPUT_CALIBRATION_MATRIX}="0.0 1.0 0.0 -1.0 0.0 1.0"
    ```

fusion44, 2021-05-18:
Touch input works perfectly with this matrix:
```
90 deg cw:
[ 0 -1 1] 
[ 1  0 0] 
```

I'll probably have to disable animations as the animations only play at 5-6 fps. 

Again, thank you very much!

ardera, 2021-05-18:
> I'll probably have to disable animations as the animations only play at 5-6 fps.

That's unexpected, for me, most animations run pretty smoothly. Maybe it works better with release mode

> Again, thank you very much!

np, glad it works!

fusion44, 2021-05-20:
One thing I notice is that the image is displayed pixel row by pixel row from top to bottom with a very tiny but noticeable delay between two rows.

I don't think this is a problem of Release / Debug mode. My 7 inch screen which is attached via HDMI is pretty smooth even in debug mode.

fusion44, 2021-05-30:
I dug a little more into the rendering speed. The problem seems to be the transfer rate of the SPI itself, which is insufficient for the resolution of the rendered image. There is an [advanced driver](https://github.com/juj/fbcp-ili9341), but it is not available for 64 bit systems yet.

I could get it to render a lot faster at the expense of visual quality by tweaking some settings in the /boot/config.txt file:
old: `dtoverlay=waveshare35a:rotate=90`
new: `dtoverlay=waveshare35a:rotate=90,speed=41000000,fps=50`

I found this snippet on the internet but was unable to find official documentation for the speed value. With this I'm getting about 15 fps which is OK but the overall visual quality degrades a lot.

fusion44, 2021-06-07:
Ok, I think we can consider this issue fixed.

Thank you for your help!

closed 2021-06-07 by fusion44

ardera, 2021-06-11:
> I dug a little more into the rendering speed. The problem seems to be the transfer rate of the SPI itself, which is insufficient for the resolution of the rendered image. There is an [advanced [...]

That's interesting! That driver is a cool project. Theoretically, you could get some more speed by cutting away the whole fbdev/dispmanx/KMS layer and make flutter-pi talk to that advanced driver directly, so a lot less memory copying, more performance etc. That's especially useful bc flutter will get support for partial screen updates some time in the future (it's being worked on right now, although it seems to be a bit stalled) so one could also pipe those partial screen updates to the driver directly, instead of the driver having to do manual memory comparison to find out what parts of the image changed. But that's probably a lot of work😄 

> Ok, I think we can consider this issue fixed.
> 
> Thank you for your help!

no problem!