ardera/flutter-pi #165: env variables are not passed to dart vm

GioPan04, 2021-03-29:
The env variables defined in the host system are not passed to the dartvm.

Example:
```dart
void main() {
  final String user = Platform.environment['USER'];
  print(user);
}
```
This will output null.

Since you can't build a bundle with `--dart-define`, there is any way to pass a env variable to the flutter app?

pezi, 2021-03-30:
I can not confirm this behavior - tested inside my flutter app

``` Dart
    print(Platform.resolvedExecutable);
    print(Platform.environment['PATH']);
```

works fine

``` cli
flutter: /usr/local/bin/flutter-pi
flutter: /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games:/opt/dart-sdk/bin
```

ardera, 2021-03-30:
From a first look it seems like `Platform.environment` internally delegates to the dart runtime, which will in turn just copy everything from from [`environ`](https://man7.org/linux/man-pages/man7/environ.7.html). Don't know why it shouldn't work, the implementation is exactly the same for linux desktop and for android too, but I'll test it

GioPan04, 2021-03-31:
Okay, I'll test this