ardera/flutter-pi #166: MissingPluginException No implementation found for method getPlatformVersion on Raspberry Pi board

sachin-lg, 2021-03-30:
Issue Summary:-
We have created one sample plugin (getPlatformVersion) for Linux desktop platform and executed the same and it executed properly.
Next we want to execute the same plugin on raspberry pi board so for that we created build folder using “flutter build bundle” command and deployed the same raspberry pi board and executed the app using flutter embedder (flutter-pi) as mentioned in below command:
<flutter embedder exe flutter assets folder path> e.g. <./flutter-pi ./flutter_assets>
With this app is being launched but it is not able to access platform APIs through plugin and got below mentioned error:

```
[ERROR:flutter/lib/ui/ui_dart_state.cc(186)] Unhandled Exception: MissingPluginException(No implementation found for method getPlatformVersion on channel version)
#0      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:156:7)
<asynchronous suspension>
#1      Version.platformVersion (package:version/version.dart:11:28)
<asynchronous suspension>
#2      _MyAppState.initPlatformState (package:version_example/main.dart:30:25)

```
Note:- We have already cross compiled flutter engine for raspberry pi board and able to execute sample app also.
Plugin Name :- version

## Steps to Reproduce
1. Run 'flutter create --template=plugin --platforms=linux version'
2. After creating plugin, go to example directory under plugin directory and run 'flutter build bundle'
3. deploy the build folder in raspberry pi board.
4. Run app in the raspberry pi board using flutter embedder flutter-pi. Run below command:
<flutter embedder exe flutter assets folder path> e.g. <./flutter-pi ./flutter_assets>

Expected results:- App should be launched and there should be platform(Linux) version display in the app

Actual results:- App is launched but version is displayed as Unknown.

Result of 'flutter analyze' on app :- OK
Result of 'flutter analyze' on plugin :- OK 

Output of running `flutter doctor -v`-

```
sachin@sachin-VirtualBox:~/Sachin/Flutter/plugins/version$ flutter doctor -v
[✓] Flutter (Channel master, 1.26.0-2.0.pre.275, on Linux, locale en_IN)
    • Flutter version 1.26.0-2.0.pre.275 at /home/sachin/Sachin/Flutter/flutter
    • Framework revision d9044a8a61 (3 months ago), 2021-01-09 04:04:04 -0500
    • Engine revision caf6a8191f
    • Dart version 2.12.0 (build 2.12.0-204.0.dev)

[✗] Android toolchain - develop for Android devices
    ✗ Unable to locate Android SDK.
      Install Android Studio from: https://developer.android.com/studio/index.html
      On first launch it will assist you in installing the Android SDK components.
      (or visit https://flutter.dev/docs/get-started/install/linux#android-setup for detailed instructions).
      If the Android SDK has been installed to a custom location, please use
      `flutter config --android-sdk` to update to that location.

[✓] Chrome - develop for the web
    • Chrome at google-chrome

[✓] Linux toolchain - develop for Linux desktop
    • clang version 6.0.0-1ubuntu2 (tags/RELEASE_600/final)
    • cmake version 3.10.2
    • ninja version 1.8.2
    • pkg-config version 0.29.1

[!] Android Studio (not installed)
    • Android Studio not found; download from https://developer.android.com/studio/index.html
      (or visit https://flutter.dev/docs/get-started/install/linux#android-setup for detailed instructions).

[✓] VS Code (version 1.54.3)
    • VS Code at /usr/share/code
    • Flutter extension version 3.20.0

[✓] Connected device (2 available)
    • Linux (desktop) • linux  • linux-x64      • Linux
    • Chrome (web)    • chrome • web-javascript • Google Chrome 87.0.4280.66

! Doctor found issues in 2 categories.
```

Query:-
1.	As we know that for native plugins, the supported platforms are android, ios, web, linux, macos, and windows. Can we support native plugins for any other embedded system also ?
If we can, what is the procedure for the same ?
2.	If we have created bundle for raspberry pi board, how we can execute the same ? Is “kernel_blob.bin” enough to launch an app with plugin or some supporting files are also required like mentioned below ?
While executing the same on linux desktop using flutter run, files are being generated in “ephemeral” folder under linux folder but while launching the same in raspberry pi board, we are making use of only kernel_blob.bin file.

ardera, 2021-03-30:
> 1\. As we know that for native plugins, the supported platforms are android, ios, web, linux, macos, and windows. Can we support native plugins for any other embedded system (e.g. raspberry pi) also ? If we can, what is the procedure for the same ?

Native plugins are supported, but they don't work the same way plugins for any other platforms work. For normal platforms, the native code resides in the `linux`, `android` subdirectories of your flutter project. For flutter-pi, this native code is inside flutter-pi's [`src/plugins`](https://github.com/ardera/flutter-pi/tree/master/src/plugins) directory. I'm afraid if you want to write native C code, you'll probably have to do it inside flutter-pi. I'd look at maybe using dart:ffi if possible. I got the GPIO, serial and SPI plugins working purely using ffi without any native code.

> 2\. If we have created bundle for raspberry pi board, how we can execute the same ? Is “kernel_blob.bin” enough to launch an app with plugin or some supporting files are also required ?

Since the native code lives inside flutter-pi, all you need is the assets bundle. (All the necessary code is inside `kernel_blob.bin` or `app.so` if using release mode)

sachin-lg, 2021-03-31:
> > 1. As we know that for native plugins, the supported platforms are android, ios, web, linux, macos, and windows. Can we support native plugins for any other embedded system (e.g. raspberry pi) also ? If we can, what is the procedure for the same ?
> 
> Native plugins are supported, but they don't work the same way plugins for any other platform work. For normal platforms, the native code resides in the `linux`, `android` subdirectories of your flutter project. For flutter-pi, this native code is inside flutter-pi's [`src/plugins`](https://github.com/ardera/flutter-pi/tree/master/src/plugins) directory. I'm afraid if you want to write native C code, you'll probably have to do it inside flutter-pi. I'd look at maybe using dart:ffi if possible. I got the GPIO, serial and SPI plugins working purely using ffi without any native code.
> 
> > 2. If we have created bundle for raspberry pi board, how we can execute the same ? Is “kernel_blob.bin” enough to launch an app with plugin or some supporting files are also required ?
> 
> Since the native code lives inside flutter-pi, all you need is the assets bundle. (All the necessary code is inside `kernel_blob.bin` or `app.so` if using release mode)

Thanks for your response.
We will analyse the same as per your direction.