ardera/flutter-pi #168: can I build app on raspberry pi 4?

icecreamzhao, 2021-04-04:
I get this error when execute `flutter build apk` or `flutter build bundle`:

```
Snapshot not compatible with the current VM configuration: the snapshot requires
'release no-dwarf_stack_traces_mode no-causal_async_stacks lazy_async_stacks
lazy_dispatchers use_bare_instructions no-dedup_instructions no-"asserts"
"use_field_guards" "use_osr" no-code-comments x64-sysv no-null-safety' but the
VM has 'release no-dwarf_stack_traces_mode no-causal_async_stacks
lazy_async_stacks lazy_dispatchers use_bare_instructions no-dedup_instructions
no-"asserts" "use_field_guards" "use_osr" no-code-comments arm64-sysv
no-null-safety'
```

ardera, 2021-04-04:
This is not really flutter-pi related, but in the readme it says you can't use a Pi as your development machine. If you look at the error message you'll find that the snapshot (= the flutter tool) requires `x64-sysv` but the dart VM trying to execute the flutter tool uses `arm64-sysv`.

closed 2021-04-04 by ardera

icecreamzhao, 2021-04-04:
> This is not really flutter-pi related, but in the readme it says you can't use a Pi as your development machine. If you look at the error message you'll find that the snapshot (= the flutter tool) requires `x64-sysv` but the dart VM trying to execute the flutter tool uses `arm64-sysv`.

OK, thanks a lot.