ardera/flutter-pi #169: I can´t install Flutter-Pi

Asmardos, 2021-04-09:
`sudo apt install cmake libgl1-mesa-dev libgles2-mesa-dev libegl1-mesa-dev libdrm-dev libgbm-dev ttf-mscorefonts-installer fontconfig libsystemd-dev libinput-dev libudev-dev  libxkbcommon-dev
[sudo] Passwort für pi: 
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
fontconfig ist schon die neueste Version (2.13.1-2).
fontconfig wurde als manuell installiert festgelegt.
libxkbcommon-dev ist schon die neueste Version (0.8.2-1).
libxkbcommon-dev wurde als manuell installiert festgelegt.
cmake ist schon die neueste Version (3.16.3-3~bpo10+1).
Einige Pakete konnten nicht installiert werden. Das kann bedeuten, dass
Sie eine unmögliche Situation angefordert haben oder, wenn Sie die
Unstable-Distribution verwenden, dass einige erforderliche Pakete noch
nicht erstellt wurden oder Incoming noch nicht verlassen haben.
Die folgenden Informationen helfen Ihnen vielleicht, die Situation zu lösen:

Die folgenden Pakete haben unerfüllte Abhängigkeiten:
 libegl1 : Hängt ab von: libegl-mesa0 soll aber nicht installiert werden
E: Fehler: Unterbrechungen durch pkgProblemResolver::Resolve hervorgerufen; dies könnte durch zurückgehaltene Pakete verursacht worden sein.
`
 libegl-mesa0 is not availabel for raspbian but libegl1 depends on it
where am i wrong?

`cmake ..
-- CMAKE_BUILD_TYPE not set, defaulting to Release.
-- The C compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Generator .............. Unix Makefiles
-- Build Type ............. Release
-- Flutter Channel not set, defaulting to stable
-- Flutter Channel ........ stable
-- Engine SHA1 ............ 2dce47073a378673f6ca095e91b8065544c3a881
-- PKG_CONFIG_PATH ........ 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking for module 'libdrm'
--   No package 'libdrm' found
CMake Error at /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:493 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:677 (_pkg_check_modules_internal)
  CMakeLists.txt:89 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/home/pi/flutter-pi/build/CMakeFiles/CMakeOutput.log".
`
an dis this related to the other problem?
libdrm is also not available for raspbian.

Asmardos, 2021-04-09:
i am relativly new to linux and raspbian and 
my englisch is also far from perfect please frogive me that

ardera, 2021-04-09:
can you execute `sudo apt update` and then try installing again?

Asmardos, 2021-04-09:
I already tryed this and dist upgrade too

Asmardos, 2021-04-09:
oh sry I tryed apt-get update

Asmardos, 2021-04-09:
I have done this and it says on all upgrades faild to retrive http//...  temporary failture to resove raspbian.raspberrypi.org

Asmardos, 2021-04-09:
it also says this when i do apt-get upgrade 
this was´t like this last time I tryed

Asmardos, 2021-04-09:
coud it be that the server is down?

Asmardos, 2021-04-09:
ok it seems like my pi is the problem I will use a backup and try it again on monday 
thanks for the help

Asmardos, 2021-04-12:
On the new device it worked,
thank you very mutch
now is my only problem that i don´t know how i can build the app.so

Asmardos, 2021-04-12:
s

closed 2021-04-12 by Asmardos