ardera/flutter-pi #173: DART SDK update

kekko7072, 2021-04-18:
After I update the dart SDK I have this issue, is there a way to fix ?
```
`[ERROR:engine/src/flutter/shell/common/shell.cc(242)] Dart Error: Can't load Kernel binary: Invalid SDK hash.
[ERROR:engine/src/flutter/runtime/dart_isolate.cc(169)] Could not prepare isolate.
[ERROR:engine/src/flutter/runtime/runtime_controller.cc(401)] Could not create root isolate.
[ERROR:engine/src/flutter/shell/common/shell.cc(571)] Could not launch engine with configuration.
flutter: Observatory listening on http://127.0.0.1:41021/1QtkyJcllj4=/`
```

ardera, 2021-04-18:
thanks! updated them

closed 2021-04-18 by ardera

kekko7072, 2021-04-18:
Ok thank you so I just  `git pull && git checkout origin/master` to make It run right?

ardera, 2021-04-18:
yep exactly

kekko7072, 2021-05-01:
@ardera 
Hey new DART UPDATE available, can you update the engine?

At the end I wasn't able to update using `git pull && git checkout origin/master` I deleted and reinstalled the flutter-pi. 
Where should I run this code? In the RPi or in the Linux VM ?  If in the RPi in witch directory?

Thank you so much I am developing crazy things using Flutter RPI you changed my life!

ardera, 2021-05-02:
> @ardera 
> Hey new DART UPDATE available, can you update the engine?
> 
> At the end I wasn't able to update using `git pull && git checkout origin/master` I deleted and reinstalled the flutter-pi. 
> Where should I run this code? In the RPi or in the Linux VM ?  If in the RPi in witch directory?

updated them (was already working on it when you wrote)

You execute them in your cloned engine binaries repo. I'm sorry but I won't go more into detail, a tutorial on git may help you further.

> Thank you so much I am developing crazy things using Flutter RPI you changed my life!

thx, appreciate it