ardera/flutter-pi #174: Color depth on device

mortenboye, 2021-04-19:
I have an issue where the colors and especially material shadows render with heavy banding, very visible shift from one color to the next.

It is only visible on the actual screen, not on screenshots taken with scrot.
Im running on a CM4 with this screen: https://www.waveshare.com/7hp-capqled.htm

Is this something I can fix in a setting somewhere? I have no understanding of how (fake) KMS works or how flutter-pi renders.

`fbset`
gives me
```
mode "1024x600"
    geometry 1024 600 1024 600 16
    timings 0 0 0 0 0 0 0
    accel true
    rgba 5/11,6/5,5/0,0/0
endmode
```
which indicate 16 bit color, which could explain the issue, but im not sure that's even relevant.

Any pointers?

ardera, 2021-05-13:
I'm afraid that's probably caused by the display/display driver/firmware.

flutter-pi definitely has some control over the pixel format, but it will always use ARGB8888 right now, so 32-bit color. Any other pixel format is not supported. So if its still banding that's probably caused by something below flutter-pi implicitly converting to another pixel format, so display driver / firmware / display controller chip.

`fbset` info isn't thaat relevant 😄 KMS is basically the "new" way to control displays in linux, fbdev the old one. On any device where KMS is supported, you'll still find a fbdev device, but it's just emulated by KMS for compatibility reasons. AFAIK any KMS-emulated fbdev device will have RGB565, at least that was the case for the devices I've tested. The only device where it was in fact 32-bit color was a device where KMS was not supported.

I'm also not too sure if RGB565 should band that badly. Can you send a picture?

Maybe some logging will give a hint:
- enable DRM debugging by appending `drm.debug=0x1ff` to the kernel commandline inside `/boot/cmdline.txt`
- restart
- run flutter-pi with your app
- send me the contents of `/var/log/syslog` or the output of `dmesg` (preferably the first one)
- send me the output of `sudo vcdbg log msg`
- make sure to disable DRM debugging again

Other than that, you can try specifying a custom EDID file or disabling the monitor supplied EDID in `/boot/config.txt`, see [here](https://www.raspberrypi.org/documentation/configuration/config-txt/video.md#hdmi_ignore_edid)

mortenboye, 2021-05-14:
Thanks for the suggestions.

This is the banding Im seeing
[image]

Here are the logs:

/var/log/syslog:
[syslog.log](https://github.com/ardera/flutter-pi/files/6478940/syslog.log)

vcdbg log msg:
```
pi@raspberrypi:~ $ sudo vcdbg log msg
005608.471: arasan: arasan_emmc_open
006085.855: brfs: File read: /mfs/sd/config.txt
006086.581: brfs: File read: 1164 bytes
006178.965: brfs: File read: /mfs/sd/config.txt
006179.684: gpioman: gpioman_get_pin_num: pin DISPLAY_SDA not defined
006179.704: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
006195.283: brfs: File read: 1164 bytes
006200.327: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
006200.342: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
006200.374: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
006200.391: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
006655.234: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
006656.514: *** Restart logging
006697.496: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
006697.509: HDMI0: hdmi_pixel_encoding: 300000000
006697.523: HDMI1: hdmi_pixel_encoding: 300000000
006698.167: gpioman: gpioman_get_pin_num: pin CAMERA_0_I2C_PORT not defined
006702.447: dtb_file 'bcm2711-rpi-cm4.dtb'
006707.525: brfs: File read: /mfs/sd/bcm2711-rpi-cm4.dtb
006707.541: Loading 'bcm2711-rpi-cm4.dtb' to 0x100 size 0xc032
006720.013: brfs: File read: 49202 bytes
006726.239: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
006790.855: brfs: File read: 1523 bytes
006792.459: brfs: File read: /mfs/sd/config.txt
006792.613: dtparam: i2c_arm=off
006801.993: dtparam: spi=on
006810.139: brfs: File read: 1164 bytes
006817.785: brfs: File read: /mfs/sd/overlays/vc4-fkms-v3d.dtbo
006834.988: Loaded overlay 'vc4-fkms-v3d'
006869.133: dtwarn: overlay 'pi3-miniuart-bt' has been renamed 'miniuart-bt'
006869.463: brfs: File read: 1446 bytes
006874.466: brfs: File read: /mfs/sd/overlays/miniuart-bt.dtbo
006904.211: Loaded overlay 'pi3-miniuart-bt'
007022.094: brfs: File read: 1819 bytes
007029.206: brfs: File read: /mfs/sd/overlays/midi-uart0.dtbo
007043.308: Loaded overlay 'midi-uart0'
007083.485: dtwarn: overlay 'pi3-disable-bt' has been renamed 'disable-bt'
007084.296: brfs: File read: 716 bytes
007087.616: brfs: File read: /mfs/sd/overlays/disable-bt.dtbo
007123.177: Loaded overlay 'pi3-disable-bt'
007203.011: brfs: File read: 1073 bytes
007204.493: brfs: File read: /mfs/sd/overlays/dwc2.dtbo
007218.945: Loaded overlay 'dwc2'
007218.969: dtparam: dr_mode=host
007244.004: brfs: File read: 801 bytes
007253.988: brfs: File read: /mfs/sd/overlays/vc4-fkms-v3d.dtbo
007297.161: Loaded overlay 'vc4-fkms-v3d'
007402.904: brfs: File read: 1446 bytes
007404.628: brfs: File read: /mfs/sd/cmdline.txt
007404.692: Read command line from file 'cmdline.txt':
007404.716: 'console=tty1 root=PARTUUID=59f7905f-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait drm.debug=0x1ff'
008596.286: brfs: File read: 114 bytes
009915.126: brfs: File read: /mfs/sd/kernel7l.img
009915.163: Loading 'kernel7l.img' to 0x8000 size 0x662688
009915.208: Device tree loaded to 0x2eff3800 (size 0xc75a)
009925.962: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
013885.263: vchiq_core: vchiq_init_state: slot_zero = 0xded00000, is_master = 1
013888.520: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
013895.988: TV service:host side not connected, dropping notification 0x00000002, 0x00000002, 0x00000057
```

ardera, 2021-05-25:
I have no idea what causes this, the logs don't say anything unusual. Have you tried supplying a custom EDID or disabling the monitor EDID?

mortenboye, 2021-05-25:
I tried disabling EDID with no change, but will try to enable dithering.

I have a couple of new displays on the way, that I can try instead. It could very well be the display.

mortenboye, 2021-05-26:
New displays arrived today and does not show the save banding. Thank you for the assistance. Closing.

closed 2021-05-26 by mortenboye