ardera/flutter-pi #176: Camera & Video Support?

jtkeyva, 2021-05-25:
Does the rpi camera module or a usb camera work with this? Can it record and playback videos at 4k?
thanks!

pezi, 2021-05-25:
I started a private project to port 
https://github.com/caprica/picam
https://github.com/caprica/picam-native
to dart using FFI. But this port will only support still images.

ardera, 2021-05-25:
Video playback works via [`omxplayer_video_player`](https://pub.dev/packages/omxplayer_video_player), but there's no plugin for camera playback right now.

But I think it's possible to get working with a workaround. On newer raspbian, the pi cam or any other camera should be present as a `/dev/video0` file. Some other people tried to get omxplayer to playback that file: https://www.raspberrypi.org/forums/viewtopic.php?t=69322

So if we take the first one as an example:
```
mkfifo /tmp/out.raw
mplayer tv:///dev/video0 -dumpstream -dumpfile /tmp/out.raw
```

And then, instead of invoking omxplayer with `omxplayer <some path or URL>`, you initialize your `VideoPlayerController` with:
```dart
VideoPlayerController.file("/tmp/out.raw")
```

This is not particularly optimal though, maybe there's delay or high CPU usage because so much memory will be copied, so it'd still be cool to have a _real_ camera plugin

Btw, `omxplayer_video_player` is just a backend of [`video_player`](https://pub.dev/packages/video_player). So for docs on how to use it to play video you actually need to look at the video_player package, or you use some higher level video player like Chewie :)

jtkeyva, 2021-05-26:
thanks @pezi for now a photo will be fine so i will check it out

@ardera got it thanks. i am familiar with omxplayer so i may tinker a bit. all sounds hopeful and look forward to the progress

jtkeyva, 2021-05-26:
> I started a private project to port
> https://github.com/caprica/picam
> https://github.com/caprica/picam-native
> to dart using FFI. But this port will only support still images.

cool! any chance of sharing it? photo is fine foe now. thx

pezi, 2021-05-26:
> cool! any chance of sharing it? photo is fine foe now. thx

I will publish this port on pub.dev in the next time.

Tek-hp, 2022-02-22:
Does anyone have an example on how to use omx player ?

madc, 2023-01-20:
@pezi Did you make any progress?