ardera/flutter-pi #177: Run Flutter App in Raspberry Pi OS Desktop

lakshanthad, 2021-06-11:
Hi,
Currently, I have installed flutter-pi on my RPi4 and have successfully run an app after logging into the command-line by executing "flutter-pi <app_name>".  However, I want to run this app after logging into Raspberry Pi OS Desktop. Preferably, if I could run a script after logging in, to open this app. Help would be appreciated!

ardera, 2021-06-11:
Hey, unfortunately that won't work.

Flutter-pi is completely incompatible with any desktop environment, so Raspberry Pi OS Desktop with flutter-pi won't work. But you can still get flutter running, flutter officially supports linux desktop. You need the 64bit Raspberry Pi OS though.

So just install Raspberry Pi OS Desktop 64bit and you can install the flutter SDK on your Pi.

lakshanthad, 2021-06-11:
Really appreciate your quick response! I will try with Raspberry Pi OS Desktop 64bit.

closed 2021-06-14 by lakshanthad