ardera/flutter-pi #179: Dart Error: Can't load Kernel binary: Invalid SDK hash.

lakshanthad, 2021-06-14:
Hi,
I am having the following errors when trying to run "flutter-pi <app_name>"

[ERROR:engine/src/flutter/shell/common/shell.cc(103)] Dart Error: Can't load Kernel binary: Invalid SDK hash.
[ERROR:engine/src/flutter/runtime/dart_isolate.cc(171)] Could not prepare isolate.
[ERROR:engine/src/flutter/runtime/runtime_controller.cc(415)] Could not create root isolate.
[ERROR:engine/src/flutter/shell/common/shell.cc(588)] Could not launch engine with configuration.

My host pc flutter config is as follows:

Flutter 2.2.2 • channel stable • https://github.com/flutter/flutter.git
Framework • revision d79295af24 (3 days ago) • 2021-06-11 08:56:01 -0700
Engine • revision 91c9fc8fe0
Tools • Dart 2.13.3

Help would be appreciated!

lakshanthad, 2021-06-14:
I just noticed that engine-binaries for flutter-pi is in version 2.2.1, whereas the latest flutter SDK is in version 2.2.2. So I was able to get it working by downgrading to SDK version 2.2.1.

Could you update the engine binaries for v2.2.2 please?

Thank you

ardera, 2021-06-14:
thx for the notice, building them right now

lakshanthad, 2021-06-14:
Great! Thank you for your contribution to the community with flutter-pi!

ardera, 2021-06-14:
> Great! Thank you for your contribution to the community with flutter-pi!

glad you appreciate it!

closed 2021-06-14 by ardera

goncalo-almeida-mc, 2021-07-02:
@ardera I just noticed to have the same issue now

I'm running Flutter latest version 2.2.3, so could update the engine binaries for this version please? :)
Thanks!

thomtomdup, 2022-02-14:
Hello.

I have the same issue. I think this is the same like before because now flutter SDK is in 2.10.1. 
Can you update to the last engine binaries for 2.10.1 please ?
Thank you very much.
Regards