ardera/flutter-pi #181: flutter-pi cannot determine the screen resolution

james-bay, 2021-06-22:
Hello! I tried to run flutter_gallery as written in the README, but encountered a problem. The flutter-pi cannot detect the screen resolution:
```
flutter-pi --release /home/pi/flutter_gallery
[modesetting] Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
[flutter-pi] Could not create drmdev from device at "/dev/dri/card1". Continuing.
[flutter-pi] WARNING: display didn't provide valid physical dimensions.
             The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
WARNING: Kernel didn't return a valid vblank timestamp. (timestamp == 0)
         VSync will be disabled.
         See https://github.com/ardera/flutter-pi/issues/38 for more info.
===================================
display mode:
  resolution: 1920 x 1080
  refresh rate: 60Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 3.1 Mesa 19.3.2"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_primitive_bounding_box GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_EXT_texture_norm16 GL_KHR_context_flush_control GL_NV_image_formats GL_OES_draw_elements_base_vertex GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_EXT_buffer_storage GL_EXT_float_blend GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_shader_image_atomic GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_MESA_framebuffer_flip_y GL_EXT_texture_query_lod "
===================================
```
The config.txt for the screen has these settings:
```
# Chalk Screen
hdmi_force_hotplug=1
hdmi_ignore_edid=0xa5000080
config_hdmi_boost=4
hdmi_group=2
hdmi_mode=28
gpu_mem=64
```
The physical resolution of the screen is 1280x800 and so `hdmi_mode=28` is the correct value. Can you tell me, please, how else can I try to set the resolution?

ardera, 2021-06-22:
can you try with `hdmi_force_mode=1`?

also make sure you've connected the monitor to HDMI port 0 since the `hdmi` options by default apply to port 0 only. (for example, use `hdmi_force_hotplug:1=1` to apply an option to HDMI port 1 instead)

james-bay, 2021-06-22:
Tried hdmi_force_mode=1 but the resolution is not detected. I connected the screen to port 0, in `config.txt` these configurations:
```ini
# Chalk Screen
hdmi_force_hotplug=0
hdmi_force_mode=0
hdmi_ignore_edid=0xa5000080
config_hdmi_boost=4
hdmi_group=2
hdmi_mode=28
gpu_mem=64
```
The edid has the following data:
```
~ $ cat /sys/devices/platform/gpu/drm/card1/card1-HDMI-A-1/edid  | edid-decode 
EDID version: 1.3
Manufacturer: SAM Model 5b0 Serial Number 0
Made in week 13 of 2009
Digital display
Maximum image size: 16 cm x 9 cm
Gamma: 2.20
RGB color display
First detailed timing is preferred timing
Display x,y Chromaticity:
  Red:   0.6396, 0.3300
  Green: 0.2998, 0.5996
  Blue:  0.1503, 0.0595
  White: 0.3125, 0.3291
Established timings supported:
  720x400@70Hz 9:5 HorFreq: 31469 Hz Clock: 28.320 MHz
  640x480@60Hz 4:3 HorFreq: 31469 Hz Clock: 25.175 MHz
  640x480@67Hz 4:3 HorFreq: 35000 Hz Clock: 30.240 MHz
  640x480@72Hz 4:3 HorFreq: 37900 Hz Clock: 31.500 MHz
  640x480@75Hz 4:3 HorFreq: 37500 Hz Clock: 31.500 MHz
  800x600@56Hz 4:3 HorFreq: 35200 Hz Clock: 36.000 MHz
  800x600@60Hz 4:3 HorFreq: 37900 Hz Clock: 40.000 MHz
  800x600@72Hz 4:3 HorFreq: 48100 Hz Clock: 50.000 MHz
  800x600@75Hz 4:3 HorFreq: 46900 Hz Clock: 49.500 MHz
  832x624@75Hz 4:3 HorFreq: 49726 Hz Clock: 57.284 MHz
  1024x768@60Hz 4:3 HorFreq: 48400 Hz Clock: 65.000 MHz
  1024x768@70Hz 4:3 HorFreq: 56500 Hz Clock: 75.000 MHz
  1024x768@75Hz 4:3 HorFreq: 60000 Hz Clock: 78.750 MHz
  1280x1024@75Hz 5:4 HorFreq: 80000 Hz Clock: 135.000 MHz
  1152x870@75Hz 192:145 HorFreq: 67500 Hz Clock: 108.000 MHz
Standard timings supported:
  1152x864@75Hz 4:3 HorFreq: 67500 Hz Clock: 108.000 MHz
  1280x800@60Hz 16:10
  1280x960@60Hz 4:3 HorFreq: 60000 Hz Clock: 108.000 MHz
  1280x1024@60Hz 5:4 HorFreq: 64000 Hz Clock: 108.000 MHz
  1440x900@60Hz 16:10 HorFreq: 55500 Hz Clock: 88.750 MHz
  1440x900@75Hz 16:10 HorFreq: 82300 Hz Clock: 156.000 MHz
  1600x1200@60Hz 4:3 HorFreq: 75000 Hz Clock: 162.000 MHz
  1680x1050@60Hz 16:10 HorFreq: 64700 Hz Clock: 119.000 MHz
Detailed mode: Clock 148.500 MHz, 160 mm x 90 mm
               1920 2008 2052 2200 hborder 0
               1080 1084 1089 1125 vborder 0
               +hsync +vsync 
               VertFreq: 60 Hz, HorFreq: 67500 Hz
Detailed mode: Clock 74.250 MHz, 160 mm x 90 mm
               1280 1720 1760 1980 hborder 0
                720  725  730  750 vborder 0
               +hsync +vsync 
               VertFreq: 50 Hz, HorFreq: 37500 Hz
Monitor ranges (GTF): 24-75Hz V, 26-81kHz H, max dotclock 230MHz
Monitor name: SyncMaster
Has 1 extension blocks
Checksum: 0xd1 (valid)

CTA extension block
Extension version: 3
31 bytes of CTA data
  Video data block
    VIC  19 1280x720@50Hz 16:9 (native) HorFreq: 37500 Hz Clock: 74.250 MHz
    VIC   4 1280x720@60Hz 16:9  HorFreq: 45000 Hz Clock: 74.250 MHz
    VIC   5 1920x1080i@60Hz 16:9  HorFreq: 33750 Hz Clock: 74.250 MHz
    VIC  20 1920x1080i@50Hz 16:9  HorFreq: 28125 Hz Clock: 74.250 MHz
    VIC   3 720x480@60Hz 16:9  HorFreq: 31469 Hz Clock: 27.000 MHz
    VIC  18 720x576@50Hz 16:9  HorFreq: 31250 Hz Clock: 27.000 MHz
    VIC  16 1920x1080@60Hz 16:9  HorFreq: 67500 Hz Clock: 148.500 MHz
    VIC  31 1920x1080@50Hz 16:9  HorFreq: 56250 Hz Clock: 148.500 MHz
    VIC  32 1920x1080@24Hz 16:9  HorFreq: 27000 Hz Clock: 74.250 MHz
    VIC  33 1920x1080@25Hz 16:9  HorFreq: 28125 Hz Clock: 74.250 MHz
    VIC  34 1920x1080@30Hz 16:9  HorFreq: 33750 Hz Clock: 74.250 MHz
  Audio data block
    Linear PCM, max channels 2
      Supported sample rates (kHz): 48 44.1 32
      Supported sample sizes (bits): 24 20 16
  Speaker allocation data block
    Speaker map:
      FL/FR - Front Left/Right
  Extended tag: Video capability data block
    YCbCr quantization: No Data (0)
    RGB quantization: No Data (0)
    PT scan behaviour: No Data (0)
    IT scan behaviour: Support both over- and underscan (3)
    CE scan behaviour: Support both over- and underscan (3)
  Vendor-specific data block, OUI 000c03 (HDMI)
    Source physical address 1.0.0.0
    Supports_AI
    DC_36bit
    DC_30bit
    DC_Y444
    Maximum TMDS clock: 225MHz
Underscans PC formats by default
Basic audio support
Supports YCbCr 4:4:4
Supports YCbCr 4:2:2
1 native detailed modes
Detailed mode: Clock 74.250 MHz, 160 mm x 90 mm
               1920 2448 2492 2640 hborder 0
                540  542  547  562 vborder 0
               +hsync +vsync interlaced 
               VertFreq: 50 Hz, HorFreq: 28125 Hz
Detailed mode: Clock 74.250 MHz, 160 mm x 90 mm
               1920 2008 2052 2200 hborder 0
                540  542  547  562 vborder 0
               +hsync +vsync interlaced 
               VertFreq: 60 Hz, HorFreq: 33750 Hz
Detailed mode: Clock 74.250 MHz, 160 mm x 90 mm
               1280 1390 1430 1650 hborder 0
                720  725  730  750 vborder 0
               +hsync +vsync 
               VertFreq: 60 Hz, HorFreq: 45000 Hz
Detailed mode: Clock 27.000 MHz, 160 mm x 90 mm
                720  732  796  864 hborder 0
                576  581  586  625 vborder 0
               -hsync -vsync 
               VertFreq: 50 Hz, HorFreq: 31250 Hz
Detailed mode: Clock 27.000 MHz, 160 mm x 90 mm
                720  736  798  858 hborder 0
                480  489  495  525 vborder 0
               -hsync -vsync 
               VertFreq: 59 Hz, HorFreq: 31468 Hz
Checksum: 0x57 (valid)
```

ardera, 2021-07-01:
Mh that's weird. flutter-pi just uses the video mode recommended by the kernel (there should _always_ be exactly one video mode recommended by the kernel), and if that's not the case it'll use the one with the highest resolution, highest refresh-rate, non-interlaced, which seems to be whats happening in your case. What happens when you run without the `hdmi_ignore_edid` option? (I guess you tried that already, but still)

Theoretically you could hardcode the resolution in the `flutter-pi.c` file, but in the long term it's probably better I provide cmdline options for forcing a specific resolution.