ardera/flutter-pi #184: Can't compile flutter-pi

goncalo-almeida-mc, 2021-06-28:
I am using a Raspberry Pi Model 3 B (2015).

I have installed:
- engine binaries, 
- cmake, graphics, system libraries and fonts;

Updated system fonts;

During the compiling process:
- $(cmake ..) command didn't raise any errors;
- $(make -j`nproc`) this is where the trouble began with several errors showing up regarding DRM variables.

What should I try to do? 
I have no idea what I did wrong. At first, cmake was raising a version error (3.7 when >3.10 was required) so I used snap's cmake (3.20))

goncalo-almeida-mc, 2021-07-01:
Fixed some problems but still fails linking

Consolidate compiler generated dependencies of target flutter-pi
[  6%] Linking C executable flutter-pi
CMakeFiles/flutter-pi.dir/src/flutter-pi.c.o: In function `init_display':
flutter-pi.c:(.text+0x30a8): undefined reference to `drmGetDevices2'
flutter-pi.c:(.text+0x387c): undefined reference to `drmCrtcGetSequence'
flutter-pi.c:(.text+0x3984): undefined reference to `gbm_surface_create_with_modifiers'
CMakeFiles/flutter-pi.dir/src/compositor.c.o: In function `gbm_bo_get_drm_fb_id':
compositor.c:(.text+0x14c): undefined reference to `gbm_bo_get_modifier'
compositor.c:(.text+0x160): undefined reference to `gbm_bo_get_plane_count'
compositor.c:(.text+0x178): undefined reference to `gbm_bo_get_stride_for_plane'
compositor.c:(.text+0x1a4): undefined reference to `gbm_bo_get_offset'
CMakeFiles/flutter-pi.dir/src/user_input.c.o: In function `user_input_on_fd_ready':
user_input.c:(.text+0x71c): undefined reference to `libinput_device_touch_get_touch_count'
user_input.c:(.text+0xf84): undefined reference to `libinput_device_touch_get_touch_count'
CMakeFiles/flutter-pi.dir/src/plugins/omxplayer_video_player.c.o: In function `mgr_entry':
omxplayer_video_player.c:(.text+0x13a4): undefined reference to `sd_bus_match_signal'
collect2: error: ld returned 1 exit status
CMakeFiles/flutter-pi.dir/build.make:304: recipe for target 'flutter-pi' failed
make[2]: *** [flutter-pi] Error 1
CMakeFiles/Makefile2:82: recipe for target 'CMakeFiles/flutter-pi.dir/all' failed
make[1]: *** [CMakeFiles/flutter-pi.dir/all] Error 2
Makefile:135: recipe for target 'all' failed
make: *** [all] Error 2

ardera, 2021-07-01:
what distro are you using? if you run raspbian, is it possible you have a really old version of it?

goncalo-almeida-mc, 2021-07-02:
Kernel version is 4.19.57-v7+

goncalo-almeida-mc, 2021-07-02:
Just update to 5.10.46-v7+ (and rebooted), the problem keeps being the same

ardera, 2021-07-02:
thx, but what distro are you using? kernel version is good to know too, but distro is really what matters here.

goncalo-almeida-mc, 2021-07-02:
Maybe this helps?

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

ardera, 2021-07-02:
you're running raspbian stretch, which is pretty old and EOL since mid 2020. You can upgrade to Raspbian buster like this: https://www.raspberrypi.org/documentation/raspbian/updating.md (See the `Upgrading from Stretch to Buster` section)

goncalo-almeida-mc, 2021-07-02:
Installation went through!
Thank you very much @ardera !

closed 2021-07-02 by goncalo-almeida-mc