ardera/flutter-pi #185: drmModeAtomicCommit permission denied

goncalo-almeida-mc, 2021-07-05:
Flutter-pi stops here:

detected VideoCore IV as underlying graphics chip, and VC4 as the driver.
Reporting modified GL_EXTENSIONS string that doesn't contain non-working extensions.
flutter: Observatory listening on http://127.0.0.1:44135/AxKHeu4r10g=/
[modesetting] Could not commit atomic request. drmModeAtomicCommit: Permission denied
[compositor] Could not present frame. drmModeAtomicCommit: Permission denied
[modesetting] Could not commit atomic request. drmModeAtomicCommit: Permission denied
[compositor] Could not present frame. drmModeAtomicCommit: Permission denied

Flutter version is matching engine-binaries (2.2.2 both). Also tried to run flutter-pi with sudo and got the same output.
Any advice to advance past this?

ardera, 2021-07-05:
1. make sure no desktop environment (X.org, wayland) is running. Select `boot into console` in `raspi-config` (see the readme)
2. make sure no other instance of flutter-pi is running. It's possible flutter-pi is running even though you see the console on your display. To make sure no instance is running execute `killall flutter-pi`

goncalo-almeida-mc, 2021-07-06:
Fixed, thanks!

closed 2021-07-06 by goncalo-almeida-mc