ardera/flutter-pi #187: Could not resolve libflutter_engine procedure FlutterEngineNotifyDisplayUpdate. dlsym: No such file or directory

Miloune, 2021-07-06:
Hello,

I have an issue with flutter-pi and I found nothing about it on internet. 

```
pi@raspberrypi:~ $ sudo flutter-pi /home/pi/flutter_gallery/
[flutter-pi] WARNING: display didn't provide valid physical dimensions.
             The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
===================================
display mode:
  resolution: 720 x 480
  refresh rate: 60Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 2.0 Mesa 19.3.2"
  shading language version: "OpenGL ES GLSL ES 1.0.16"
  vendor: "Broadcom"
  renderer: "VC4 V3D 2.1"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_EGL_image GL_OES_depth_texture GL_AMD_performance_monitor GL_OES_packed_depth_stencil GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_separate_shader_objects GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_KHR_parallel_shader_compile GL_MESA_tile_raster_order "
===================================
[flutter-pi] Could not resolve libflutter_engine procedure FlutterEngineNotifyDisplayUpdate. dlsym: No such file or directory

```
What did I miss?

Regards

JerichoCross, 2021-08-23:
I am having the same problem. I have walked through the tutorial step-by-step but I get exactly the same error. I also tried it with the flutter starter project but I get the same thing. I would also like to know what I missed.

ardera, 2021-08-23:
`FlutterEngineNotifyDisplayUpdate` is a procedure that was introduced in flutter 2.2 (or 2.0, not sure).

Can you make sure you've installed the latest engine binaries from the [engine binaries repo](https://github.com/ardera/flutter-engine-binaries-for-arm) on your Pi? Could be that's the problem

JerichoCross, 2021-08-24:
No problem. I believe that I have installed the engine binaries as suggested but I will install them again to be sure. Is there an easy way to verify that they have actually installed? I'm a bit new to all this, sorry.

ardera, 2021-08-24:
no problem, you can maybe check the modification dates with `stat`

JerichoCross, 2021-09-01:
This may be a n00b kind of question but do I need to have any of the Flutter SDK deployed on the Pi?

ardera, 2021-09-01:
> Hi, sorry I got caught up with work. Once these have been installed, where would they be? I have the git clone in my home folder and I have run the install.sh as per the instructions. I only did this on Monday 23rd so I assume they are the latest.

they're installed in `/usr/lib`. (see the `install.sh`)

> This may be a n00b kind of question but do I need to have any of the Flutter SDK deployed on the Pi?

Nope, you don't need that and it's also not possible because the flutter SDK doesn't work on armhf.

JerichoCross, 2021-09-01:
Thanks. The libflutter_engine.so.debug and .release are both dated 2021-08-26 which would be consistent with me reinstalling them.

ardera, 2021-09-01:
Just to be sure, did you also run `git pull` inside the engine repo before installing? What does `git log -n 1` output for you? (when run inside the engine binaries directory)

JerichoCross, 2021-09-02:
Git pull returned 'Already up to date'.
Git log -n 1 gives:
`commit 1dcfddb52f838e2665f964803dd3d4b903ff2b29 (grafted, HEAD -> engine-binaries, origin/engine-binaries)
Author: Hannes Winkler <hanneswinkler2000@web.de>
Date:   Thu Jan 28 22:19:41 2021 +0100

    updated for flutter 1.22.6
    - now contains install script for simpler installation
    - updated the readme with new instructions
`

ardera, 2021-09-13:
I see, right now you've checked out the `engine-binaries` branch of the `flutter-pi` project. That's where the engine binaries used to be, but now they have their separate repo: https://github.com/ardera/flutter-engine-binaries-for-arm

Clone that instead & install, that should work

closed 2021-09-13 by ardera