ardera/flutter-pi #189: Splash Screen

suyulin, 2021-07-14:
hi,
How to add Splash Screen  ?

ardera, 2021-07-14:
that's not really in the scope of flutter-pi, but first of all:
- you need to disable the kernel console / show it on some other display
- raspbian uses [plymouth](https://freedesktop.org/wiki/Software/Plymouth/) for boot splash, so you can add a theme to that and set is as default
- enable the boot splash (not sure it's enabled by default)

there's a lot of resources about the individual steps online, I think you can google them all

suyulin, 2021-07-19:
> that's not really in the scope of flutter-pi, but:
> 
> * you need to disable the kernel console / show it on some other display
> * raspbian uses [plymouth](https://freedesktop.org/wiki/Software/Plymouth/) for boot splash, so you can add a theme to that and set is as default
> * enable the boot splash (not sure it's enabled by default)
> 
> there's a lot of resources about the individual steps online, I think you can google them all

thx

closed 2021-07-19 by suyulin

ardera, 2023-04-20:
@sdabet-orpheo What is shown on the display in the time between splash screen quit and first flutter frame, for you, right now? The kernel console? Just a black screen? (is the backlight active?)

I have no idea really how plymouth works and what makes it quit, but I can try looking into it a bit

sdabet-orpheo, 2023-04-20:
@ardera There's a black screen, then a login prompt. Then the first flutter frame appears a few seconds after.