ardera/flutter-pi #192: How do I use a custom plugin?

MateoSegura, 2021-07-23:
Hello, 

I managed to successfully build my app and run it on a Raspberry Pi 4.

I wanted to use [https://pub.dev/packages/flutter_libserialport](url) to talk to a serial port through my pi.

When I build the app.so, and run the app using flutter-pi --release, the following is outputted, before the app launches into a grey screen:

`flutter: Invalid argument(s): Failed to load dynamic library 'libserialport.so': libserialport.so: cannot open shared object file: No such file or directory`

I am aware this is because I don't have the .so file in the _src_ folder of flutter-pi.

However, I have NO IDEA how to build/get this file.

Any suggestions, tips, examples or explanations are much appreciated.

Thanks in advance.

Here's the entire output btw:

`flutter: Invalid argument(s): Failed to load dynamic library 'libserialport.so': libserialport.so: cannot open shared object file: No such file or directory
flutter: #0      _open (dart:ffi-patch/ffi_dynamic_library_patch.dart:11)
flutter: #1      new DynamicLibrary.open (dart:ffi-patch/ffi_dynamic_library_patch.dart:20)
flutter: #2      dylib (package:libserialport/src/dylib.dart:32)
flutter: #3      _SerialPortImpl.availablePorts.<anonymous closure> (package:libserialport/src/port.dart:230)
flutter: #4      Util.call (package:libserialport/src/util.dart:37)
flutter: #5      _SerialPortImpl.availablePorts (package:libserialport/src/port.dart:230)
flutter: #6      SerialPort.availablePorts (package:libserialport/src/port.dart:82)
flutter: #7      _ExampleAppState.initPorts.<anonymous closure> (package:flutter_libserialport_example/main.dart:39)
flutter: #8      State.setState (package:flutter/src/widgets/framework.dart:1088)
flutter: #9      _ExampleAppState.initPorts (package:flutter_libserialport_example/main.dart:39)
flutter: #10     _ExampleAppState.initState (package:flutter_libserialport_example/main.dart:33)
flutter: #11     StatefulElement._firstBuild (package:flutter/src/widgets/framework.dart:4711)
flutter: #12     ComponentElement.mount (package:flutter/src/widgets/framework.dart:4548)
flutter: #13     Element.inflateWidget (package:flutter/src/widgets/framework.dart:3611)
flutter: #14     Element.updateChild (package:flutter/src/widgets/framework.dart:3363)
flutter: #15     RenderObjectToWidgetElement._rebuild (package:flutter/src/widgets/binding.dart:1189)
flutter: #16     RenderObjectToWidgetElement.mount (package:flutter/src/widgets/binding.dart:1159)
flutter: #17     RenderObjectToWidgetAdapter.attachToRenderTree.<anonymous closure> (package:flutter/src/widgets/binding.dart:1104)
flutter: #18     BuildOwner.buildScope (package:flutter/src/widgets/framework.dart:2535)
flutter: #19     RenderObjectToWidgetAdapter.attachToRenderTree (package:flutter/src/widgets/binding.dart:1103)
flutter: #20     WidgetsBinding.attachRootWidget (package:flutter/src/widgets/binding.dart:937)
flutter: #21     WidgetsBinding.scheduleAttachRootWidget.<anonymous closure> (package:flutter/src/widgets/binding.dart:917)
flutter: #29     TickerFuture.whenCompleteOrCancel.thunk (package:flutter/src/scheduler/ticker.dart:407)
flutter: (elided 10 frames from class _RawReceivePortImpl, class _Timer, and dart:async)`

ardera, 2021-07-23:
It's actually just an `apt` package, you can install it using `sudo apt install libserialport`

(Though there's other plugins where you'd have to compile some `.so` yourself)

MateoSegura, 2021-07-23:
Thank you, 

`sudo apt install libserialport-dev`

solved my problem

closed 2021-07-23 by MateoSegura

trapeye, 2022-02-04:
I get same issue.. already install

`[ERROR:flutter/lib/ui/ui_dart_state.cc(209)] Unhandled Exception: Invalid argument(s): Failed to load dynamic library '/usr/local/bin/lib/libdart_vlc_plugin.so': /usr/local/bin/lib/libdart_vlc_plugin.so: cannot open shared object file: No such file or directory`