ardera/flutter-pi #194: DRM issue while trying to run app on raspberry pi

raul215, 2021-08-03:
I suddenly started facing the following issue.

```[flutter-pi] Could not create drmdev from device at "/dev/dri/card1". Continuing.
[flutter-pi] WARNING: display didn't provide valid physical dimensions.
             The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
===================================
display mode:
  resolution: 1920 x 1080
  refresh rate: 50Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 3.1 Mesa 19.3.2"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_primitive_bounding_box GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_EXT_texture_norm16 GL_KHR_context_flush_control GL_NV_image_formats GL_OES_draw_elements_base_vertex GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_EXT_buffer_storage GL_EXT_float_blend GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_shader_image_atomic GL_EXT_texture_compression_s3tc_srgb GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_MESA_framebuffer_flip_y GL_EXT_texture_query_lod "
===================================
flutter: Observatory listening on http://127.0.0.1:35911/0f253WQN62E=/
flutter: Directory: '/home/pi'
[modesetting] Could not commit atomic request. drmModeAtomicCommit: Permission denied
[compositor] Could not present frame. drmModeAtomicCommit: Permission denied
[modesetting] Could not commit atomic request. drmModeAtomicCommit: Permission denied
[compositor] Could not present frame. drmModeAtomicCommit: Permission denied
[modesetting] Could not commit atomic request. drmModeAtomicCommit: Permission denied
[compositor] Could not present frame. drmModeAtomicCommit: Permission denied```

raul215, 2021-08-03:
And after killing the flutter-pi processes manually I am getting the following:

`[modesetting] Could not set DRM client universal planes capable. drmSetClientCap: Operation not supported
[flutter-pi] Could not create drmdev from device at "/dev/dri/card0". Continuing.
[flutter-pi] WARNING: display didn't provide valid physical dimensions.
             The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
===================================
display mode:
  resolution: 1280 x 720
  refresh rate: 60Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 3.1 Mesa 19.3.2"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_primitive_bounding_box GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_EXT_texture_norm16 GL_KHR_context_flush_control GL_NV_image_formats GL_OES_draw_elements_base_vertex GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_EXT_buffer_storage GL_EXT_float_blend GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_shader_image_atomic GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_MESA_framebuffer_flip_y GL_EXT_texture_query_lod "
===================================
flutter: Observatory listening on http://127.0.0.1:35809/ZOC-Vh1qssE=/
`

ardera, 2021-08-03:
So it worked before? You probably did that but just to make sure, did you try rebooting?

Can you send me the complete output of `raspinfo`?

raul215, 2021-08-05:
Yes I did actually. 

`System Information
------------------

Raspberry Pi 4 Model B Rev 1.2 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2021-01-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 21090519d85bdaa1615d5d5057d37b09368ea5d2, stage4

Linux pod2 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
Revision	: b03112
Serial		: 100000000805ce7c
Model		: Raspberry Pi 4 Model B Rev 1.2
Throttled flag  : throttled=0x0
Camera          : supported=1 detected=0

Videocore information
---------------------

Feb 25 2021 12:11:39 
Copyright (c) 2012 Broadcom
version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start_x)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       30169392 7720884  21170340  27% /
devtmpfs          829636       0    829636   0% /dev
tmpfs             961732       0    961732   0% /dev/shm
tmpfs             961732   48940    912792   6% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             961732       0    961732   0% /sys/fs/cgroup
/dev/loop2         46976   46976         0 100% /snap/core18/2001
/dev/loop1        100608  100608         0 100% /snap/cmake/890
/dev/loop0        101376  101376         0 100% /snap/cmake/914
/dev/loop3         46976   46976         0 100% /snap/core18/2067
/dev/loop4        101376  101376         0 100% /snap/cmake/920
/dev/loop5         46976   46976         0 100% /snap/core18/2072
/dev/mmcblk0p1    258095   48791    209305  19% /boot
tmpfs             192344       0    192344   0% /run/user/1000

Filename				Type		Size		Used		Priority
/var/swap                               file		102396		0		-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: 1.20201210+nmu1
raspberrypi-sys-mods:
  Installed: 20210310
openbox:
  Installed: 3.6.1-8+rpt5
lxpanel:
  Installed: 0.10.0-2+rpt14
pcmanfm:
  Installed: 1.3.1-1+rpt26
rpd-plym-splash:
  Installed: 0.26

Networking Information
----------------------
[... 605 lines cut ...]
[  192.176331] usb 1-1.2: Product: USB Receiver
[  192.176348] usb 1-1.2: Manufacturer: Logitech
[  192.189445] input: Logitech USB Receiver as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2:1.0/0003:046D:C534.0001/input/input0
[  192.257099] hid-generic 0003:046D:C534.0001: input,hidraw0: USB HID v1.11 Keyboard [Logitech USB Receiver] on usb-0000:01:00.0-1.2/input0
[  192.268910] input: Logitech USB Receiver Mouse as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2:1.1/0003:046D:C534.0002/input/input1
[  192.269456] input: Logitech USB Receiver Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2:1.1/0003:046D:C534.0002/input/input2
[  192.336393] input: Logitech USB Receiver System Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2:1.1/0003:046D:C534.0002/input/input3
[  192.336981] hid-generic 0003:046D:C534.0002: input,hiddev96,hidraw1: USB HID v1.11 Mouse [Logitech USB Receiver] on usb-0000:01:00.0-1.2/input1
[  192.607838] logitech-djreceiver 0003:046D:C534.0001: hidraw0: USB HID v1.11 Keyboard [Logitech USB Receiver] on usb-0000:01:00.0-1.2/input0
[  192.929255] logitech-djreceiver 0003:046D:C534.0002: hiddev96,hidraw1: USB HID v1.11 Mouse [Logitech USB Receiver] on usb-0000:01:00.0-1.2/input1
[  193.012034] logitech-djreceiver 0003:046D:C534.0002: device of type eQUAD nano Lite (0x0a) connected on slot 1
[  193.015599] input: Logitech Wireless Keyboard PID:4023 Keyboard as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2:1.1/0003:046D:C534.0002/0003:046D:4023.0003/input/input6
[  193.017448] input: Logitech Wireless Keyboard PID:4023 Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2:1.1/0003:046D:C534.0002/0003:046D:4023.0003/input/input7
[  193.017864] input: Logitech Wireless Keyboard PID:4023 System Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2:1.1/0003:046D:C534.0002/0003:046D:4023.0003/input/input8
[  193.018349] hid-generic 0003:046D:4023.0003: input,hidraw2: USB HID v1.11 Keyboard [Logitech Wireless Keyboard PID:4023] on usb-0000:01:00.0-1.2/input1:1
[  193.028012] logitech-djreceiver 0003:046D:C534.0002: device of type eQUAD nano Lite (0x0a) connected on slot 2
[  193.030044] input: Logitech Wireless Mouse PID:4022 Mouse as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2:1.1/0003:046D:C534.0002/0003:046D:4022.0004/input/input13
[  193.030605] hid-generic 0003:046D:4022.0004: input,hidraw3: USB HID v1.11 Mouse [Logitech Wireless Mouse PID:4022] on usb-0000:01:00.0-1.2/input1:2
[  193.604393] input: Logitech Wireless Keyboard PID:4023 as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2:1.1/0003:046D:C534.0002/0003:046D:4023.0003/input/input17
[  193.604879] logitech-hidpp-device 0003:046D:4023.0003: input,hidraw2: USB HID v1.11 Keyboard [Logitech Wireless Keyboard PID:4023] on usb-0000:01:00.0-1.2/input1:1
[... cut ...] 1:00.0/usb1/1-1/1-1.2/1-1.2:1.1/0003:046D:C534.0002/0003:046D:4022.0004/input/input18
[  193.716459] logitech-hidpp-device 0003:046D:4022.0004: input,hidraw3: USB HID v1.11 Mouse [Logitech Wireless Mouse PID:4022] on usb-0000:01:00.0-1.2/input1:2
[  205.374422] usb 1-1.4: reset high-speed USB device number 5 using xhci_hcd
[  239.339048] logitech-hidpp-device 0003:046D:4023.0003: HID++ 2.0 device connected.
[  776.712380] v3d fec00000.v3d: MMU error from client L2T (0) at 0x21c1000, pte invalid`

ardera, 2021-08-06:
Okay nothing unusual, could you please send me the syslog with DRM debugging enabled?
- append `drm.debug=0x1ff` to `/boot/cmdline.txt`
- reboot
- run flutter-pi
- send me the full contents of `/var/log/syslog` (it's best to attach it as a file instead of pasting the contents in a code block)
- disable debugging again

Jollastro, 2021-09-08:
I'm also facing the same problem, but I have a few more details to work on:

When i run my app with flutter-pi on my raspberry 4 (with raspi os 64 bit) it works without that drm message.

When i run the same app, on the same rasp with same OS but using docker this time, then that message appears, exactly the same as the one of raul215, with the only difference that:

`Could not create drmdev from device at "/dev/dri/card1". Continuing.` is not card1 but it is card0 in my case.

Here is the configuration of the environment:

- The host is the same rasp with raspi os 64 bit.
- I made the necessary modification on the raspi-config,
- while for what regards the dependencies, those are scripted in the dockerfile:

`Dockerfile start`

FROM ubuntu

RUN apt -y update && apt -y install git && apt -y install wget 

WORKDIR /TOWER_DISPLAY

# STEP 1 DIPENDENZE
RUN git clone --depth 1 https://github.com/ardera/flutter-engine-binaries-for-arm.git engine-binaries && \ 
    cd engine-binaries && \
    ./install.sh

# STEP 2 DIPENDENZE
ENV TZ=Europe/Rome
RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone
RUN apt -y install cmake \
    libgl1-mesa-dev \
    libgles2-mesa-dev \
    libegl1-mesa-dev \
    libdrm-dev \
    libgbm-dev

RUN DEBIAN_FRONTEND='noninteractive' apt -y install ttf-mscorefonts-installer
RUN apt -y install fontconfig \
    libsystemd-dev \
    libinput-dev \
    libudev-dev \
    libxkbcommon-dev

#STEP 3 DIPENDENZE
RUN fc-cache
###########################
# COMPILING
RUN git clone https://github.com/ardera/flutter-pi
WORKDIR /TOWER_DISPLAY/flutter-pi/build
RUN cmake ..
RUN make -j `nproc`
RUN make install

WORKDIR /TOWER_DISPLAY

COPY ../build/flutter_assets ./flutter_assets

CMD flutter-pi ./flutter_assets

`Dockerfile end`

I'm waiting for a possible exlanation to this, thanks in advance.

ardera, 2021-09-13:
I've never used docker and never tried to run docker with flutter-pi, so not sure I can debug that; but I'll look into it

Cir0X, 2021-11-25:
@Jollastro How do you run the container? I'm running it with [`--priviliged` ](https://docs.docker.com/engine/reference/run/#runtime-privilege-and-linux-capabilities).

```sh
# build image
docker build -t flutter-pi .

# run container
docker run --rm --privileged --network host -v /run/udev:/run/udev:ro --name flutter-pi flutter-pi:latest
```

Depending on your setup, you probably don't need the `/run/udev` read only mount.

Jollastro, 2021-12-02:
I'm also running with `--priviliged` but it doesn't seem to solve the drmdev message

AtharvanDogra, 2022-09-15:
Hi, 
I'm facing the same issue, and I've done nothing unusual. Only followed the readme