ardera/flutter-pi #196: Building kernel snapshot for release mode works only without -Ddart.vm.product=true

Jollastro, 2021-09-08:
Hi,

I'm following the guide in the section:

**[ Building the app.so (for running your app in Release/Profile mode) ]**

I'm on Windows 10 and when i run the first command:

`dart` C:\Users\david\flutter\bin\cache\dart-sdk\bin\snapshots\frontend_server.dart.snapshot --sdk-root C:\Users\david\flutter\bin\cache\artifacts\engine\common\flutter_patched_sdk_product\ --target=flutter --aot --tfa -Ddart.vm.product=true --packages .packages --output-dill build\kernel_snapshot.dill --verbose --depfile build\kernel_snapshot.d package:tower_display/main.dart)`

**I receive 2 errors:**

**Error when reading '.vm.product=true': Impossible to find the specified file**

**Error: No 'main' method found.
Try adding a method named 'main' to your program.**

After a few tries I omitted the option `-Ddart.vm.product=true` and it worked without errors.
The rest of the guide worked fine and I managed to build the app.so file and run the app in release mode.

### The problem is that in my code I use the kReleaseMode constant, and without that option it doesn't work.

I can't explain why that option doesn't work, any help is much appreciated, thanks.

ardera, 2021-09-13:
are you sure you invoked the command correctly? From what it sounds like you maybe had a space between the `-Ddart` and `.vm.product=true` (so `-Ddart .vm.product=true`)

Jollastro, 2021-09-13:
I double-checked right now, and sadly yes, i'm invoking the command properly without the space.

ardera, 2021-09-13:
I just tried and for me that command is running fine. But when I add a space between `-Ddart` and `.vm.product=true`, I get the error you have:
```
C:\Users\hanne\devel\hello_world>dart C:\Users\hanne\devel\flutter_sdk\bin\cache\artifacts\engine\windows-x64\frontend_server.dart.snapshot --sdk-root C:\Users\hanne\devel\flutter_sdk\bin\cache\artifacts\engine\common\flutter_patched_sdk/ --target=flutter -Ddart.vm.profile=false -Ddart .vm.product=true --aot --tfa --packages C:\Users\hanne\devel\hello_world\.packages --output-dill C:\Users\hanne\devel\hello_world\.dart_tool\flutter_build\c4ecb8b638df8e36ee6b0d17b4921591\app.dill --depfile C:\Users\hanne\devel\hello_world\.dart_tool\flutter_build\c4ecb8b638df8e36ee6b0d17b4921591\kernel_snapshot.d package:hello_world/main.dart
result d555fb6b-1616-41ba-b2ac-9c0aade3bfde
Error: Error when reading '.vm.product=true': Das System kann die angegebene Datei nicht finden.

.vm.product=true: Error: No 'main' method found.
Try adding a method named 'main' to your program.
d555fb6b-1616-41ba-b2ac-9c0aade3bfde

C:\Users\hanne\devel\hello_world>
```

(With `Das System kann die angegebene Datei nicht finden.` meaning `the specified file can not be found`)

Jollastro, 2021-09-13:
Your answer made me tought about a syntax problem, so I tried with cmdline instead of powershell and....

**it worked!**

Still not working in the powershell 😕

Ahhhh.. the magic world of computers

closed 2021-09-13 by Jollastro