ardera/flutter-pi #197: The app flickers continuously on the rpi4

Jollastro, 2021-09-08:
Hi,

as said in the title the app running with flutter-pi flickers continuously on the rpi4.

I have 2 strange things:

- **on the rpi3 the app doesn't flicker!** (same configuration and same environment, except for the os which is 32 bit, but on the rpi4 it flickers also with the 32 bit os)
- if I append `drm.debug=0x1ff` to /boot/cmdline.txt, I really don't know why, but app flicker is dramatically reduced (almost non-existent).

The only problem to the drm.debug=0x1ff workaround is that with that the cpu works much much harder.
1 of the 4 core is almost always at 100% and also the other work much more.

Look forward to hering from you, thanks in advance.

ardera, 2021-09-13:
> as said in the title the app running with flutter-pi flickers continuously on the rpi4.

what kind of flickering? What app are you running? Does the flickering ocurr only in that specific app or other apps (like the gallery app, or the blank `flutter create`) too? Do you maybe have a video of that flickering?

> The only problem to the drm.debug=0x1ff workaround is that with that the cpu works much much harder.
> 1 of the 4 core is almost always at 100% and also the other work much more.

yeah and besides that, it'll also fill up your disk space pretty quickly

Jollastro, 2021-10-04:
I just tested with flutter gallery app, and it doesn't flicker with such application.

In my app, it flickers (bottom half of the screen), and especially when i use the touchscreen.

I attach a video

https://user-images.githubusercontent.com/17902313/135825822-934f96d6-7841-45c7-b4c0-eb16b648f432.mp4

ardera, 2021-10-05:
can you try with the full-KMS driver? that fixed some flickering in the past (though that flickering looked a bit different)

- change `dtoverlay=vc4-fkms-v3d` to `dtoverlay=vc4-kms-v3d` in `/boot/config.txt`

other than that, it could also be a flutter / skia or graphics driver issue

Jollastro, 2021-10-05:
It works!! No more flickering!!

Thank you so much!!

closed 2021-10-05 by Jollastro