ardera/flutter-pi #199: Can't consume messages

goncalo-almeida-mc, 2021-09-16:
I'm running dart_amqp libs to enable RabbitMQ communication on my Flutter app.

I don't know for sure if the faulty side is on flutter-pi but when running on the main machine it works and Flutter consumes the Rabbit messages.
I just tried to validate the app on my RPI and turns out the screen doesn't update according to the messages that are being sent, meaning it isn't consuming the messages.

Internally on RPI, I can send and receive the messages, so maybe it is some issue from flutter-pi?

Version: 2.2.3 on both ends

ardera, 2021-09-16:
Sounds very unlikely to be a flutter-pi issue.
Flutter-pi just provides flutter with graphics output / user input (+ some other stuff, but nothing related to what you're doing). I've also never heard of RabbitMQ or dart_ampq, so I'm afraid I can't help you with that

If you have a minimal code sample that reproduces the issue, which works on other devices but not on pi using flutter-pi I'd be happy to look at it, but generally speaking, this doesn't sound like a flutter-pi issue

goncalo-almeida-mc, 2021-09-16:
Thanks!

To install the RMQ server:
`$ sudo apt-get install rabbitmq-server`
`$ sudo systemctl enable rabbitmq-server`

To start it:
`$ sudo service rabbitmq-server start`
`$ sudo rabbitmqctl start_app`

To stop it:
`$ sudo rabbitmqctl stop_app`
`$ sudo service rabbitmq-server stop`

You can try with the attached `zipped python scripts` to send (and receive, just to check it is actually being sent) and the code below to consume on Flutter, I think I didn't forget anything :)
[send-receive.tar.gz](https://github.com/ardera/flutter-pi/files/7177213/send-receive.tar.gz)

`- - Flutter code - -`
import "package:dart_amqp/dart_amqp.dart";

void recvMsg() async {
    
    String queueName = 'hello';
    List<String> routingKeys = [queueName];
    Client client = Client();
    // auto-connect to localhost:5672 using guest credentials
    Channel channel = await client.channel();
    Exchange exchange = await channel.exchange(queueName, ExchangeType.DIRECT);
    Consumer consumer = await exchange.bindQueueConsumer(queueName, routingKeys);

    consumer.listen((AmqpMessage message) {
      // Get the payload as a string
      print(" [x] Received string: ${message.payloadAsString}");

      /*
      // Or unserialize to json
      print(" [x] Received json: ${message.payloadAsJson}");

      // Or just get the raw data as a Uint8List
      print(" [x] Received raw: ${message.payload}");
      */

      // The message object contains helper methods for
      // replying, ack-ing and rejecting
      message.ack();
    });
  }

goncalo-almeida-mc, 2021-09-16:
I store that `${message.payloadAsString}` on a String variable that I am displaying on screen, so I know for sure it isn't being consume (as Pi only shows screen app).

Through ssh it also isn't printing any message on terminal.