ardera/flutter-pi #201: Use specific display output

shahinv, 2021-10-04:
I could not find any specific way to control display output (DSI/HDMI0/HDMI1)?
As far as I know display output could be controlled through dispmanx display id:

0 = Default display
2 = HDMI0 (Pi4)
4 = Touchscreen 
5 = HDMI (Pi3-)
6 = Select non-default display
7 = HDMI1 (Pi4)

ardera, 2021-10-05:
Yeah right now flutter-pi just uses the first display it finds, though it uses KMS not dispmanx (dispmanx is deprecated AFAIK)

Right now it's pretty easy to workaround, so lower priority, but once I work on it the Qt EGLFS config file (https://doc.qt.io/qt-5/embedded-linux.html) could serve as a template

shahinv, 2021-10-05:
Thank you for the help.

I take a look at the source code and could not recognize where display selection happen.
Can you share a tip on the workaround?

ardera, 2021-10-05:
Sure, output selection happens here: https://github.com/ardera/flutter-pi/blob/e60ccd19d753674dc9481d87a073dd7e9c9bf4f9/src/flutter-pi.c#L1267
    
.    

That loop will just select the first connected connector. If you want a specific connector like HDMI you can just skip all non-HDMI connectors.

Though I don't know how to differentiate HDMI0 and HDMI1 via KMS. Though the [`modetest` utility from libdrm](https://gitlab.freedesktop.org/mesa/drm/-/tree/main) is really useful for finding that out.

You can maybe differentiate them via the [PATH property](https://www.kernel.org/doc/html/latest/gpu/drm-kms.html#standard-connector-properties), or via the `connector_type_id`. (That's at least how [modeset determines the connector name](https://gitlab.freedesktop.org/mesa/drm/-/blob/main/tests/modetest/modetest.c#L664))

For example
```C
for_each_connector_in_drmdev(flutterpi.drm.drmdev, connector) {
  if (connector->connector->connection == DRM_MODE_CONNECTED && connector->connector->connector_type == DRM_MODE_CONNECTOR_HDMIA) {
    // only update the physical size of the display if the values
    //   are not yet initialized / not set with a commandline option
    if ((flutterpi.display.width_mm == 0) || (flutterpi.display.height_mm == 0)) {
      if ((connector->connector->connector_type == DRM_MODE_CONNECTOR_DSI) &&
          (connector->connector->mmWidth == 0) &&
          (connector->connector->mmHeight == 0))
      {
        // if it's connected via DSI, and the width & height are 0,
        //   it's probably the official 7 inch touchscreen.
        flutterpi.display.width_mm = 155;
        flutterpi.display.height_mm = 86;
      } else if ((connector->connector->mmHeight % 10 == 0) &&
	      (connector->connector->mmWidth % 10 == 0)) {
        // don't change anything.
      } else {
        flutterpi.display.width_mm = connector->connector->mmWidth;
        flutterpi.display.height_mm = connector->connector->mmHeight;
      }
    }
    
    break;
  }
}
```

Taha-Firoz, 2022-08-13:
> Yeah right now flutter-pi just uses the first display it finds, though it uses KMS not dispmanx (dispmanx is deprecated AFAIK)
>
> Right now it's pretty easy to workaround, so lower priority, but once [...]

🤣 I had the same problem on my pi and I was only able to debug it by looking into QT's EGLFS config file. This is exactly what I want to add into this embedder, a config file that's an alternative to the command line params. If you're interested in having something like that merged into the project I'll be more than happy to help because I've already extensively studied QT's KMS config file implementation.

ardera, 2022-08-19:
@Taha-Firoz Sounds awesome! Though if you do that, I think it's a good idea to write that stuff on top of the `feature/compositor-ng` branch. That's a huge refactor/rewrite of the display infra I've been working on, I just want to get it up to feature parity with master branch (mouse support is still missing for example) before I merge it. Don't want to add to many changes to `master` now since all that has to be added to `feature/compositor-ng` as well.

Taha-Firoz, 2022-08-20:
Your design seems to be a lot similar to [ivi-homescreen json configuration file](https://github.com/toyota-connected/ivi-homescreen/blob/c3abadee678da430203bc23b2a11c7718d857ad5/README.md?plain=1#L139)  implementation. Since Joel is interested in eventually pulling flutter-pi in for ivi-homescreen's DRM implementation do you think we should just take the implementation from there and modify it. 

Ivi-homescreen does use paths to automatically pick up config files from instead of passing their paths in as command line parameters, I think the simplest approach should be taken. Only allowing either a config file (path given via command line) or a command line configurations to used at a time, this would allow a majority of flutter-pi's command line argument processing to stay the same and we can add in the config file module separately.

I do enjoy QT's implementation of a kms conf file, very to the point
```
{
    "device": "/dev/dri/card1",
    "hwcursor" : false,
    "outputs": [
        { "name": "DSI1", "mode": "800x1280", "primary": true },
        { "name": "Composite1", "mode": "off" } 
    ]
}
```

ardera, 2022-08-20:
> Your design seems to be a lot similar to [ivi-homescreen json configuration [...]

If you can reuse code that's great, but not sure we can take from ivi-homescreen bc Apache 2.0 license is slightly more restrictive than MIT AFAIK. Other than that, C-code should be preferred when possible (though if you want to write in C++ we can talk about that 😄). We can try to stay compatible to ivi-homescreens config though so pulling in flutter-pi is not that much of a breaking change.

> Only allowing either a config file (path given via command line) or a command line configurations to used at a time, this would allow a majority of flutter-pi's command line argument processing to stay the same and we can add in the config file module separately.

I think that approach is fine at the start, but IMO ultimately the cmdline arg processing should be modified to rely on the new configuration module as well. It's not like that's a huge change, the cmdline processing function is only a few lines :) And I much rather have a single, well-used codepath.

Taha-Firoz, 2022-08-20:
>  Other than that, C-code should be preferred when possible (though if you want to write in C++ we can talk about that 😄)
 I prefer Rust but since we aren't going down that rabbit hole I'll try my hand with C, the only annoying part might be working with json in C instead of C++.
 
> If you can reuse code that's great, but not sure we can take from ivi-homescreen bc Apache 2.0 license is slightly more restrictive than MIT AFAIK.
Yes I saw that but I'm sure our C implementation will be different enough not get us into any trouble.

What are you using for parsing json in C, futter-elinux is using Tencent rapidJson.

ardera, 2022-08-20:
> What are you using for parsing json in C, futter-elinux is using Tencent rapidJson.

I'm using `jsmn.h`, it's a header-only library, though tbh calling it a parser is probably an overstatement, it's more like a tokenizer.

But if you feel like using something nicer for json handling, I'd be open for that as well.

Taha-Firoz, 2022-08-20:
Basically I've used flutter-pi as the first embedder for my project and in between moved over to flutter-elinux where I seem to be having a similar conversation on. There are certain problems with flutter-elinux that have come up which is pushing to move back to flutter-pi however the lack of control over DKMS for both projects is a major problem for me which is why I want to work on it. 

However as a disclaimer since I've spent a lot of time going through and working with the flutter-elinux code base, I am heavily influenced by it.

Taha-Firoz, 2022-08-23:
Is this library [fine](https://github.com/DaveGamble/cJSON) and I'll use [this as reference](https://github.com/toyota-connected/ivi-homescreen/blob/c3abadee678da430203bc23b2a11c7718d857ad5/shell/configuration/configuration.h)?

ardera, 2022-08-28:
@Taha-Firoz yes, that's fine by me. (Both things.) I could also migrate the platform channel code to cJSON too, seems nicer than jsmn.

Taha-Firoz, 2022-08-28:
Checkout [my fork](https://github.com/Taha-Firoz/flutter-pi/tree/feature/compositor-ng) should I create a PR to get a review on what I've done or should we coordinate on issues for each review?

I just added basic config file parsing based on the template you sent, tbh I didn't really do anything I just pasted the example and copilot wrote the whole thing for me 😅

ardera, 2022-08-28:
You can create a draft PR if you want, so we can coordinate the work better :) Or you get your prototype going first, how you prefer.

Looks pretty good already tbh, it's nice that you follow the rest of the flutter-pi codestyle :D Did copilot really write all that? That's amazing, I should get it too

Btw one small thing I've spotted: https://github.com/Taha-Firoz/flutter-pi/blob/538fc927c162586cf0d6d98f84b321f3fd1d3dc3/src/config.c#L179-L188

you can use
```C
char *newpath;
ok = asprintf(&newpath, "%s/.config/", path);
```

there, it'll do the correct allocation for you. (May need to `#define _GNU_SOURCE` for that) (Though you can't do `asprintf(&path, "%s/.config/", path);` because the args are marked as `restrict`)

Mostly I try to do any declarations at the top and sort them in decending length, so
```C
struct configuration *c;
char *path;
int ok;
```

That's probably a bit pedantic but I think it helps with readability.

th0f, 2025-03-26:
Bumping this old thread, but I would be very interested in a similar solution. Has anything from this idea ever been implemented?

Taha-Firoz, 2025-03-26:
I started an implementation but unfortunately I haven't been able to make time in my schedule for this, also given that high standard for the implementation didn't make it very straight forward to add. We ended up completing our implementation use Sony FlutterElinux but I wouldn't recommend going with that right now because the maintainer left Sony for Nvidia so it's basically on life support.