ardera/flutter-pi #207: Touch of the official raspbery pi 7 inch display is not working

PySchilling, 2021-11-10:
I am having problems getting the touchscreen of the official raspbery pi 7 inch display to work having a Raspberry Pi 4 8GB with tha latest updates.

I am using Flutter 2.5.2 with the latest commits from the repositories (flutter-pi and flutter-engine).
The graphical display is fine and with the keyboard I can run the application. But with the touchscreen I can't use the gui.

I have already checked with evtest if events arrive and it looks like the events arrive:
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:	NOVATEK USB Keyboard
/dev/input/event1:	NOVATEK USB Keyboard System Control
/dev/input/event2:	NOVATEK USB Keyboard Consumer Control
/dev/input/event3:	NOVATEK USB Keyboard
/dev/input/event4:	vc4
/dev/input/event5:	vc4
/dev/input/event6:	generic ft5x06 (00)
Select the device event number [0-6]: 6
Input driver version is 1.0.1
Input device ID: bus 0x18 vendor 0x0 product 0x0 version 0x0
Input device name: "generic ft5x06 (00)"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value    750
      Min        0
      Max      799
    Event code 1 (ABS_Y)
      Value    409
      Min        0
      Max      479
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max        4
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max      799
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max      479
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max    65535
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
Testing ... (interrupt to exit)
Event: time 1636542629.512494, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 352
Event: time 1636542629.512494, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 288
Event: time 1636542629.512494, type 3 (EV_ABS), code 0 (ABS_X), value 352
Event: time 1636542629.512494, type 3 (EV_ABS), code 1 (ABS_Y), value 288
Event: time 1636542629.512494, -------------- SYN_REPORT ------------
Event: time 1636542629.572485, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 354
Event: time 1636542629.572485, type 3 (EV_ABS), code 0 (ABS_X), value 354
Event: time 1636542629.572485, -------------- SYN_REPORT ------------
Event: time 1636542629.602519, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 359
Event: time 1636542629.602519, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 293
Event: time 1636542629.602519, type 3 (EV_ABS), code 0 (ABS_X), value 359
Event: time 1636542629.602519, type 3 (EV_ABS), code 1 (ABS_Y), value 293
Event: time 1636542629.602519, -------------- SYN_REPORT ------------
Event: time 1636542629.632488, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 372
Event: time 1636542629.632488, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 327
Event: time 1636542629.632488, type 3 (EV_ABS), code 0 (ABS_X), value 372
Event: time 1636542629.632488, type 3 (EV_ABS), code 1 (ABS_Y), value 327
Event: time 1636542629.632488, -------------- SYN_REPORT ------------
[... 62 lines cut ...]
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_device EGL_EXT_platform_wayland EGL_KHR_platform_wayland EGL_EXT_platform_x11 EGL_KHR_platform_x11 EGL_MESA_platform_gbm EGL_KHR_platform_gbm EGL_MESA_platform_surfaceless"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 3.1 Mesa 20.3.5"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_draw_instanced GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_pack_reverse_row_order GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_NV_pixel_buffer_object GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_primitive_bounding_box GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_NV_image_formats GL_OES_draw_elements_base_vertex GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_EXT_buffer_storage GL_EXT_float_blend GL_EXT_geometry_point_size GL_EXT_geometry_shader GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_geometry_point_size GL_OES_geometry_shader GL_OES_shader_image_atomic GL_EXT_texture_compression_s3tc_srgb GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_EXT_EGL_image_storage GL_MESA_framebuffer_flip_y GL_EXT_texture_query_lod "
===================================
flutter: Observatory listening on http://127.0.0.1:36371/Kb9ApP3ev-w=/
[compositor] Could not move cursor to front. Mouse cursor may be invisible. drmdev_plane_get_max_zpos_value: Invalid argument
[compositor] GPU does not supported the desired HW plane order.
             Some UI layers may be invisible.
[ERROR:flutter/lib/ui/ui_dart_state.cc(209)] Unhandled Exception: FormatException: Unexpected extension byte (at offset 3)
#0      _Utf8Decoder.convertSingle (dart:convert-patch/convert_patch.dart:1788:7)
#1      Utf8Decoder.convert (dart:convert/utf.dart:318:42)
#2      StringCodec.decodeMessage (package:flutter/src/services/message_codecs.dart:44:25)
#3      JSONMessageCodec.decodeMessage (package:flutter/src/services/message_codecs.dart:99:44)
#4      JSONMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:147:54)
#5      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:156:18)
<asynchronous suspension>
#6      SystemChrome.setApplicationSwitcherDescription (package:flutter/src/services/system_chrome.dart:398:5)
<asynchronous suspension>

ardera, 2021-11-10:
can you try running without the `-i (path here)` argument? I thought the sentence `In most cases, there's no need to specify this option.` in the usage and not providing an example invocation in the usage would be enough to stop people from using it 😄

if it still doesn't work without it, are you running the latest pi OS based on debian 11? bc I haven't tested with that yet

PySchilling, 2021-11-10:
Thanks for your quick response!

Same problem withput ```-i```

> if it still doesn't work without it, are you running the latest pi OS based on debian 11? bc I haven't tested with that yet

Yes, I use the latest image

> EDIT: That compositor error is interesting though. Maybe they switched to the full KMS driver now. The FormatException also shouldn't be happening, I need to look into that too.

I switched to the full KMS driver

I will try with a previous Raspian OS image.

Thanks!

PySchilling, 2021-11-10:
Do it with 2021-05-07-raspios-buster works fine!!

closed 2021-11-10 by PySchilling

ardera, 2021-11-30:
This is a kernel bug, was fixed recently: https://github.com/raspberrypi/linux/pull/4736
That's not in any stable release though, so you need to use `rpi-update` to get the fix.

Touch will work after that, but you'll get spurious touch events at the bottom right of the screen (see https://github.com/raspberrypi/linux/issues/4749)

(Just noting in case anyone else encounters this problem)