ardera/flutter-pi #208: Y Scale

ali77gh, 2021-11-23:
Hi.
First of all thank you for making this project public, its really helping.
<hr>
The problem is:<br>
I have a 1280px x 720px display (ratio is  16x9).<br>
but its 256mm x 163mm (ratio is 14x9).<br>
<br>
So when i try to render a square it actually shows a rectangle (its taller then what it should be).

Then i tried to run program with -d switch:
```bash 
flutter-pi -d "256, 163" ./fps_assets/
```
It actually not changing ratio and just effects on widget sizes, and i tried to randomly increase and decrease sizes and ratio is not changing.

ardera, 2021-11-24:
The issue is that your screen has non-square pixels (the official pi screen has too). That's actually an open problem in flutter right now: https://github.com/flutter/flutter/issues/71865

Theoretically I could add a "framebuffer size" argument to flutter-pi, so you can set some custom framebuffer size (that's a bit shorter/longer in the y direction) and then scale to framebuffer to fit the screen accordingly, but that's not optimal. text & edges generally will probably look blurry

Best way to fix this is actually in the flutter app. See https://github.com/flutter/flutter/issues/32115#issuecomment-551336545 for example. You can inject that widget into your app using the `MaterialApp.builder` property. (It's not perfect though, see the replies in that issue)

You can change this to only scale in the y direction, then it should work

ali77gh, 2021-11-27:
@ardera
Thanks for your replay. 

I fixed this by using this [class](https://gist.github.com/ali77gh/bd01162e2b0f8c5b4794ef5792818831).
It actually using Scale Y but you should provide physical width and height of your screen in millimeters.

And it my fixed in flutter some day [here](https://github.com/flutter/flutter/issues/71865).

closed 2021-11-27 by ali77gh

Arka2009, 2022-03-20:
@ali77gh , First of all thanks for the invaluable code snipet.
I am unable to compile your code. I encounter error "Target URI doesn't exist: package:fps/build_conf.dart".

Can you help me out how, to fix this?

Below is my pubspec.yaml
--------------------------------------------------------------------------------------------------------------------------
name: test2_web
description: A new Flutter project.
version: 1.0.0+1

environment:
  sdk: ">=2.12.0 <3.0.0"
dependencies:
  flutter:
    sdk: flutter
  cupertino_icons: ^1.0.2
  flutter_gpiod: ^0.4.0-nullsafety
  flutter_fps: ^0.1.1

dev_dependencies:
  flutter_test:
    sdk: flutter

ali77gh, 2022-03-20:
Hello @Arka2009.

1. Remove the import line

2. Replace physicalScreenSize with your screen size(in millimeters)

I hope it helps🙂.