ardera/flutter-pi #210: Error compiling: DRM_FORMAT_MOD_LINEAR’ undeclared (first use in this function)

jairoareyes, 2021-11-27:
Hi, I'm trying to build and compile the project, but it shows me this error: 

```
/home/pi/flutter-pi/src/flutter-pi.c:1464:27: error: ‘DRM_FORMAT_MOD_LINEAR’ undeclared (first use in this function)
  flutterpi.gbm.modifier = DRM_FORMAT_MOD_LINEAR;
                           ^~~~~~~~~~~~~~~~~~~~~
/home/pi/flutter-pi/src/flutter-pi.c:1464:27: note: each undeclared identifier is reported only once for each function it appears in

```

My setup is a Rpi 2 Model B with Raspbian GNU/Linux 9.9 (stretch) (Linux raspberrypi 4.19.50-v7+ #1234 SMP Thu Jun 13 11:06:37 BST 2019 armv7l GNU/Linux)

What should I do? (not suggest upgrade the OS, please)

ardera, 2021-11-27:
Seems like `DRM_FORMAT_MOD_LINEAR` was only introduced in a more recent linux kernel.

Try running cmake with `-DCMAKE_C_FLAGS="-DDRM_FORMAT_MOD_LINEAR=0"`

jairoareyes, 2021-11-27:
Now I have another issues compiling: 
```
/home/pi/flutter-pi/src/compositor.c: In function ‘rendertarget_gbm_present’:
/home/pi/flutter-pi/src/compositor.c:376:86: error: ‘DRM_MODE_ROTATE_0’ undeclared (first use in this function)
 e_supports_setting_rotation_value(atomic_req->drmdev, drm_plane_id, DRM_MODE_ROTATE_0, &supported);
...
 ev_plane_supports_setting_rotation_value(req->drmdev, drm_plane_id, DRM_MODE_ROTATE_0 | DRM_MODE_REFLECT_Y, &supported);

```

ardera, 2021-11-28:
Add these additional flags to `CMAKE_C_FLAGS`:
- `-DDRM_MODE_ROTATE_0=1`,
- `-DDRM_MODE_ROTATE_90=2`,
- `-DDRM_MODE_ROTATE_180=4`,
- `-DDRM_MODE_ROTATE_270=8`,
- `-DDRM_MODE_REFLECT_X=16` and
- `-DDRM_MODE_REFLECT_Y=32`

(maybe you need more, if yes you can find the definitions for any `DRM_MODE_...` macros in the linux kernel source here: https://elixir.bootlin.com/linux/latest/source/include/uapi/drm/drm_mode.h#L178)

I don't really support raspbian stretch, even if I add fallbacks for these macros inside the flutter-pi source code, things will probably break again later on because I'm not testing on stretch

jairoareyes, 2021-11-29:
Hi, I'm having this issue now.  

``` bash
CMakeFiles/flutter-pi.dir/src/flutter-pi.c.o: In function `init_display':
flutter-pi.c:(.text+0x3160): undefined reference to `drmGetDevices2'
flutter-pi.c:(.text+0x3934): undefined reference to `drmCrtcGetSequence'
flutter-pi.c:(.text+0x3a34): undefined reference to `gbm_surface_create_with_modifiers'
CMakeFiles/flutter-pi.dir/src/compositor.c.o: In function `gbm_bo_get_drm_fb_id':
compositor.c:(.text+0x14c): undefined reference to `gbm_bo_get_modifier'
compositor.c:(.text+0x160): undefined reference to `gbm_bo_get_plane_count'
compositor.c:(.text+0x178): undefined reference to `gbm_bo_get_stride_for_plane'
compositor.c:(.text+0x1a4): undefined reference to `gbm_bo_get_offset'
CMakeFiles/flutter-pi.dir/src/user_input.c.o: In function `user_input_on_fd_ready':
user_input.c:(.text+0x9c4): undefined reference to `libinput_device_touch_get_touch_count'
user_input.c:(.text+0x1094): undefined reference to `libinput_device_touch_get_touch_count'
CMakeFiles/flutter-pi.dir/src/plugins/omxplayer_video_player.c.o: In function `mgr_entry':
omxplayer_video_player.c:(.text+0x13a4): undefined reference to `sd_bus_match_signal'
```
I notice that the library is in the path:

``` bash
pi@raspberrypi:/usr/lib/arm-linux-gnueabihf $ ls -l | grep libGL
lrwxrwxrwx  1 root root       14 Aug  4  2017 libGL.so -> libGL.so.1.2.0
lrwxrwxrwx  1 root root       14 Aug  4  2017 libGL.so.1 -> libGL.so.1.2.0
-rw-r--r--  1 root root   400092 Aug  4  2017 libGL.so.1.2.0
lrwxrwxrwx  1 root root       21 Aug  4  2017 libGLESv1_CM.so.1 -> libGLESv1_CM.so.1.1.0
-rw-r--r--  1 root root    17704 Aug  4  2017 libGLESv1_CM.so.1.1.0
lrwxrwxrwx  1 root root       18 Aug  4  2017 libGLESv2.so -> libGLESv2.so.2.0.0
lrwxrwxrwx  1 root root       18 Aug  4  2017 libGLESv2.so.2 -> libGLESv2.so.2.0.0
-rw-r--r--  1 root root    38184 Aug  4  2017 libGLESv2.so.2.0.0
lrwxrwxrwx  1 root root       16 Sep 18  2016 libGLEW.so.2.0 -> libGLEW.so.2.0.0
-rw-r--r--  1 root root   528308 Sep 18  2016 libGLEW.so.2.0.0
lrwxrwxrwx  1 root root       15 Sep 19  2015 libGLU.so.1 -> libGLU.so.1.3.1
-rw-r--r--  1 root root   358228 Sep 19  2015 libGLU.so.1.3.1
```

Is it really necessary to update the OS version?

ardera, 2021-11-29:
You can try compiling mesa, libinput and libsystemd yourself and linking against those self-build libraries instead, that should work.

closed 2022-03-18 by ardera