ardera/flutter-pi #212: SPI  touchscreen not working [ADS7846]

Marouen-Hammami, 2021-12-30:
Hi, 
I'm having trouble using the touchscreen on RPi 4. I started the app using `flutter-pi -i "/dev/input/event5" -d "154,86" /test.app` .The app starts fine and works well with the mouse, but the touchscreen make any action. 
This is what i get on the terminal when i start the app through ssh.

###### [modesetting] Could not set DRM client universal planes capable. drmSetClientCap: Opération non supportée
###### [flutter-pi] Could not create drmdev from device at "/dev/dri/card1". Continuing.
###### [flutter-pi] WARNING: display has non-square pixels. Non-square-pixels are not supported by flutter.
###### =================================== 
###### display mode:
######   resolution: 1024 x 600
######   refresh rate: 60Hz
######   physical size: 154mm x 86mm
######   flutter device pixel ratio: 1,749829
###### ===================================
###### EGL information:
######   version: 1.4
######   vendor: "Mesa Project"
######   client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_device EGL_EXT_platform_wayland EGL_KHR_platform_wayland EGL_EXT_platform_x11 EGL_KHR_platform_x11 EGL_MESA_platform_gbm EGL_KHR_platform_gbm EGL_MESA_platform_surfaceless"
 ###### display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
###### ===================================
###### OpenGL ES information:
######  version: "OpenGL ES 3.1 Mesa 20.3.5"
######  shading language version: "OpenGL ES GLSL ES 3.10"
######  vendor: "Broadcom"
######  renderer: "V3D 4.2"
######  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_compression_s3tc GL_EXT_texture_compression_dxt1 GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_draw_instanced GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_pack_reverse_row_order GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_NV_pixel_buffer_object GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_primitive_bounding_box GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_NV_image_formats GL_OES_draw_elements_base_vertex GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_EXT_buffer_storage GL_EXT_float_blend GL_EXT_geometry_point_size GL_EXT_geometry_shader GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_geometry_point_size GL_OES_geometry_shader GL_OES_shader_image_atomic GL_EXT_texture_compression_s3tc_srgb GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_EXT_EGL_image_storage GL_MESA_framebuffer_flip_y GL_EXT_texture_query_lod "
###### ===================================
###### flutter: Observatory listening on http://127.0.0.1:39413/PJaqwLDaxqY=/
###### [ERROR:flutter/lib/ui/ui_dart_state.cc(209)] Unhandled Exception: FormatException: Unexpected extension byte (at offset 3)
###### #0      _Utf8Decoder.convertSingle (dart:convert-patch/convert_patch.dart:1789:7)
###### #1      Utf8Decoder.convert (dart:convert/utf.dart:318:42)
###### #2      StringCodec.decodeMessage (package:flutter/src/services/message_codecs.dart:44:25)
###### #3      JSONMessageCodec.decodeMessage (package:flutter/src/services/message_codecs.dart:99:44)
###### #4      JSONMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:147:54)
###### #5      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:167:18)
###### <asynchronous suspension>
###### #6      SystemChrome.setApplicationSwitcherDescription (package:flutter/src/services/system_chrome.dart:394:5)
###### <asynchronous suspension>
_______________________________________________________
And here's the evtest log when i tap on the screen :

###### Input driver version is 1.0.1
###### Input device ID: bus 0x0 vendor 0x0 product 0x0 version 0x0
###### Input device name: "ADS7846 Touchscreen"
###### Supported events:
######  Event type 0 (EV_SYN)
######  Event type 1 (EV_KEY)
######    Event code 330 (BTN_TOUCH)
######  Event type 3 (EV_ABS)
######    Event code 0 (ABS_X)
######      Value   2418
######      Min        0
######      Max     4095
######    Event code 1 (ABS_Y)
######      Value   1728
######      Min        0
######      Max     4095
######    Event code 24 (ABS_PRESSURE)
######      Value      0
######      Min        0
######      Max      255
###### Properties:
###### Testing ... (interrupt to exit)
###### Event: time 1640882423.626831, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
###### Event: time 1640882423.626831, type 3 (EV_ABS), code 0 (ABS_X), value 1946
###### Event: time 1640882423.626831, type 3 (EV_ABS), code 1 (ABS_Y), value 1501
###### Event: time 1640882423.626831, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 253
###### Event: time 1640882423.626831, -------------- SYN_REPORT ------------
###### Event: time 1640882423.646784, type 3 (EV_ABS), code 1 (ABS_Y), value 1486
###### Event: time 1640882423.646784, -------------- SYN_REPORT ------------
[... 25 lines cut ...]
###### Event: time 1640882424.796727, -------------- SYN_REPORT ------------
###### Event: time 1640882424.813583, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
###### Event: time 1640882424.813583, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 0
###### Event: time 1640882424.813583, -------------- SYN_REPORT ------------
###### Event: time 1640882425.616826, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
###### Event: time 1640882425.616826, type 3 (EV_ABS), code 0 (ABS_X), value 1763
###### Event: time 1640882425.616826, type 3 (EV_ABS), code 1 (ABS_Y), value 1182
###### Event: time 1640882425.616826, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 253
###### Event: time 1640882425.616826, -------------- SYN_REPORT ------------
###### Event: time 1640882425.636822, type 3 (EV_ABS), code 0 (ABS_X), value 1761
###### Event: time 1640882425.636822, type 3 (EV_ABS), code 1 (ABS_Y), value 1149
###### Event: time 1640882425.636822, -------------- SYN_REPORT ------------
###### Event: time 1640882425.656829, type 3 (EV_ABS), code 0 (ABS_X), value 1762
###### Event: time 1640882425.656829, type 3 (EV_ABS), code 1 (ABS_Y), value 1221
###### Event: time 1640882425.656829, -------------- SYN_REPORT ------------
###### Event: time 1640882425.676810, type 3 (EV_ABS), code 0 (ABS_X), value 1732
###### Event: time 1640882425.676810, type 3 (EV_ABS), code 1 (ABS_Y), value 1213
###### Event: time 1640882425.676810, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 252
###### Event: time 1640882425.676810, -------------- SYN_REPORT ------------
###### Event: time 1640882425.693607, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
###### Event: time 1640882425.693607, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 0
###### Event: time 1640882425.693607, -------------- SYN_REPORT ------------
__________________________________________
 
Am I doing something wrong ? I already checked the exisiting issues and none of them actually helped.

ardera, 2021-12-30:
Hey,
to be sure, have you tried running without the `-i <...>` option? 

Can you try running `libinput debug-events`, touch the screen a few times and send me the complete output?

Marouen-Hammami, 2021-12-30:
Thank you for the quick answer.

> to be sure, have you tried running without the `-i <...>` option?
Yes i did and nothing changed.
> Can you try running `libinput debug-events`, touch the screen a few times and send me the complete output?

-event0   DEVICE_ADDED     Telink Wireless Receiver Mouse    seat0 default group1  cap:p left scroll-nat scroll-button
-event1   DEVICE_ADDED     Telink Wireless Receiver Consumer Control seat0 default group1  cap:kp scroll-nat
-event2   DEVICE_ADDED     Telink Wireless Receiver System Control seat0 default group1  cap:k
-event3   DEVICE_ADDED     Telink Wireless Receiver          seat0 default group1  cap:k
-event4   DEVICE_ADDED     ADS7846 Touchscreen               seat0 default group2  cap:t ntouches 1 calib
 event4   TOUCH_DOWN       +0.000s      -1 (0) 42.50/27.59 (1741.00/1130.00mm)
 event4   TOUCH_FRAME      +0.000s
 event4   TOUCH_MOTION     +0.020s      -1 (0) 42.58/27.44 (1744.00/1124.00mm)
 event4   TOUCH_FRAME      +0.020s
 event4   TOUCH_MOTION     +0.040s      -1 (0) 42.55/27.56 (1743.00/1129.00mm)
 event4   TOUCH_FRAME      +0.040s
 event4   TOUCH_MOTION     +0.060s      -1 (0) 42.58/27.59 (1744.00/1130.00mm)
 event4   TOUCH_FRAME      +0.060s
 event4   TOUCH_MOTION     +0.080s      -1 (0) 42.41/27.49 (1737.00/1126.00mm)
 event4   TOUCH_FRAME      +0.080s
 event4   TOUCH_UP         +0.097s
 event4   TOUCH_FRAME      +0.097s
 event4   TOUCH_DOWN       +2.100s      -1 (0) 41.87/75.02 (1715.00/3073.00mm)
 event4   TOUCH_FRAME      +2.100s
 event4   TOUCH_MOTION     +2.120s      -1 (0) 41.92/75.24 (1717.00/3082.00mm)
 event4   TOUCH_FRAME      +2.120s
 event4   TOUCH_MOTION     +2.140s      -1 (0) 41.89/75.05 (1716.00/3074.00mm)
 event4   TOUCH_FRAME      +2.140s
 event4   TOUCH_MOTION     +2.160s      -1 (0) 41.87/75.00 (1715.00/3072.00mm)
 event4   TOUCH_FRAME      +2.160s
 event4   TOUCH_MOTION     +2.180s      -1 (0) 41.89/75.05 (1716.00/3074.00mm)
 event4   TOUCH_FRAME      +2.180s
 event4   TOUCH_MOTION     +2.200s      -1 (0) 41.94/75.12 (1718.00/3077.00mm)
 event4   TOUCH_FRAME      +2.200s
 event4   TOUCH_MOTION     +2.220s      -1 (0) 41.85/75.15 (1714.00/3078.00mm)
 event4   TOUCH_FRAME      +2.220s
 event4   TOUCH_UP         +2.237s
 event4   TOUCH_FRAME      +2.237s
 event4   TOUCH_DOWN       +3.920s      -1 (0) 47.02/27.56 (1926.00/1129.00mm)
 event4   TOUCH_FRAME      +3.920s
 event4   TOUCH_MOTION     +3.940s      -1 (0) 47.90/27.34 (1962.00/1120.00mm)
 event4   TOUCH_FRAME      +3.940s
 event4   TOUCH_MOTION     +3.960s      -1 (0) 48.00/27.29 (1966.00/1118.00mm)
 event4   TOUCH_FRAME      +3.960s
 event4   TOUCH_MOTION     +3.980s      -1 (0) 48.00/27.27 (1966.00/1117.00mm)
 event4   TOUCH_FRAME      +3.980s
 event4   TOUCH_MOTION     +4.000s      -1 (0) 47.85/27.54 (1960.00/1128.00mm)
 event4   TOUCH_FRAME      +4.000s
 event4   TOUCH_UP         +4.037s
 event4   TOUCH_FRAME      +4.037s
 event4   TOUCH_DOWN       +4.680s      -1 (0) 45.53/76.42 (1865.00/3130.00mm)
 event4   TOUCH_FRAME      +4.680s
 event4   TOUCH_MOTION     +4.700s      -1 (0) 45.53/76.32 (1865.00/3126.00mm)
 event4   TOUCH_FRAME      +4.700s
 event4   TOUCH_MOTION     +4.720s      -1 (0) 45.46/76.29 (1862.00/3125.00mm)
 event4   TOUCH_FRAME      +4.720s
 event4   TOUCH_MOTION     +4.740s      -1 (0) 45.56/76.29 (1866.00/3125.00mm)
 event4   TOUCH_FRAME      +4.740s
 event4   TOUCH_MOTION     +4.760s      -1 (0) 45.51/76.25 (1864.00/3123.00mm)
 event4   TOUCH_FRAME      +4.760s
 event4   TOUCH_UP         +4.777s
 event4   TOUCH_FRAME      +4.777s

ardera, 2021-12-30:
Mh okay nothing unusual there. It probably has something to do with the fact that it's single touch, I've never tried running with that, but not sure what's the problem exactly.

I'll add some logging to `user_input.c`

Marouen-Hammami, 2022-01-01:
So , I added some logging to `user_input.c` in function `process_libinput_events()`.
When randomly taping on the screen in different positions, I got a response. After further digging i noticed that the screen responds but the x and y axis are swapped.

Marouen-Hammami, 2022-01-01:
UPDATE:
I swapped X and Y and inverted the Y axis. I also had to manually set `display_width` and `display_height` in line 833 and 834 of `user_input.c`. I think I have to do the same thing on `on_touch_motion()` to get more reliable results.

ardera, 2022-01-01:
If the touchscreen doesn't report correct coordinates, you should calibrate it, see [here](https://github.com/ardera/flutter-pi/issues/164#issuecomment-841879291). What changes did you have to make to `display_width` and `display_height` though?

Marouen-Hammami, 2022-01-02:
> If the touchscreen doesn't report correct coordinates, you should calibrate it, see [here](https://github.com/ardera/flutter-pi/issues/164#issuecomment-841879291). 
I tried the calibration process that you mentioned. The X and Y are now in the right place but : 
- X range is actually Y range.
- Y axis is inverted. the 0 is on top and max at bottom 
- Y range is X range.

The transforamtion matrix doesn't actually invert axes, it either translate or / and rotate them.

>What changes did you have to make to `display_width` and `display_height` though?

They had 0 as a value which is a bit odd. So i replaced them with 1024 and 600. 
here's how i made it work :
`y = 599 - libinput_event_touch_get_x_transformed(touch_event, 600-1);`
`x = libinput_event_touch_get_y_transmformed(touch_event,1024-1);`
I know this is not the most elegant way to do it , but it might help cover more use cases.

ardera, 2022-01-02:
> The transforamtion matrix doesn't actually invert axes, it either translate or / and rotate them.

Sorry but that's wrong. Those are pretty basic use cases and that's the reason why libinput uses a 3x3 matrix, even though we only have 2-dimensional coordinates. It's also the reason computer games use 4x4 matrices for 3-dimensional coordinates.

Even if you're screen had some weird setup, with the transformation matrix you can basically represent any equation system of the following form: *(Note that x, y, x' and y' are normalized device coordinates, so they are in the range 0..1)*
```
x' = a*x + b*y + c*1
y' = d*x + e*y + f*1
```

For example, if you want `y'` to be the inverted `x` range, you set `d = -1`, `e = 0` and `f = 1`. That'll make the bottom equation `y' = 1 - x`. 

> Y axis is inverted. the 0 is on top and max at bottom

Not sure that's a problem, the origin (0, 0) should be at the top left, so if 0 is on top and max is at the bottom that's all fine I think.

The props you'd need to specify are:
1. If you want to swap X and Y and also invert Y:
    ```
    ATTRS{name}=="device name", ENV{LIBINPUT_CALIBRATION_MATRIX}="0.0 1.0 0.0 -1.0 0.0 1.0"
    ```
2. If you just want to swap X and Y, you need to set:
    ```
    ATTRS{name}=="device name", ENV{LIBINPUT_CALIBRATION_MATRIX}="0.0 1.0 0.0 1.0 0.0 0.0"
    ```

> They had 0 as a value which is a bit odd. So i replaced them with 1024 and 600.

`display_width` and `display_height` are 0 at first (when `user_input_new` is called) because user input is initialized before the display is initialized. They only get filled with real values inside [`flutterpi_fill_view_properties`](https://github.com/ardera/flutter-pi/blob/821a5c6a222e5ac05471ee7610b64e745e63bc4d/src/flutter-pi.c#L1185). Did you maybe read them out before `flutterpi_fill_view_properties` is called?

Marouen-Hammami, 2022-01-03:
oh ,i get it now.
It seems to work fine with the right transformation matrix.
Thank you for the help .

closed 2022-01-03 by Marouen-Hammami

ardera, 2022-01-04:
@Marouen-Hammami np, glad it works now 👍