ardera/flutter-pi #213: "Artifacts" with official 7" touchscreen RPI 4

Toernblom, 2022-01-06:
Hey, getting these artifacts on a specific widget and only on this specific index in the listview. I did have some weird artifacts going on in the "hello-word" flutter gallery aswell. Has anyone else encountered them? Running Flutter 2.8.1 • channel stable • with 64bit buster os.

[image]

ardera, 2022-01-12:
I've seen those before in the flutter gallery, it's a flutter bug I think.
Try fiddling with the rendering a bit, maybe add a repaint boundary or an invisible container below. That's fixed it for some people

ardera, 2022-11-01:
Could be it's caused by this: https://github.com/ardera/flutter-pi/issues/255