ardera/flutter-pi #214: Failed to load dynamic library 'libserialport.so'

eximius313, 2022-01-10:
I'm trying to use [flutter_libserialport](https://pub.dev/packages/flutter_libserialport) library.
I've followed the guidelines of [Libserialport](https://sigrok.org/wiki/Libserialport) and built the library.
I clearly have a file `/usr/local/lib/libserialport.so`
Unfortunately, when I run ` sudo flutter-pi my_apps_flutter_assets` I see red screen with message:
```
Failed to load dynamic library 'libserialport.so'
```
How to force `flutter-pi` to see the `libserialport.so`

Kind regards,

eximius313, 2022-01-13:
Thanks!
```
# cd /usr/lib/
# ln -s /usr/local/lib/libserialport.so .
```
did the job!

closed 2022-01-13 by eximius313

trapeye, 2022-02-04:
I don't have `libdart_vlc_plugin.so` at there folder, but when I create `flutter build -d linux` it create this `libdart_vlc_plugin.so`

do i need to copy paste, or just ln -s

`[ERROR:flutter/lib/ui/ui_dart_state.cc(209)] Unhandled Exception: Invalid argument(s): Failed to load dynamic library '/usr/local/bin/lib/libdart_vlc_plugin.so': /usr/local/bin/lib/libdart_vlc_plugin.so: cannot open shared object file: No such file or directory`

ardera, 2022-02-04:
@trapeye that's a different issue

dart_vlc will not work with flutter-pi without modification. You can't just copy the dart_vlcs plugin .so that was built for the linux desktop embedder, since linux desktop embedder and flutter-pi have different interfaces.

If you want to make dart_vlc working with flutter-pi you need to write some custom C code (in the flutter-pi source tree) to integrate it with flutter-pi's APIs