ardera/flutter-pi #216: Has anyone tested Raspberry Pi Zero 2 W

tobiasht, 2022-01-17:
Just wanted to know if anyone has tested this library with Raspberry Pi Zero 2 W.

ardera, 2022-01-17:
@jwinarske afaik you wanted to test on Pi Zero 2 W? Have you got it running?

pezi, 2022-01-20:
Tested the gallery example on a Pi Zero 2W, only TV attached, no mouse, no keybaord using SSH to start the app (debug).

Some messages, but the app is available on the screen/TV.

`detected VideoCore IV as underlying graphics chip, and VC4 as the driver.
Reporting modified GL_EXTENSIONS string that doesn't contain non-working extensions.
flutter: Observatory listening on http://127.0.0.1:41123/Wzg528JGbcQ=/
[ERROR:flutter/lib/ui/ui_dart_state.cc(209)] Unhandled Exception: FormatException: Missing extension byte (at offset 10)
#0      _Utf8Decoder.convertSingle (dart:convert-patch/convert_patch.dart:1789:7)
#1      Utf8Decoder.convert (dart:convert/utf.dart:318:42)
#2      StringCodec.decodeMessage (package:flutter/src/services/message_codecs.dart:44:25)
#3      JSONMessageCodec.decodeMessage (package:flutter/src/services/message_codecs.dart:99:44)
#4      JSONMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:147:54)
#5      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:167:18)
<asynchronous suspension>
#6      SystemChrome.setApplicationSwitcherDescription (package:flutter/src/services/system_chrome.dart:394:5)
<asynchronous suspension>

[compositor] Could not move cursor to front. Mouse cursor may be invisible. drmdev_plane_get_max_zpos_value: Invalid argument
[compositor] GPU does not supported the desired HW plane order.
             Some UI layers may be invisible.`

tobiasht, 2022-01-21:
Could you check if interaction is working or if it needs some additional work?

pezi, 2022-01-21:
I own a PI compatible touch screen. But currently I have no access to this device - probablay I can test the zero on this device in the next week.

jwinarske, 2022-01-21:
> @jwinarske afaik you wanted to test on Pi Zero 2 W? Have you got it running?

@ardera I have Pi Zero 2 W images building/running with flutter-pi a month ago.  Last build is from 11 days ago:
https://github.com/meta-flutter/meta-flutter/actions/workflows/honister-rpi-zero2w-64.yml

The build failures should go green today with hot fix going into engine beta channel.

I haven't done a smoke test with STM32157MP yet, busy with Vulkan stuff.  Should get to it next week!

tobiasht, 2022-03-10:
If Pi Zero 2 W works great with this library. I think it should be added to the documentation for supported devices.

closed 2022-03-18 by ardera