ardera/flutter-pi #217: Cannot Build Asset Bundle for Raspberry Pi  When Using Package msal_js

raspberrypi9, 2022-01-17:
Hi,

For a while now, I have been successfully building my flutter app's asset bundle and deploying debug and release versions to my Raspberry Pi using flutter-pi.  I now need my app to do Microsoft Authentication, so I recently added the msal_js 2.14.0 package/plugin (https://pub.dev/packages/msal_js) to my project.  Now, on my Mac, my code builds fine and successfully deploys to Chrome.  When I run 'flutter build bundle', however, so that I can run it on flutter-pi, I get the following errors (which are not in my code):

../../.pub-cache/hosted/pub.dartlang.org/http-0.13.4/lib/src/browser_client.dart
:6:8: Error: Not found: 'dart:html'
import 'dart:html';
       ^
../../.pub-cache/hosted/pub.dartlang.org/js-0.6.3/lib/js.dart:8:1: Error: Not
found: 'dart:js'
export 'dart:js' show allowInterop, allowInteropCaptureThis;
^
../../.pub-cache/hosted/pub.dartlang.org/js-0.6.3/lib/js_util.dart:8:1: Error:
Not found: 'dart:js_util'
export 'dart:js_util';
^
../../.pub-cache/hosted/pub.dartlang.org/http-0.13.4/lib/src/browser_client.dart
:32:18: Error: 'HttpRequest' isn't a type.
  final _xhrs = <HttpRequest>{};
                 ^^^^^^^^^^^
../../.pub-cache/hosted/pub.dartlang.org/msal_js-2.14.0/lib/src/utils/error_util
s.dart:18:18: Error: Method not found: 'promiseToFuture'.
    return await promiseToFuture<T>(promise);
                 ^^^^^^^^^^^^^^^
../../.pub-cache/hosted/pub.dartlang.org/msal_js-2.14.0/lib/src/utils/promise_ut
ils.dart:9:26: Error: Method not found: 'allowInterop'.
  return interop.Promise(allowInterop((resolve, reject) {
                         ^^^^^^^^^^^^
../../.pub-cache/hosted/pub.dartlang.org/msal_js-2.14.0/lib/src/logger.dart:46:3
4: Error: The method 'allowInterop' isn't defined for the class 'LoggerOptions'.
 - 'LoggerOptions' is from 'package:msal_js/msal_js.dart'
 ('../../.pub-cache/hosted/pub.dartlang.org/msal_js-2.14.0/lib/msal_js.dart').
Try correcting the name to the name of an existing method, or defining a method
named 'allowInterop'.
      _jsObject.loggerCallback = allowInterop(jsCallback);
                                 ^^^^^^^^^^^^
../../.pub-cache/hosted/pub.dartlang.org/msal_js-2.14.0/lib/src/navigation_clien
t.dart:19:9: Error: Method not found: 'allowInterop'.
        allowInterop((String url, interop.NavigationOptions options) {
        ^^^^^^^^^^^^
../../.pub-cache/hosted/pub.dartlang.org/msal_js-2.14.0/lib/src/navigation_clien
t.dart:24:9: Error: Method not found: 'allowInterop'.
        allowInterop((String url, interop.NavigationOptions options) {
        ^^^^^^^^^^^^
../../.pub-cache/hosted/pub.dartlang.org/msal_js-2.14.0/lib/src/navigation_clien
t.dart:17:10: Error: Method not found: 'jsify'.
  return jsify(<String, dynamic>{
         ^^^^^
../../.pub-cache/hosted/pub.dartlang.org/msal_js-2.14.0/lib/src/network_module.d
art:21:28: Error: Method not found: 'allowInterop'.
    'sendGetRequestAsync': allowInterop((String url,
                           ^^^^^^^^^^^^
../../.pub-cache/hosted/pub.dartlang.org/msal_js-2.14.0/lib/src/network_module.d
art:32:29: Error: Method not found: 'allowInterop'.
    'sendPostRequestAsync': allowInterop((String url,
                            ^^^^^^^^^^^^
../../.pub-cache/hosted/pub.dartlang.org/msal_js-2.14.0/lib/src/network_module.d
art:20:10: Error: Method not found: 'jsify'.
  return jsify(<String, dynamic>{
         ^^^^^
../../.pub-cache/hosted/pub.dartlang.org/msal_js-2.14.0/lib/src/public_client_ap
plication.dart:88:41: Error: The method 'allowInterop' isn't defined for the
class 'PublicClientApplication'.
 - 'PublicClientApplication' is from 'package:msal_js/msal_js.dart'
 ('../../.pub-cache/hosted/pub.dartlang.org/msal_js-2.14.0/lib/msal_js.dart').
Try correcting the name to the name of an existing method, or defining a method
named 'allowInterop'.
      return _jsObject.addEventCallback(allowInterop((message) {
                                        ^^^^^^^^^^^^
../../.pub-cache/hosted/pub.dartlang.org/msal_js-2.14.0/lib/src/request.dart:34:
61: Error: The method 'allowInterop' isn't defined for the class
'EndSessionRequest'.
 - 'EndSessionRequest' is from 'package:msal_js/msal_js.dart'
 ('../../.pub-cache/hosted/pub.dartlang.org/msal_js-2.14.0/lib/msal_js.dart').
Try correcting the name to the name of an existing method, or defining a method
named 'allowInterop'.
      _jsObject.onRedirectNavigate = value == null ? null : allowInterop(value);
                                                            ^^^^^^^^^^^^
../../.pub-cache/hosted/pub.dartlang.org/msal_js-2.14.0/lib/src/request.dart:144
:61: Error: The method 'allowInterop' isn't defined for the class
'RedirectRequest'.
 - 'RedirectRequest' is from 'package:msal_js/msal_js.dart'
 ('../../.pub-cache/hosted/pub.dartlang.org/msal_js-2.14.0/lib/msal_js.dart').
Try correcting the name to the name of an existing method, or defining a method
named 'allowInterop'.
      _jsObject.onRedirectNavigate = value == null ? null : allowInterop(value);
                                                            ^^^^^^^^^^^^
../../.pub-cache/hosted/pub.dartlang.org/http-0.13.4/lib/src/browser_client.dart
:44:15: Error: The method 'HttpRequest' isn't defined for the class
'BrowserClient'.
 - 'BrowserClient' is from 'package:http/src/browser_client.dart'
 ('../../.pub-cache/hosted/pub.dartlang.org/http-0.13.4/lib/src/browser_client.d
 art').
Try correcting the name to the name of an existing method, or defining a method
named 'HttpRequest'.
    var xhr = HttpRequest();
              ^^^^^^^^^^^
../../.pub-cache/hosted/pub.dartlang.org/msal_js-2.14.0/lib/src/js_proxies/js_pr
oxies.dart:49:15: Error: Method not found: 'jsify'.
      value = jsify(value);
              ^^^^^
../../.pub-cache/hosted/pub.dartlang.org/msal_js-2.14.0/lib/src/js_proxies/js_ob
ject_map_proxy.dart:20:21: Error: The method 'getProperty' isn't defined for the
class 'JsObjectMapProxy<V>'.
 - 'JsObjectMapProxy' is from 'package:msal_js/src/js_proxies/js_proxies.dart'
 ('../../.pub-cache/hosted/pub.dartlang.org/msal_js-2.14.0/lib/src/js_proxies/js
 _proxies.dart').
Try correcting the name to the name of an existing method, or defining a method
named 'getProperty'.
    return jsDecode(getProperty(_jsObject, key!));
                    ^^^^^^^^^^^
../../.pub-cache/hosted/pub.dartlang.org/msal_js-2.14.0/lib/src/js_proxies/js_ob
ject_map_proxy.dart:25:5: Error: The method 'setProperty' isn't defined for the
class 'JsObjectMapProxy<V>'.
 - 'JsObjectMapProxy' is from 'package:msal_js/src/js_proxies/js_proxies.dart'
 ('../../.pub-cache/hosted/pub.dartlang.org/msal_js-2.14.0/lib/src/js_proxies/js
 _proxies.dart').
Try correcting the name to the name of an existing method, or defining a method
named 'setProperty'.
    setProperty(_jsObject, key, jsEncode(value));
    ^^^^^^^^^^^
Unhandled exception:
FileSystemException(uri=org-dartlang-untranslatable-uri:dart%3Ahtml;
message=StandardFileSystem only supports file:* and data:* URIs)
#0      StandardFileSystem.entityForUri
(package:front_end/src/api_prototype/standard_file_system.dart:34:7)
#1      asFileUri (package:vm/kernel_front_end.dart:614:37)
#2      writeDepfile (package:vm/kernel_front_end.dart:754:21)
<asynchronous suspension>
#3      FrontendCompiler.compile
(package:frontend_server/frontend_server.dart:571:9)
<asynchronous suspension>
#4      starter (package:flutter_frontend_server/server.dart:85:12)
<asynchronous suspension>
#5      main
(file:///opt/s/w/ir/cache/builder/src/flutter/flutter_frontend_server/bin/starte
r.dart:13:24)
<asynchronous suspension>

Target kernel_snapshot failed: Exception
Failed to build bundle.

I'd be very grateful for any assistance you can offer in resolving these errors.  Thanks so much for your help.

ardera, 2022-01-18:
Hey,
msal_js only works for flutter web I'm afraid. Look at the line below the publishing time: 
[image]

If it was working with other platforms too, it'd say something like "android, ios, linux, windows, " etc right there.

raspberrypi9, 2022-01-18:
Flutter-pi clearly isn't web, windows, android, or iOS.  It doesn't seem like it is 100% Linux either, so which of those platform labels must appear on the pub.dev page so that I know a package is compatible with flutter-pi?  Thanks.

ardera, 2022-01-18:
It's not easy or maybe impossible to tell by that platform label
every plugin with native code has to have custom native code for every platform it supports. Most plugin authors don't do that for flutter-pi (obviously) so they don't work with flutter-pi. Exceptions are for example the video_player plugin, I've written native code to support that (you need to use the omxplayer_video_player package though, there's also a new one based on gstreamer I'm working on)

Plugins that don't contain any native code should work fine on flutter-pi

closed 2022-02-28 by ardera