ardera/flutter-pi #218: not able to use dart_vlc (dart_vlc_ffi) together with flutter-pi to play video in widget

radeksalomon, 2022-02-04:
We are not able to pass the obstacles to run video in flutter widget on rpi with flutter-pi. We run flutter-pi on archlinuxarm 32 bit armv7h and when we try to run the video in flutter, the error appears that we are missing `/usr/local/bin/lib/libdart_vlc_plugin.so`. I guess the missing lib is the issue, but we do not know how to build for it. Any kind of your help is much appriciated.

I also noticed flutter-pi has some support for gstreamer in you feature branch. Play-pause of file or stream is what we need to use. So maybe this should be the way, how to implement the video to the widget.

ardera, 2022-02-04:
`dart_vlc` looks for that lib in this line, because it assumes it's running on the linux desktop embedder: https://github.com/alexmercerind/dart_vlc/blob/61f8180324c89aaad897663c3f37c0387fad536c/lib/dart_vlc.dart#L101

That library is only built when using the linux desktop embedder and only works with the linux desktop embedder, it doesn't work with flutter-pi (at all).

It's definitely possible to make dart_vlc working with flutter-pi, but that requires modification to both dart_vlc and flutter-pi, so let me know if you want to go down that route.

As I mentioned on discord, there's two video players for flutter-pi: one based on omxplayer and one based on gstreamer.
- I didn't test the omxplayer one in a long time, it also was a bit of a hack. Also omxplayer is deprecated now
- the gstreamer one is the one I'm working on right now. It works pretty well, I'd definitely recommend it over the omxplayer video player. Though I haven't implemented audio yet, if you need that

radeksalomon, 2022-02-05:
Omxplayer is not the way, we should go further. I agree, it is deprecated and does not make sense to develop with it. 
We like that you implemented gstreamer to the flutter-pi. Gstreamer is great tool and has some good advantages comparing to vlc (gapless play etc.). So we would like to implement it. Our solution does not support audio yet, so it looks like good match :). We will be happy to support you with further development, further tests etc.

DisDis, 2022-02-07:
Hi @ardera , Where can I find an example of gstreamer video playback in flutter-pi?
This topic is very interesting to me. I can be a tester.

ardera, 2022-03-18:
Closing this since dart_vlc is not supported right now.

closed 2022-03-18 by ardera