ardera/flutter-pi #219: flutter-pi in debug mode no longer runs on flutter 2.10, it only runs on 2.8.1

gmmota, 2022-02-05:
OpenGL ES information:
          version: "OpenGL ES 2.0 Mesa 20.3.5"
          shading language version: "OpenGL ES GLSL ES 1.0.16"
          vendor: "Broadcom"
          renderer: "VC4 V3D 2.1"
          extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888         GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_EGL_image GL_OES_depth_texture GL_AMD_performance_monitor GL_OES_packed_depth_stencil GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_ANGLE_pack_reverse_row_order GL_EXT_occlusion_query_boolean GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_NV_pixel_buffer_object GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_separate_shader_objects GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_KHR_parallel_shader_compile GL_MESA_tile_raster_order "

        detected VideoCore IV as underlying graphics chip, and VC4 as the driver.
        Reporting modified GL_EXTENSIONS string that doesn't contain non-working extensions.
        [ERROR:flutter/shell/common/shell.cc(94)] Dart Error: Can't load Kernel binary: Invalid kernel binary format version.
        [ERROR:flutter/runtime/dart_isolate.cc(177)] Could not prepare isolate.
        [ERROR:flutter/runtime/runtime_controller.cc(384)] Could not create root isolate.
        [ERROR:flutter/shell/common/shell.cc(574)] Could not launch engine with configuration.
        flutter: Observatory listening on http://127.0.0.1:36121/7HrzW9dHtkQ=/

gmmota, 2022-02-05:
downgrade to 2.81 :(

ardera, 2022-02-05:
Yeah this is related to https://github.com/ardera/flutter-engine-binaries-for-arm/issues/7.
I'll have to build the engine binaries for 2.10

ardera, 2022-02-06:
fixed by [ardera/flutter-engine-binaries-for-arm/8358952893831b4486ac097abc7a438dd25d251c](https://github.com/ardera/flutter-engine-binaries-for-arm/commit/8358952893831b4486ac097abc7a438dd25d251c)

closed 2022-02-06 by ardera